tagyoureit/nodejs-poolController #114: Pool and Spa not showing separate and/or last temperature readings

afgoody, 2019-02-03:
I have two bodies of water, but one pump. My Pentair EasyTouch 4 reports either the current temp (green in ScreenLogic iOS app) for whichever mode (pool or spa) is running or else the last known temp for the respective mode the last time it ran (white in ScreenLogic iOS app). This works well.

But in nodejs-poolController, when these two temps are different from each other in ScreenLogic, they're still the same as each other in nodejs-poolController and often also different from ScreenLogic. I'll post a ScreenLogic desktop screenshot on top of a nodejs-poolController refreshed page and the ScreenLogic iOS app showing the same moment with all different temperature readings.

1. What version of the code are you using?
5.3.0

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
I don't know how to get these files off the RPi SD card through an SSH session.

3. Any errors that are in the console
None

4. Your config.json or in-use configuration file
Attached (copy/paste from SSH session into .txt)
[custom.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/2825212/custom.json.txt)

5. Your pool equipment
Pentair 520538 EasyTouch 4 - Pool/Spa

[image]
[image]

dhop90, 2019-02-05:
I'm seeing this as well

ronf71, 2019-02-23:
Seeing the same thing.  Both pool temp and spa temp are showing whatever the spa temp is.

tagyoureit, 2019-05-17:
I included this in the latest build of the [6.0-Dev](https://github.com/tagyoureit/nodejs-poolController/tree/6.0-DEV) build.  Please test it out.

afgoody, 2019-10-01:
> I included this in the latest build of the [6.0-Dev](https://github.com/tagyoureit/nodejs-poolController/tree/6.0-DEV) build. Please test it out.

Will 6.0 work with ActionTiles and all other capabilities currently in 5.3?

GoodyGizmos, 2020-02-02:
@tagyoureit How is development of 6.0 going?

tagyoureit, 2020-02-02:
>How is development of 6.0 going?
Coming along... going to push up some new changes here shortly.  

>Will 6.0 work with ActionTiles and all other capabilities currently in 5.3?
Yes, and hopefully even better.  ActionTiles is purely based on the SmartThings devices and I'm currently working on that part of it.

GoodyGizmos, 2020-02-02:
>

Nice. I can’t get 5.3 to stay running on my RPi reliably enough to use even with PM2 (logs off), so any chance I could start trying out 6.0?

tagyoureit, 2020-02-02:
Of course... I need beta testers. :)
There are a few things you'll need.
1. [Next](https://github.com/tagyoureit/nodejs-poolController/tree/next) branch
2. One of the [web clients](https://github.com/tagyoureit/nodejs-poolController/tree/next#clients)
3. If you want to try the ST integration, it's inside a script in the [scripts](https://github.com/tagyoureit/nodejs-poolController-webClient/tree/master/scripts) directory on my web client.  Run it separately with Node (check the IP's in the file).

Keep everything on the same box for now.  Later, we'll be able to separate the client/server but I need to do a little more plumbing to make that happen.

GoodyGizmos, 2020-02-09:
@tagyoureit I was able to find the output.log file that was filling my RPi and get 5.X running again w/ PM2. But, I assumed I could still start beta testing 6.0. I updated Node and NPM, but don't see a 6.0 source code release here...[https://github.com/tagyoureit/nodejs-poolController/releases](url) to download.

tagyoureit, 2020-02-09:
You will need to use ‘git clone xyz’ where xyz is the Branch that you want to clone. Look for the button in the upper right with the path. My power is out now so I can work more exact instructions later if you are stuck.

tagyoureit, 2020-05-13:
This issue should have been fixed in v5, but will work in v6.

closed 2020-05-13 by tagyoureit