tagyoureit/nodejs-poolController #115: Uncaught Exception

afgoody, 2019-02-10:
I'm running nodejs-poolController on my RPi using tmux. It seems to stop about once a day. The error messages are below. Any suggestions to try based on the error codes below?

15:56:15.262 ERROR uncaughtException: Cannot read property 'name' of undefined
        {"date":"Sat Feb 09 2019 15:56:15 GMT-0800 (PST)","process":{"pid":4559,"uid":0,"gid":0,"cwd":"/home/pi/nodejs-poolController-5.3.0","execPath":"/usr/bin/node","version":"v8.15.0","argv":["/usr/bin/node","/home/pi/nodejs-poolController-5.3.0/src/index.js","custom.json"],"memoryUsage":{"rss":73093120,"heapTotal":26542080,"heapUsed":21790944,"external":4340130}},"os":{"loadavg":[0.02099609375,0.02294921875,0],"uptime":412901},"trace":[{"column":45,"file":"/home/pi/nodejs-poolController-5.3.0/src/lib/comms/server.js","function":null,"line":150,"method":null,"native":false},{"column":null,"file":null,"function":"Array.forEach","line":null,"method":"forEach","native":false},{"column":28,"file":"/home/pi/nodejs-poolController-5.3.0/src/lib/comms/server.js","function":null,"line":148,"method":null,"native":false},{"column":13,"file":"events.js","function":"emitTwo","line":126,"method":null,"native":false},{"column":7,"file":"events.js","function":"EventEmitter.emit","line":214,"method":"emit","native":false},{"column":43,"file":"/home/pi/nodejs-poolController-5.3.0/node_modules/multicast-dns/index.js","function":null,"line":49,"method":null,"native":false},{"column":13,"file":"events.js","function":"emitTwo","line":126,"method":null,"native":false},{"column":7,"file":"events.js","function":"Socket.emit","line":214,"method":"emit","native":false},{"column":8,"file":"dgram.js","function":"UDP.onMessage [as onmessage]","line":659,"method":"onMessage [as onmessage]","native":false}],"stack":["TypeError: Cannot read property 'name' of undefined","   at /home/pi/nodejs-poolController-5.3.0/src/lib/comms/server.js:150:45","    at Array.forEach (<anonymous>)","    at EventEmitter.<anonymous> (/home/pi/nodejs-poolController-5.3.0/src/lib/comms/server.js:148:28)","    at emitTwo (events.js:126:13)","    at EventEmitter.emit (events.js:214:7)","    at Socket.<anonymous> (/home/pi/nodejs-poolController-5.3.0/node_modules/multicast-dns/index.js:49:43)","    at emitTwo (events.js:126:13)","    at Socket.emit (events.js:214:7)","    at UDP.onMessage [as onmessage] (dgram.js:659:8)"]}
nodejs-poolController has closed successfully.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nodejs-poolcontroller@5.2.0 start: `node src/index.js "custom.json"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nodejs-poolcontroller@5.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-02-09T23_56_15_324Z-debug.log

tagyoureit, 2019-02-10:
Usually when I've seen this it is because you have more circuits or schedules configured in the `custom.json` than you actually have on your pool.  

Feel free to follow a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach it here and I'll track it down.

tagyoureit, 2019-03-14:
Closing due to lack of activity.

closed 2019-03-14 by tagyoureit