tagyoureit/nodejs-poolController #116: Intellicenter Messages

rstrouse, 2019-02-21:
Tag,

Here are the files from the capture.
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/2891549/replay.zip)

tagyoureit, 2019-02-22:
There seems to be a LOT of new commands here.  Some of them may be similar but are not packets that we've seen/decoded yet.  

Do you have Screenlogic or a remote panel?  If so, redo the packet capture and wait until all the packets from the app fail.  Then unplug/replug in your screen logic (or change the pool/spa temp) and this should trigger a broadcast of all the configuration packets again.  Then end the packet capture and re-upload it here.

rstrouse, 2019-03-02:
Intellicenter does not use ScreenLogic.  It has its own hosting www.intellicenter.com (there is a demo) and has its own mobile app.  Included in the zip is a captureChanges.js which can be run with multiple options to filter out specific types of messages, a word doc explaining what I have discovered so far with the configuration, and a capture of the related messages for the configuration.
[Pentair Messages.zip](https://github.com/tagyoureit/nodejs-poolController/files/2921725/Pentair.Messages.zip)

rstrouse, 2019-03-02:
Didn't mean to close it.

rstrouse, 2019-03-04:
Here is an update document of the configuration data from Intellicenter.  It is by no means complete but this probably encompasses everything that is required to manage the panel.
[Pentair Messages.docx](https://github.com/tagyoureit/nodejs-poolController/files/2924465/Pentair.Messages.docx)

tagyoureit, 2019-03-13:
I've started to work on this.  Your doc is very helpful . I'm at the point of just duct taping everything together but if you want to see the progress I'll push up the beta build for you to explore/review to see if the data is right.

Also, if you can send another packet capture with all of the settings I'll need those eventually.  The little .js program you wrote is great.  Only minor tweak is that the json should output `packet` instead of `Broadcast` so I don't need to do a find/replace.  (I do try to track packets, socket calls and other comms hence the type.)

rstrouse, 2019-03-14:
I've begun decoding even more messages and have decoded pretty much all of the system configuration items as well as much of the system status.  I have also figured out the versioning at the panel so that it only sends back information that is new.  My usb stick took a dump and am waiting on a new one to arrive but as soon as I get it I will put up another doc and a packet capture.

As for the program I wasn't sure what that data was used for but it makes much more sense as I look deeper into the messages transmitted back and forth and your code.  I wasn't using that data for anything as I was simply reading the packets into a hacked up .NET app so that I could quickly analyze the payloads.
[image]

tagyoureit, 2019-03-15:
Nice... yes, I can use your replay.json with [npm run start:replay](https://github.com/tagyoureit/nodejs-poolController/blob/master/package.json#L9) and then load up [replay.html](https://github.com/tagyoureit/nodejs-poolController/blob/master/src/www/public/replay.html) and replay the packets as if they were coming directly from your pool controller.

rstrouse, 2019-03-17:
Well here is the information I promised.  Weird the usb dongle just started being intermittent.  New one fixed the issue.  You may find that the changes are so far reaching with the previous generation that this is a bridge too far.  I couldn't imagine doing this without the available equipment.  There is crazy stuff going on like it sends specific equipment packets to see if the equipment is installed then asks for the rest of the configuration later.

Anyhow here it is and don't feel obligated.
[Intellicenter Messages.zip](https://github.com/tagyoureit/nodejs-poolController/files/2975584/Intellicenter.Messages.zip)

tagyoureit, 2019-03-18:
Thanks for posting this.  Since this intellicenter stuff is so different I need to separate out the internal storage layer from the current implementation.  That way, either the Intellitouch or Intellicenter logic can both update a consistent/known implementation of the internal circuit, pump, etc objects.  In this process, much of the code is pretty outdated and a lot of efficiencies can be gained by implementing es5/es6 patterns and I'm also taking the opportunity to add TypeScript to the project as that will significantly reduce the number of coding errors that I've experienced in the past.

Long story short - stay tuned.  I'll get this added but it will be a long process.

And yes, doing all of this without the equipment is really hard.  I have done stuff in the past with users reporting what their equipment is doing by visually watching it and it was a nightmare.  But with a guide like you put together and the replay functionality it should be much simpler (but still far from simple).

tagyoureit, 2019-06-04:
Just FYI, I've started working on this.  And yes, it is pretty tricky!  But I'm starting to get the flow of how the packets work.  Hopefully will have an initial something to test here shortly.

tagyoureit, 2019-06-07:
Ok - I've got a few questions and I'm sure there will be a few more...
From your Word Doc...
4.3.3 - Is there any setting to display circuit/features in the Pool/Spa section of the app?  This is possible in screenlogic but maybe it was removed in Intellicenter?
4.3.4 - Circuit names.  I'm just a little confused on which name goes with which circuit.
```
// first packet
165,63,15,16,30,34,1,3,83,112,97,0,0,0,0,0,0,0,0,0,0,0,0,0,83,112,97,32,74,101,116,115,0,0,0,0,0,0,0,0,5,69
// is "Spa" and "Spa Jets" --> is this circuit 1 and 2?  
...
// second packet
165,63,15,16,30,34,1,4,65,85,88,32,50,0,0,0,0,0,0,0,0,0,0,0,65,85,88,32,51,0,0,0,0,0,0,0,0,0,0,0,3,201
"AUX 2", "AUX 3"
...
// third packet
165,63,15,16,30,34,1,5,80,111,111,108,32,76,105,103,104,116,0,0,0,0,0,0,80,111,111,108,0,0,0,0,0,0,0,0,0,0,0,0,6,149
"Pool Light", "Pool"
...
// sixth packet
165,63,15,16,30,34,1,6,83,112,97,32,76,105,103,104,116,0,0,0,0,0,0,0,70,105,114,101,32,66,111,119,108,115,0,0,0,0,0,0,8,51
"Spa Light", "Fire Bowls"
```
thx.

rstrouse, 2019-06-10:
Russ,

Sorry for the delay but like always nobody can just be whelmed.  You are either over or under and I am over at the moment.  Anyhow,  I did spend some more time testing and sending messages to the panel so I have a more complete document.  I will try to answer your questions as I have observed them.
1.  **Is there any setting to display circuit/features in the pool/spa section of the app?**  There is but it manifests itself differently.  Intellicenter has split the old method of paging between displays into 2 distinct sections.  Lights and Features.  Circuits can be hidden in that they can be created but not shown in the features menu.  You can create a circuit that simply controls the position of the valves or flips a relay that users will never see in the interface.  See 4.3.3 and 4.3.9.  For lights you can display them in the feature menu but if they are one of the identified light types then they will always be shown in the light menu.  I think they may be gearing up to allow some more advanced lighting control in the future.  The light group functions look an awful lot like what you would find in DMX control.

2.  **Circuit Names - I am just a little confused as to which name goes with which circuit**  It took me a bit to figure out what they were doing with this.  As it turns out there are two names returned per packet.  Intellicenter does not really make a distinction between the ID and the index of the circuit.  For the purposes of a C programmer who looks at the world from address offsets then index 0 is the first circuit or circuit 1.  In the doc I refer to index when I am looking at the offset and ID when I am referring to the quantitative.  Long story short index 0 is the first circuit and your assessment of the document is correct.

[Intellicenter Communications.docx](https://github.com/tagyoureit/nodejs-poolController/files/3273220/Intellicenter.Communications.docx)

tagyoureit, 2019-06-10:
Thanks, I'll take a look at this new document.  Would you be interested in having it hosted somewhere that we can collaborate on it (Google Sheets?  Dropbox?).  I can host it if you don't have a way to do it.  I have lots of little comments/suggestions that would be easier to see/ask in context.

Also, I am a making my way through the document and up to the Pumps section.  Like you noted, I'm trying to figure these out and more importantly maintain compatibility between intellitouch/intellicenter.  

I'll try to push up something soon so you can test against what you see in intellicenter and we can course correct along the way.

rstrouse, 2019-06-10:
I don't have a problem with that but I don't have any way to host it.  I can actually send you a node.js project that performs all the functions on Intellicenter.  I used it to create this down and dirty dashboard.
[image]

tagyoureit, 2019-06-10:
Sure... that would be great.  And if you [email me](mailto:russ.goldin@gmail.com) I'll setup a hosted version of the doc and invite you as a collaborator.

0xJoeMartin, 2019-10-11:
What progress have you guys made on this Intellicenter node server?  Anything I can help with?

rstrouse, 2019-10-12:
We pretty much have all the messaging decoded as well as the command structure.  At this point it can control and report all devices on the pool with the exception of IntelliChem.  

Which version of IntelliCenter do you have and do you have IntelliChem?  This has been written entirely with an i10PS but I don't suspect there to be many if any differences between models.

0xJoeMartin, 2019-10-13:
I have the i8PS and I also have IntelliChem.

How are you guys capturing the messaging?  I have tried looking at the requests by logging in to prod.intellicenter.com in a browser and using the mobile app.  It looks like it switches over to websockets but then after that I can't see anything.  Every WS plugin/decoder I have tried comes up empty.

rstrouse, 2019-10-13:
This project is using a pi attached to an RS-485 stick.  The D+ and D- lines of the bus are connected directly.  It then communicates of the RS-485 bus.

We are not using the underlying rest api.

theRealMCQwerty, 2019-12-01:
@tagyoureit Do you have an alpha version of v6 you could share? My IntelliCenter is on the way and I'm hoping to supplement it with this project. I already have a USB RS485 adapter and I have it hooked up to an intellivalve. I am seeing what looks like a heartbeat message periodically from the valve but hoping to get into it a bit more. Thanks

tagyoureit, 2019-12-01:
Absolutely.  [Next branch](https://github.com/tagyoureit/nodejs-poolController/tree/next).  Note there are two web clients  the app now (linked in the readme).  You'll probably want to use @rstrouse's because he has an Intellicenter.

Neither of us have Intellivalves so it will be interesting to see what those comms look like.

cheers.

rstrouse, 2019-12-01:
> @tagyoureit Do you have an alpha version of v6 you could share? My IntelliCenter is on the way and I'm hoping to supplement it with this project. I already have a USB RS485 adapter and I have it [...]

Actually I have 7 IntelliValves hooked up on my pool.  I am waiting on the new firmware that was supposed to be released by Pentair this year to start tackling it.  I suspect that the heartbeats are probably reporting position this will probably look a lot like the pump reporting.  I expect to see valve position, mode, and whether it is transitioning in that message.

theRealMCQwerty, 2020-03-06:
@tagyoureit Finally got my IntelliCenter setup and connected the RS485 adapter. I am running the latest code from the 'next' branch but not having much luck. This is what I see in the console:
[image]
The GET is when I tried to access http://localhost:4200/ in a browser. All I got in the browser windows was: `Cannot GET /`

Any idea what I'm doing wrong?

Full config.json below:
```
{
  "controller": {
    "comms": {
      "rs485Port": "COM7",
      "portSettings": {
        "baudRate": 9600,
        "dataBits": 8,
        "parity": "none",
        "stopBits": 1,
        "flowControl": false,
        "autoOpen": false,
        "lock": false
      },
      "mockPort": false,
      "netConnect": false,
      "netHost": "raspberrypi",
      "netPort": 9801,
      "inactivityRetry": 10
    }
  },
  "web": {
    "servers": {
      "http2": {
        "enabled": false
      },
      "http": {
        "enabled": true,
        "ip": "127.0.0.1",
        "port": 4200,
        "httpsRedirect": false,
        "authentication": "none",
        "authFile": "/users.htpasswd"
      },
      "https": {
        "enabled": true,
        "ip": "127.0.0.1",
        "port": 4201,
        "authentication": "none",
        "authFile": "/users.htpasswd",
        "sslKeyFile": "",
        "sslCertFile": ""
      },
      "mdns": {
        "enabled": false
      },
      "ssdp": {
        "enabled": false
      }
    },
    "services": {}
  },
  "log": {
    "packet": {
      "enabled": true,
      "logToConsole": false,
      "logToFile": false,
      "filename": "packetLog",
      "replay": false,
      "broadcast": {
        "enabled": true,
        "includeActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeActions": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "pump": {
        "enabled": true,
        "includeActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeActions": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "chlorinator": {
        "enabled": false,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "unknown": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      }
    },
    "app": {
      "enabled": true,
      "level": "info"
    }
  },
  "appVersion": "0.0.1"
}
```

Thanks

theRealMCQwerty, 2020-03-06:
> Actually I have 7 IntelliValves

@rstrouse Have you had a chance to look into this more? I have been watching for the illusive Pentair update for months but it seems no closer now than before.
I was wondering if it may be possible to figure out the commands to move the intelliValves to a given and bypass the IntelliCenter, have you looked into this at all?

Thanks

tagyoureit, 2020-03-06:
For your first question, you should download one of the [two clients](https://github.com/tagyoureit/nodejs-poolController/tree/next#web-clients).  

You should be able to get http://localhost:4200/state/all or http://localhost:4200/config/all to get the configurations.  I haven't documented the API's yet.

theRealMCQwerty, 2020-03-07:
> You should be able to get http://localhost:4200/state/all or http://localhost:4200/config/all to get the configurations. I haven't documented the API's yet.

I was able to get a response on both of the above. Below is the `equipment` section:
```
"equipment": {
		"model": "IntelliTouch i5+3S",
		"controllerType": "intellitouch",
		"shared": false,
		"maxBodies": 2,
		"equipmentType": "equipment"
	},
```

Is as expected that my IntelliCenter is detected as an IntelliTouch? I actually have an IntelliCenter 8PS with a additional single 6-valve expansion board installed. 

> download one of the [two clients](https://github.com/tagyoureit/nodejs-poolController/tree/next#web-clients).

I downloaded both :-) But am confused how to get them setup.  Tried various combinations of `nmp install`, copying various files into the `web` dir of the maincontroller and also opening the index.html files directly. None are working. 
Sorry, node.js is not a stack I am familiar with so I'm a real noob.

Thanks!

tagyoureit, 2020-03-07:
Yes, this should be picking up the Intellicenter.  Try to delete the poolConfig.json and poolState.json in the data directory and restart.  If it still picks up Intellitouch by mistake, set `logToFile: true` in the config.json and attach it here.  I'll replay the packets locally and see what's up.

For the clients, they both run separately from this app.  For now, they should be on the same machine until we document all the configurations.  For the current version of nodejs-poolController-webclient run `npm run dev`.  This will change in the next version I push up.  

For dashpanel it should just be `npm start`.

theRealMCQwerty, 2020-03-07:
> Try to delete the poolConfig.json and poolState.json in the data directory and restart.
Tried twice and both times still picks up Intellitouch

> set `logToFile: true` in the config.json and attach it here
[packetLog(2020-03-07_14-08-53).log](https://github.com/tagyoureit/nodejs-poolController/files/4302460/packetLog.2020-03-07_14-08-53.log)

> For the current version of nodejs-poolController-webclient run `npm run dev`.
[image]
But was still able to open the site:
[image]

> For dashpanel it should just be `npm start`.
[image]

From referenced log:

```
0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Users\\M\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'start' ]
2 info using npm@6.13.1
3 info using node@v8.16.2
4 verbose stack Error: missing script: start
4 verbose stack     at run (C:\Users\M\AppData\Roaming\npm\node_modules\npm\lib\run-script.js:155:19)
4 verbose stack     at C:\Users\M\AppData\Roaming\npm\node_modules\npm\lib\run-script.js:63:5
4 verbose stack     at C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\read-package-json\read-json.js:116:5
4 verbose stack     at C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\read-package-json\read-json.js:435:5
4 verbose stack     at checkBinReferences_ (C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\read-package-json\read-json.js:391:45)
4 verbose stack     at final (C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\read-package-json\read-json.js:433:3)
4 verbose stack     at then (C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\read-package-json\read-json.js:161:5)
4 verbose stack     at C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\read-package-json\read-json.js:281:12
4 verbose stack     at C:\Users\M\AppData\Roaming\npm\node_modules\npm\node_modules\graceful-fs\graceful-fs.js:115:16
4 verbose stack     at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:511:3)
5 verbose cwd C:\nodejs-poolControl.dashPanel
6 verbose Windows_NT 10.0.18363
7 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\M\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "start"
8 verbose node v8.16.2
9 verbose npm  v6.13.1
10 error missing script: start
11 verbose exit [ 1, true ]
```

tagyoureit, 2020-03-10:
Pull down the latest code from each branch to start.

For the packet log, I'm not seeing much in there.  Can you try again and also set the `replay:true` and send along the *.json replay file that will be in the /replay directory?  Also, please make sure it is capturing all pump/controller/etc messages as specified in the config.json.

The nodejs-poolController-webclient looks fine starting up.  It isn't talking to the main app, though.  Do you see any errors if you open up the developer console in Chrome?
 
For the dashPanel, you'll need to run `tsc` (typescript compile; if you don't have typescript then first `npm i typescript`) and then `node app.js`.

theRealMCQwerty, 2020-03-12:
> Pull down the latest code from each branch to start.

Done

> send along the *.json replay file

[packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/4321932/packetCapture.zip)

---------------------------------------
nodejs-poolController-webclient connected this time... strange but at least it is connecting now.

dashpanel: your instructions worked, I am able to load the dashpanel also now, thanks!

Neither UI is functional though of course since the controller still thinks I have an IntelliTouch, hopefully the packet capture helps figure out the mystery.

-----------------------------
By the way, not sure if it matters but I have my IntelliCenter setup on my bench. The only RS485 devices connected to the bus are a single IntelliValve and the USB adapter. There is no power to my equipment pad yet so I am unable to hook up the pumps or the IntelliChem.

tagyoureit, 2020-03-15:
I ran the packet capture and you have a type of IntelliCenter that is new to me.  I've updated the code so please to pull it and try again.  We'll have to see if anything else needs to be changed.  

1. Can you tell me exactly which configuration of IntelliCenter you have?
2. Can you resend me another packet capture after the controller finishes loading (and let me know if anything you see if poolConfig.json or poolState.json are incorrect)?

theRealMCQwerty, 2020-03-15:
Thanks for your fast responses and sticking with this.

> 1. Can you tell me exactly which configuration of IntelliCenter you have?

IntelliCenter i8PS with a valve expansion module (522038Z) installed in expansion slot 1.
Running firmware v 11:14:14

> 2\. Can you resend me another packet capture

[packetLog(2020-03-14_20-59-31).log](https://github.com/tagyoureit/nodejs-poolController/files/4333700/packetLog.2020-03-14_20-59-31.log)

[packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/4333701/packetCapture.zip)

> let me know if anything you see if poolConfig.json or poolState.json are incorrect

From `State`:
```
"equipment": {
		"controllerType": "intellicenter",
		"model": "IntelliCenter i10PS",
		"maxBodies": 2,
		"shared": true,
		"maxCircuits": 11,
		"maxValves": 10,
		"maxSchedules": 100,
		"maxPumps": 16,
		"equipmentType": "equipment"
	},
```
- i8PS incorrectly reported as i10PS.

From `Config`:
```
	"equipment": {
		"maxBodies": 2,
		"maxCircuits": 11,
		"maxSchedules": 100,
		"maxPumps": 16,
		"maxValves": 10,
		"maxCircuitGroups": 16,
		"maxLightGroups": 40,
		"maxIntelliBrites": 8,
		"maxChlorinators": 1,
		"maxCustomNames": 10,
		"model": "IntelliCenter i10PS",
		"maxFeatures": 32,
		"name": "Master Panel",
		"type": 15,
		"expansions": [
			{
				"id": 1,
				"type": 0,
				"name": "Expansion 1",
				"isActive": false
			},
			{
				"id": 2,
				"type": 0,
				"isActive": false,
				"name": "Expansion 2"
			},
			{
				"id": 3,
				"type": 0,
				"isActive": false,
				"name": "Expansion 3"
			}
		],
		"shared": true
	},
```

- i8PS incorrectly reported as i10PS.
- I have a valve expansion module installed in 'expansion' slot 1, but it shows as `"isActive" : false`. However the `"maxValves"` is correctly showing `10` (4 from the main board and 6 extra from the expansion module)

tagyoureit, 2020-03-15:
Oh, very interesting!  If you have the i8PS shouldn't `maxCircuits` only be 8 (or 9) and not 11?

If you have this unit on your bench are you able to start a fresh packet capture (and delete poolConfig/poolState) **without** the 6 valve expansion panel?  We currently have a small sample set and seeing the differences with/without the panel in the i8PS compared to the i10PS will help to nail down the differences in the packets.

theRealMCQwerty, 2020-03-15:
> shouldn't `maxCircuits` only be 8 (or 9) and not 11?

Yes it should, sorry I forgot to mention that.

> fresh packet capture ... **without** the 6 valve expansion panel?

[packetLog(2020-03-15_09-49-26).log](https://github.com/tagyoureit/nodejs-poolController/files/4334367/packetLog.2020-03-15_09-49-26.log)
[packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/4334368/packetCapture.zip)

Even without the valve expansion module, `maxValves` is still reporting as `10` which is incorrect as it can only control 4 without the expansion module.

From `state`:
```
 "equipment": {
    "shared": true,
    "model": "IntelliCenter i10PS",
    "controllerType": "intellicenter",
    "maxBodies": 2,
    "maxCircuits": 11,
    "maxValves": 10,
    "maxSchedules": 100,
    "maxPumps": 16
  },
```

From `config`:
```
"equipment": {
    "maxBodies": 2,
    "maxCircuits": 11,
    "maxSchedules": 100,
    "maxPumps": 16,
    "maxValves": 10,
    "maxCircuitGroups": 16,
    "maxLightGroups": 40,
    "maxIntelliBrites": 8,
    "maxChlorinators": 1,
    "maxCustomNames": 10,
    "maxFeatures": 32,
    "name": "Master Panel",
    "type": 15,
    "expansions": [
      {
        "id": 1,
        "type": 0,
        "name": "Expansion 1",
        "isActive": false
      },
      {
        "id": 2,
        "type": 0,
        "isActive": false,
        "name": "Expansion 2"
      },
      {
        "id": 3,
        "type": 0,
        "isActive": false,
        "name": "Expansion 3"
      }
    ],
    "shared": true,
    "model": "IntelliCenter i10PS"
  },
```
---------------------------
For fun, I repeated the above with the i8PS personality board removed. So now all I have connected is the mainboard. The mainboard has only 4 Aux circuit and 2 valve actuator connections on board.

[packetLog(2020-03-15_10-01-23).log](https://github.com/tagyoureit/nodejs-poolController/files/4334379/packetLog.2020-03-15_10-01-23.log)

[packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/4334380/packetCapture.zip)

I deleted the `poolState.json` and `poolConfig,json` before this run, but the results are exactly the same as above. 😕 

```
 "equipment": {
    "maxBodies": 2,
    "maxCircuits": 11,
    "maxSchedules": 100,
    "maxPumps": 16,
    "maxValves": 10,
    "maxCircuitGroups": 16,
    "maxLightGroups": 40,
    "maxIntelliBrites": 8,
    "maxChlorinators": 1,
    "maxCustomNames": 10,
    "maxFeatures": 32,
    "name": "Master Panel",
    "type": 15,
    "expansions": [
      {
        "id": 1,
        "type": 0,
        "name": "Expansion 1",
        "isActive": false
      },
      {
        "id": 2,
        "type": 0,
        "isActive": false,
        "name": "Expansion 2"
      },
      {
        "id": 3,
        "type": 0,
        "isActive": false,
        "name": "Expansion 3"
      }
    ],
    "shared": true,
    "model": "IntelliCenter i10PS"
  },
```

tagyoureit, 2020-03-15:
Thanks, I'll take a look at these... just curious, after each hardware change are you able to verify that IC knows you have removed the boards?  In the picture below (from the [manual](https://www.pentair.com/content/dam/extranet/product-related/product-manuals/pool-and-spa-equipment/intellicenter/IntelliCenter_Install_Guide.pdf), page 45), you should be able to re-detect and verify what the base board thinks is present:

[image]

theRealMCQwerty, 2020-03-15:
That is the setup wizard, the detected hardware can also be accessed at any time via:
Advanced System Configuration -> System -> Detected Hardware. 

The system correctly detects that the relevant cards are installed/removed. See images below for each of the three configurations I tested:

[image]

[image]

[image]

Gibby, 2020-03-22:
I just got an intellicenter have been digging around seeing how to integrate it with Home Assistant and this seems the closest. I want to be able to do it without having to use a RS485. On https://www.intellicenter.com/ looking at the source looks like the API is all there and uses web sockets. I noticed the Android App for the intellicenter supports local IP mode. When I get home later tonight I'll unpack it and see how it connects locally to the intellicenter. 
Also, I am able to connect to the intellicenter locally on port 6081 and send commands and request parameters, I just haven't figured out how to get status updates when say I changed a setting from Android app.
I am no developer my any means but I am pretty good and digging around and modifying code as needed. Let me know if I can be of any help.

Gibby, 2020-03-23:
I was trying to figure out how to get changes of the system when a telnet session is opened to 6081 and I found if you send the below message, you start getting status messages for temp, lights(on/off, color changes) and a whole lot of other info.

> Send

`{"command":"RequestParamList","objectList":[{"objnam":"UFFFE","keys":["ENABLE","OBJTYP","PASSWRD","SHOMNU"]},{"objnam":"C0002","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0003","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0002","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0003","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0004","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0005","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"_5451","keys":["OBJTYP","MODE","AVAIL","TEMPNC","VACFLO","VACTIM","START","STOP","HEATING","VALVE","TIMZON","VER","MANHT"]},{"objnam":"_A135","keys":["OBJTYP","SUBTYP","PROBE","STATUS"]},{"objnam":"B1101","keys":["TEMP","HITMP","LOTMP","HEATER","STATUS","HTMODE","HTSRC","VOL","MANUAL","LSTTMP","HNAME","MODE"]},{"objnam":"C0002","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0003","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0004","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0005","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"C0006","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY"]},{"objnam":"GRP01","keys":["STATUS","MODE","LISTORD","USAGE","FREEZE","LIMIT","USE","MANUAL","FEATR","DNTSTP","CHILD","HNAME","SNAME","RLY","OBJNAM","OBJTYP","SHOMNU","TIME","TIMOUT","SOURCE","SUBTYP","BODY","CHILD","SYNC","SET","SWIM"]},{"objnam":"_C105","keys":["OBJTYP","SOURCE"]},{"objnam":"_C10C","keys":["OBJTYP","DAY","MIN","OFFSET","SNAME","TIMZON","SRIS","SSET","CLK24A","LOCX","LOCY","ZIP","DLSTIM"]},{"objnam":"_5451","keys":["OBJTYP","MODE","AVAIL","TEMPNC","VACFLO","VACTIM","START","STOP","HEATING","VALVE","TIMZON","VER","MANHT"]},{"objnam":"_57A7","keys":["PARTY","BADGE"]}],"messageID":"1234"}`

> Response 

`{"command":"NotifyList","messageID":"1234","timeSince":"0","timeNow":"1584990960","response":"200","objectList":[{"objnam":"UFFFE","params":{"ENABLE":"OFF","OBJTYP":"PERMIT","PASSWRD":"xxxxxx","SHOMNU":"vogqkzactefluphmnbxiPSC"}},{"objnam":"C0002","params":{"STATUS":"ON","LISTORD":"2","USAGE":"sdyMbaSowglTe","FREEZE":"OFF","LIMIT":"4","USE":"REDR","FEATR":"OFF","DNTSTP":"OFF","HNAME":"AUX 1","SNAME":"Pool Light 1","OBJNAM":"C0002","OBJTYP":"CIRCUIT","SHOMNU":"fcsreplhzmto","TIME":"240","SUBTYP":"INTELLI"}},{"objnam":"C0003","params":{"STATUS":"ON","LISTORD":"3","USAGE":"sdyMbaSowglTe","FREEZE":"OFF","LIMIT":"2","USE":"BLUER","FEATR":"OFF","DNTSTP":"OFF","HNAME":"AUX 2","SNAME":"Pool Light 2","OBJNAM":"C0003","OBJTYP":"CIRCUIT","SHOMNU":"fcsreplhzmto","TIME":"240","SUBTYP":"INTELLI"}},{"objnam":"C0002","params":{"STATUS":"ON","LISTORD":"2","USAGE":"sdyMbaSowglTe","FREEZE":"OFF","LIMIT":"4","USE":"REDR","FEATR":"OFF","DNTSTP":"OFF","HNAME":"AUX 1","SNAME":"Pool Light 1","OBJNAM":"C0002","OBJTYP":"CIRCUIT","SHOMNU":"fcsreplhzmto","TIME":"240","SUBTYP":"INTELLI"}},{"objnam":"C0003","params":{"STATUS":"ON","LISTORD":"3","USAGE":"sdyMbaSowglTe","FREEZE":"OFF","LIMIT":"2","USE":"BLUER","FEATR":"OFF","DNTSTP":"OFF","HNAME":"AUX 2","SNAME":"Pool Light 2","OBJNAM":"C0003","OBJTYP":"CIRCUIT","SHOMNU":"fcsreplhzmto","TIME":"240","SUBTYP":"INTELLI"}},{"objnam":"C0004","params":{"STATUS":"OFF","LISTORD":"4","USAGE":"","FREEZE":"OFF","LIMIT":"0","USE":"WHITER","FEATR":"OFF","DNTSTP":"OFF","HNAME":"AUX 3","SNAME":"AUX 3","OBJNAM":"C0004","OBJTYP":"CIRCUIT","SHOMNU":"fcsrepvhzmto","TIME":"720","SUBTYP":"GENERIC"}},{"objnam":"C0005","params":{"STATUS":"OFF","LISTORD":"5","USAGE":"","FREEZE":"OFF","LIMIT":"0","USE":"WHITER","FEATR":"OFF","DNTSTP":"OFF","HNAME":"AUX 4","SNAME":"AUX 4","OBJNAM":"C0005","OBJTYP":"CIRCUIT","SHOMNU":"fcsrepvhzmto","TIME":"720","SUBTYP":"GENERI [... cut ...]
{"command":"NotifyList","messageID":"c15f731d-72a5-48c8-927d-03aaf649adcb","objectList":[{"objnam":"C0002","params":{"ACT":"65535","USE":"BLUER","LIMIT":"2"}}]}
{"command":"NotifyList","messageID":"dc6111f9-7f3b-4fb9-9e00-1e536f6ac982","objectList":[{"objnam":"B1101","params":{"TEMP":"53"}}]}`

All I did was go to intellicenter.com, open up developer tools in Chrome and watched the console. You can filter on sending and see everything it is doing. I think there are a few other sends it does to build the above command but I think it is a pretty good start.

tagyoureit, 2020-03-23:
Hi @Gibby, that's pretty cool.  Sounds like there could be some benefit to continuing down that path.  I, unfortunately, don't have an IntelliCenter.  There was another user who was figuring out the network calls on ScreenLogic for the *Touch series but I never got into trying to debug them.  Keep us posted!

theRealMCQwerty, 2020-03-24:
@tagyoureit any thoughts on my latest findings and why my system is discovered as an i10PS? Is there any more info I can provide? Hoping to get some more equipment installed soon and power to my equipment pad but that will mean my IntelliCenter will move off my bench and be more difficult to mess with. 
Thanks

tagyoureit, 2020-03-24:
Hmmm, I looked at the packets that are marked as identifying the personality boards and it is the 30,40,0 packet.   There are two packets (17 & 18) that are different from each, but it is a pretty small sample to be able to decode it so far.  I'll try to look at some other examples I have when I get some time.  (I think, in order, they are the naked base board; base + personality; and base + personality + expansion but I need to double check.)

```
[255,0,255,165,1,15,16,30,40,0,0,1,0,0,0,0,0,0,0,0,0,0,240,211,28,86,2,0,0,78,100,104,103,0,3,0,0,15,0,0,0,0,100,0,0,0,0,0,0,5,58]
[255,0,255,165,1,15,16,30,40,0,0,1,0,0,0,0,0,0,0,0,0,0,240,211,28,75,2,0,0,78,100,104,103,0,3,0,0,15,0,0,0,0,100,0,0,0,0,0,0,5,47]
[255,0,255,165,1,15,16,30,40,0,0,1,0,0,0,0,0,0,0,0,0,0,240,210,28,233,4,0,0,78,100,104,103,0,3,0,0,15,0,0,0,0,100,0,0,0,0,0,0,5,206]
```

@rstrouse, how did you determine this packet identified the expansion (and personality) boards?  Do these help in any way?

rstrouse, 2020-03-24:
> Hmmm, I looked at the packets that are marked as identifying the personality boards and it is the 30,40,0 packet. There are two packets (17 & 18) that are different from each, but it is a pretty [...]

Guys,

The 30 message above does not indicate the model or the personality boards.  That 30 you are looking at represents the genera; options associated with the panel.  This includes the sensor adjustments, clock mode, body setpoints, body heat modes, and manual operations.

The model is extracted from the message action 30, xx, 13, 0 message.  This contains the information for the panels.  It may not be correct because of the limited sample but the system looks at the master panel type from the payload byte(35).  I simulated a scenario where the board actually reported an i5, and i10.  From there the max circuits are determined.  If you have the message for 30, 40, 13, 0 then I can see the differences between the two payloads.

Payload byte(35) is a bitmask where 
bit 1 = indicates whether a valve expansion exists
bit 2 = i5P installed
bit 3 = i8P installed
bit 4 = i10P installed
bit 8 = Shared equipment (S).

theRealMCQwerty, 2020-03-25:
> bit 1 = indicates whether a valve expansion exists

Hi @rstrouse  I don't pretend to understand all that you said but I have a question regarding the above. If there is only one bit allocated to the indicating the presence or lack of a valve expansion board then what happens in the case where two or three of the expansion slots are filled with valve expansion boards? Does bit1 just indicate if _any_ expansion boards exist and then there is another part of the payload that indicates the number of available valves?

Also, there is an optional pool cover expansion module (522039Z) where (if at all) would the presence of this board show in the payload?

rstrouse, 2020-03-25:
I'll address the pool cover first since that is the easy one.   The pool cover is actually a dedicated port.  It's existence is determined by asking for the cover configuration.  In fact you can enable it without even having the board.

As for the detection of the personality boards that is another issue altogether.  The reason why you are experiencing issues is that they moved the location of this byte from 1.023 to 1.040.  I just hadn't upgraded yet but I did just about a half hour ago.   Turns out that there is a message 204 that had this information duplicated in it and they removed it from the master panel config.

Anyhow just to clarify a few things.  There are different types of expansion here.  The OCP0 is the outdoor panel.  If you add an i10x or i5x you get OCP1 as well and you can add up to 3 of these in some sort of nebulous configuration.  That being said, each of these can only have 1 valve expansion (I believe).  There are actually 4 bits in the new location that indicate these.

I have your 3 configurations so this should help me to see the differences.

theRealMCQwerty, 2020-03-25:
> If you add an i10x or i5x you get OCP1 as well and you can add up to 3 of these in some sort of nebulous configuration. That being said, each of these can only have 1 valve expansion (I believe).

Very interesting. So just to clarify, are you saying that each master board can only have one valve expansion module per board even though each main board has 3 expansion slots? 
If I understand correctly, the i10x and i5x are essentially additional IntelliCenter mainboards that are slaved to the main-main board?

> In fact you can enable it without even having the board.

This is also very interesting to me :-) Before I start asking a lot of questions about _how_ to enable it, is there and way to signal to the IntelliCenter that a cover is open/closed without having the cover module installed? As far as I can tell that is really the only point of the cover module, to provide a physical interface, since 'reaction' to an open/closed cover occurs on the mainboard.

rstrouse, 2020-03-25:
> Very interesting. So just to clarify, are you saying that each master board can only have one valve expansion module per board even though each main board has 3 expansion slots?
That is correct but I may just be mistaken on this.  I'll have to look at it again but I think you can't piggy back more than one.
> If I understand correctly, the i10x and i5x are essentially additional IntelliCenter mainboards that are slaved to the main-main board?
> 
That is correct
> This is also very interesting to me :-) Before I start asking a lot of questions about _how_ to enable it, is there and way to signal to the IntelliCenter that a cover is open/closed without having [...]
I am pretty sure the only thing the cover board gives you is the electrical connectors.  Everything else exists without the board.

theRealMCQwerty, 2020-03-25:
> That is correct but I may just be mistaken on this. I'll have to look at it again but I think you can't piggy back more than one.

I was under the impression it was valid to fill all three slots with valve expansion modules (in fact my current design relies on it!) 😟 - From poking around in the API I noticed the mainboard seems to support a max of 26 valves, (6+6+6+6+2) ◀️ that is actually _4_ valve expansion modules because the mainboard can be run without the personality card. I happen to have another valve expansion board. I will go and add it to my setup now and test with 2 valve expansion boards plus the i8PS personality card.... 

-------------------------------

> I am pretty sure the only thing the cover board gives you is the electrical connectors. Everything else exists without the board.

👍  - So it may be fun to enable it without the cover board but ultimately pointless? 😄

theRealMCQwerty, 2020-03-25:
Well...... 😞😢😭

Second valve expansion module is correctly detected:
[image]

But only 10 valves are detected in Installation Setup -> Valves and only 10 are show in in IntelliCenter.com

rstrouse, 2020-03-25:
While it is connected like this shoot a log.  I need to see what the 204 message looks like.  All might not be lost here.  Perhaps it simply isn't showing in the UI properly.

theRealMCQwerty, 2020-03-25:
[2020-03-25_08-56-00.zip](https://github.com/tagyoureit/nodejs-poolController/files/4381362/2020-03-25_08-56-00.zip)
[packetLog(2020-03-25_08-56-00).log](https://github.com/tagyoureit/nodejs-poolController/files/4381363/packetLog.2020-03-25_08-56-00.log)

Thanks and 🤞 🤞

rstrouse, 2020-03-25:
Well I think I can get it recognize it but the manual says that there is only one valve expansion per load center.  Page 65
https://www.pentair.com/content/dam/extranet/product-related/product-manuals/pool-and-spa-equipment/intellicenter/IntelliCenter_Install_Guide.pdf

I can say that I have seen conflicting info here.  Not sure how one would get to 26 valves on a single load center.  But then again you have only 11 supported circuits per load center.  Are you sure you want that many valves?  Typically, the plumbing is set up as to make sure you don't dead head a pump so 1 valve in a 3 way position often covers places where you may be thinking 2 valves.  From a guy who has had a 100% failure rate on CV24s that would be catastrophic.  And dangerous, at least to your wallet.  Good news is that I have yet to have an Intellivalve failure.  Btw my pool has 8 Intellivalves.

theRealMCQwerty, 2020-03-25:
> I think I can get it recognize it
If you could would that make the additional valves available to assign to feature circuits? Regarding the 11 circuits per load center, I am seeing 32 available _feature_ circuits and it is feature circuits I have been assigning valves too. Since I have an i8PS I do have 9 (8+1) _aux_ circuits available but as far as I can see these are just for controlling relays. 😕 

> manual says that there is only one valve expansion per load center. Page 65
This is very confusing, from the manual:
```
The Expansion enclosure can be expanded to include up to five actuators (with valve module P/N
520285). Refer to the IntelliCenter Control System Expansion Kit Installation Guide (P/N 520471) for
more information. 
```

Part 520285 is an expansion module that adds 3 actuator ports but it looks like it is for the IntelliTouch not IntelliCenter https://www.amazon.com/Pentair-520285-Replacement-IntelliTouch-Automatic/dp/B008GTWG9A
Also that whole paragraph is lifted directly from page 36 of the IntelliTouch manual: https://www.pentair.com/content/dam/extranet/product-related/product-manuals/pool-and-spa-equipment/easytouch/EasyTouch_and_IntelliTouch_Pool_and_Spa_Control_System_Load_Center_Installation_Guide_English.pdf
I wonder if they just forgot to update that part for the IntelliCenter or do you think that when using an i5x or an i10x then that module is what should be used rather than the 6 port valve expansion module?

With 3 expansion boards each with the 3-port module and a main board with a personality board and a 6-port valve module that would equal a total of (4+6+2+3+2+3+2+3) 25 valves (almost but not quite 26 😄)

> Are you sure you want that many valves?
...Yes 😀 - the pump dead-head issue is a concern but I already have my pad fully plumbed. I have a lot of features and two pumps, I made the whole thing very complicated but it gives me ultimate control! My hope is that if Pentair ever release the mythical firmware update to allow RS485 control of the valves then they will provide feedback as to their current position so I could avoid issues by processing the positions outside of the IntelliCenter before issuing commands to close other valves. 

Worst case I will rig up an arduino and a bank of relays to control the valves, best case we can find a way to make the IntelliCenter recognize more than 10 (without shelling out $$$ on expansion load centers etc.). Middle case, if we get the firmware update and can figure out the RS485 calls then I could control the intellivalves and know their position just without connecting them to the IntelliCenter.

Thanks for all your help so far!

rstrouse, 2020-03-26:
Russ and I spent a couple of hours collaborating last night and he is building configuration screens.  I hope to steal his config code for the dashpanel.  The system will pull all 26 from the configuration and the code for assigning circuits to valves does already exist in the poolController.  If IntelliCenter doesn't reject the setting then we are probably golden.  What it needs is a service layer and UX.

I will be checking in some code over the next couple of days that pulls the OCP modules in.  They strangely removed the data from the panel dump but the secondary location (which we ignored) contains what looks like the correct location to be pulling from.  This will register 16 valves with your two P/N 522440 cards installed.  I don't quite know how they are addressed as they seem to hard-code these on the board as C-H.  Internally we slap an ordinal on them as the position within the configuration so the poolController will deal with that like butter.  We will know the addressing from your poolconfig.json when you run it with the new code and the two modules installed.

It also looks like they shifted one other location but you are too early in the process to notice it.  It relates to lighting groups (which they called out in the release notes).  This looks like we can actually remove some code that we put in to get over an issue they had with synchronizing the lights.  Just need to figure out their transition bytes.

Theoretically, if I knew the protocol I could talk to the valves today but I don't have time to decode a protocol right now.  In the end, these valves will exist on the comm bus and I assume that they will be sending a status periodically.  Hopefully, not too often because we really don't need more chatter on the bus.  Especially, if they follow the chlorinator model.  That thing is an interrupting cow.  In the end it will follow the OCP protocol or the Header/Terminator protocol found with the chlorinator.

Anyhoo, I'll post back here when it is ready.

rstrouse, 2020-04-07:
@theRealMCQwerty,

Alright I made some changes that should recognize the installed cards and their appropriate expansion attributes.  Pull down a new poolController and give it a try.  It should now address all installed personality cards and expansions.   The dashPanel has also been updated to include a yet to be completed configuration section.  Click the gear icon on the upper right corner to see the valves that have been listed from the panel.  It doesn't save anything yet but if the OCP recognizes the two valve expansions on the master panel then these should show in the list.

theRealMCQwerty, 2020-04-08:
thanks @rstrouse !! 

poolConfig.json:

```
"equipment": {
    "maxBodies": 2,
    "maxCircuits": 9,
    "maxSchedules": 100,
    "maxPumps": 16,
    "maxValves": 16,
    "maxCircuitGroups": 16,
    "maxLightGroups": 40,
    "maxChlorinators": 1,
    "maxFeatures": 32,
    "modules": [
      {
        "id": 0,
        "name": "i8PS",
        "desc": "i8PS Personality Card",
        "type": 3,
        "part": "521968Z",
        "bodies": 2,
        "circuits": 9,
        "valves": 4
      },
      {
        "id": 1,
        "name": "Valve Exp",
        "desc": "Valve Expansion Module",
        "type": 8,
        "part": "522440",
        "valves": 6
      },
      {
        "id": 3,
        "name": "Valve Exp",
        "desc": "Valve Expansion Module",
        "type": 8,
        "part": "522440",
        "valves": 6
      }
    ],
    "expansions": [
      {
        "id": 1,
        "modules": [],
        "type": 0,
        "name": "Expansion 1",
        "isActive": false
      },
      {
        "id": 2,
        "modules": [],
        "type": 0,
        "isActive": false,
        "name": "Expansion 2"
      },
      {
        "id": 3,
        "modules": [],
        "type": 0,
        "isActive": false,
        "name": "Expansion 3"
      }
    ],
    "shared": true,
    "model": "IntelliCenter i8PS",
    "softwareVersion": "1.04",
    "name": "Master Panel",
    "type": 15
  },
...
"valves": [
    {
      "id": 1,
      "circuit": 129,
      "type": 0,
      "isActive": true,
      "name": "Valve A"
    },
    {
      "id": 2,
      "circuit": 130,
      "type": 0,
      "isActive": true,
      "name": "VALVEB"
    },
    {
      "id": 3,
      "circuit": 247,
      "type": 0,
      "isActive": true,
      "name": "Intake"
    },
    {
      "id": 4,
      "circuit": 247,
      "type": 0,
      "isActive": true,
      "name": "Return"
    },
    {
      "id": 5,
      "circuit": 131,
      "type": 0,
      "isActive": true,
      "name": "VALVEC"
    },
    {
      "id": 6,
      "circuit": 132,
      "type": 0,
      "isActive": true,
      "name": "VALVED"
    },
    {
      "id": 7,
      "circuit": 133,
      "type": 0,
      "isActive": true,
      "name": "VALVEE"
    },
    {
      "id": 8,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Valve F"
    },
    {
      "id": 9,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Valve G"
    },
    {
      "id": 10,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Valve H"
    },
    {
      "id": 11,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Valve A (XP 1)"
    },
    {
      "id": 12,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Valve B (XP 1)"
    },
    {
      "id": 13,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Intake (XP 1)"
    },
    {
      "id": 14,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Return (XP 1)"
    },
    {
      "id": 15,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Intake Pool 2 ("
    },
    {
      "id": 16,
      "circuit": 256,
      "type": 0,
      "isActive": true,
      "name": "Return Pool 2 ("
    }
  ],
```

Logs:
[packetLog(2020-04-07_17-30-25).log](https://github.com/tagyoureit/nodejs-poolController/files/4447714/packetLog.2020-04-07_17-30-25.log)

[2020-04-07_17-30-25-20200408T012928Z-001.zip](https://github.com/tagyoureit/nodejs-poolController/files/4447716/2020-04-07_17-30-25-20200408T012928Z-001.zip)

Looks like its working!! 😁 
I wasn't able to test dashpanel with it all connected as my IntelliCenter is now out on the equipment pad and it started to rain just as I got everything hooked up. Grabbed the logs and ran inside.

I did check the IntelliCenter screen and it was still only shoing 10 valves (Intake, Return, A, B, C, D, E, F, G and H) but I think that is as expected? Is the idea that I would be able to control these additional 6 actuators from dashpanel and also via the pool-controller API?

This is very exciting, I actually have yet another 6-port valve expansion module (yes I know I'm crazy), do you think that would work also?

Another bonkers question, I know there are only 4 expansion bays (3 when the personality card is installed) but since the cards connect with edge connectors do you think I could connect more than 4 cards? I currently have in my possession, the i8PS personality card 3x valve expansion cards and a pool cover card. I think I could make them fit and I can certainly test with the panel open but do you think it would work?

Thanks again!!!

rstrouse, 2020-04-08:
We aren't out of the woods yet.  While I am pretty confident that I can send the configuration to the OCP I am not sure I can get it to activate the valve when the circuit turns on.  I am hoping that they didn't hardcode anything after Valve H to XP1.  

In short, I can set them and return them but let's see if I can get the circuit to rotate a valve.  This will have to wait a bit as I have work and life in the way of my fun.

As for chaining that many cards together, I wouldn't know.  One issue might be power limitations the other potential issue is how it communicates on the bus.  I assume at least one of those lanes on the connector are interrupt for each connected card.  Who knows maybe it is more sophisticated than that.

theRealMCQwerty, 2020-04-08:
> This will have to wait a bit as I have work and life in the way of my fun.

Thank you very much for all of your help so far!! My pad is currently being wired so I will run all the actuators back to the IC and sit tight until I hear back from you.

Thanks again and please let me know if there is anything I can test or info I can provide to help in the mean time.

kkimball1, 2020-04-17:
Hi,

I'm very interested in the addition of IntelliCenter into this project as I want to integrate control of my pool locally into Home Assistant. Let me know how I can assist and here is my pool setup.

IntelliCenter Load Center w/SWG support
IntelliCenter I10ps Pool/Spa Personality Kit
Pentair IC60 Cell
4x IntelliValve Actuators
Pentair Mastertemp Gas Heater

Do you have an update of what is working and not working?

Thanks for all your efforts on this to date!

tagyoureit, 2020-04-17:
Those should all be working in the `next` branch.  The notables are that IntelliValve is actively being worked on and if the MasterTemp talks over RS485 then that would need to be decoded.  Everything else should just work.  Please do give it a try.

theRealMCQwerty, 2020-04-18:
Hi @tagyoureit - your response is very interesting, a couple of follow up questions:

> IntelliValve is actively being worked on

Could you provide some more detail on this? I know that @rstrouse mentioned that it appeared to be possible to control the intellivalves via rs485 but I think he said he did not have the time to reverse engineer the protocol. I have had one hooked up to my bus but of course I cant do anythign with it. Direct control of intellivalves is of very high interest to me since I have 24 of them!

> if the MasterTemp talks over RS485

I am 99% sure that the Mastertemp does not have the ability to talk over RS485. I have one of these heaters and I have scoured the install manual to see if there is any mention of it. The only 'automation' is via the two wire fireman's switch, which I believe to be a dry contact. Is there something that I'm missing that you have seen regarding RS485 capabilities?

[69 more replies]

closed 2020-07-05 by tagyoureit