tagyoureit/nodejs-poolController #117: Question, besides getting the rs485 cable...?

gbdavidx, 2019-03-12:
No issue with software, but for a newb, how do i get the software to connect to the pool pump? I have a pentair intelliflow variable speed pump, do i connect the rs485 to a raspberry pi or something else?

gbdavidx, 2019-03-12:
would something like this work? https://www.amazon.com/Eletechsup-Distance-Wireless-Transceiver-Passthrough/dp/B07H5HFFNX/ref=sr_1_1?keywords=rs485+serial&qid=1552357291&s=hi&sr=1-1

tagyoureit, 2019-03-12:
I have a recommendation for a RS-485 adapter on the [RS-485 wiki page](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details).  I am running this USB dongle in my Raspberry Pi and others are using Windows based PC's.  I have my Raspberry Pi next to my pool equipment in a waterproof utility box and it connects directly to my home wifi.  While that product looks interesting (and you can try it) I am a bit leery about the "half-duplex" specification.  Pentair makes the [Screenlogic Wireless](https://advancesolar.com/wp-content/uploads/2017/08/WirelessconnectionkitIG.pdf) but I'm honestly not sure if that is full/half duplex.

gbdavidx, 2019-03-12:
Thanks!! I didn’t even see that. Sweet deal. Do I have to do anything special on the pi? What kind of water proof enclosure do you have?

tagyoureit, 2019-03-12:
I have one like [this](https://www.amazon.com/Altelix-Vented-Enclosure-Polycarbonate-Weatherproof/dp/B0791KLM3F/ref=sr_1_fkmrnull_12_sspa?crid=K45BVWFQT85O) (I have more equipment out there) but anything that is big enough for your needs should suffice.

tagyoureit, 2019-03-14:
Closing issue.  Re-open/ask a new one if you still have questions.

closed 2019-03-14 by tagyoureit

gbdavidx, 2019-03-14:
I just bought the device, do i need any drivers or with rasbian pick it up once i plug it in?

tagyoureit, 2019-03-15:
[RS-485 wiki page](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details).  Already documented.