tagyoureit/nodejs-poolController #118: Question regarding implementation

gbdavidx, 2019-03-12:
Dont have this up and running yet, but how difficult would it be to implement mqtt?  Some cool possible sensors could be to turn on/off pool pump, see how long its been running for, what mode is currently set to, and see what the schedule is to too, if you can implement mqtt, we could import this awesome settings into home-assistant.io!

tagyoureit, 2019-03-12:
I would check out the [integrations](https://github.com/tagyoureit/nodejs-poolController#integrations-1) and [examples](https://github.com/tagyoureit/nodejs-poolController/tree/master/src/integrations).  We use SSDP and MDNS to talk to SmartThings.  Shouldn't be too hard...

gbdavidx, 2019-03-12:
Thanks. Might just have to buy a isy device since that syncs with home assistant too

gbdavidx, 2019-03-12:
What type of data can be sent over from the api?

tagyoureit, 2019-03-12:
Load up the pool app and run `http://{server}:3000/all` and it is a single JSON with all the info in the entire app.  It is made up of individual JSON.  It's all documented in the README.

tagyoureit, 2019-03-14:
Closing issue.  Re-open/ask a new one if you still have questions.

closed 2019-03-14 by tagyoureit