tagyoureit/nodejs-poolController #119: Docker command?

gbdavidx, 2019-04-03:
is there support for installing this with docker?

tagyoureit, 2019-04-04:
I don’t have any documentation on this. Others may have set it up this way but I haven’t. 

If someone wants to help write up instructions I’ll be happy to include in the readme or wiki.

dhop90, 2019-06-05:
I may start looking at this in a few days, but here is a tutorial I found https://nodejs.org/de/docs/guides/nodejs-docker-webapp/

wurmr, 2020-05-19:
I've created a quick Dockerfile for my usage that grabs the latest in `next` and builds it:
https://github.com/wurmr/nodejs-poolController-docker

It is hosted on hub here with multiarch builds.
https://hub.docker.com/r/wurmr/nodejs-pool-controller

Perhaps that is useful, or you can use it as an example for your needs.

tagyoureit, 2020-05-19:
Thanks so much!  I'll add these to the docs for sure.

If we wanted to add either [dashpanel](https://github.com/rstrouse/nodejs-poolControl.dashPanel) or [webclient](http://github.com/tagyoureit/nodejs-poolController-webClient) for a front-end interface should that be part of the same container or should they be their own separate containers and talk to the server?

wurmr, 2020-05-19:
> If we wanted to add either dashpanel or webclient for a front-end interface should that be part of the same container or should they be their own separate containers and talk to the server?

They should be separate containers and composed together. I'm actually do this right now with my example graphql wrapper:

```
version: '3.8'
services:
  controller:
    image: wurmr/nodejs-pool-controller
    ports:
      - 4200:4200
    volumes:
      - ./config.json:/usr/local/nodejs-poolController/config.json
  graph:
    image: wurmr/nodejs-pool-controller-graph
    ports:
      - 4000:4000
    environment:
      - POOL_CONTROLLER_ADDRESS=http://controller:4200
      - NODE_ENV=dev
```

tagyoureit, 2020-07-05:
Closing this issue.  Link to docker file was added to `next` Readme already.  Thanks @wurmr

closed 2020-07-05 by tagyoureit