tagyoureit/nodejs-poolController #120: Suntouch controller issues

jstewart99, 2019-04-16:
### Expected behavior and actual behavior
Some data is reported but Pool and Spa state and set point not reported.  Clicks in spa section result in "ERROR: Client, equipment = undefined" or "malformed packet" errors.
### What is actually happening (screenshots are helpful)
[image]
### Steps to reproduce the problem

### Specifications

1. What version of the code are you using?
Downloaded and installed 5.3 and although folder shows 5.3.0, at launch it says it's version 5.2...?

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
  _see below_
3. Any errors that are in the console
4. Your config.json or in-use configuration file
Packet capture and config.json files attached:
[Archive.zip](https://github.com/tagyoureit/nodejs-poolController/files/3082939/Archive.zip)

5. Your pool equipment
Suntouch, VSF pump, intellichlor, solar and gas heat.

tagyoureit, 2019-04-16:
Thanks, I'll take a look.  Can you also do a packet capture where you turn on/off the pool and spa from your working controller (and note the times so I can correspond in the logs)?  Maybe leave 30 seconds between each one.

jstewart99, 2019-04-16:
Sure thing, packet files attached.  Here's a log of what I did from the Suntouch controller (times may be off by a minute or so):
[Archive 2.zip](https://github.com/tagyoureit/nodejs-poolController/files/3086520/Archive.2.zip)

12:03  Spa ON
12:05  Spa heat change from 98 to 95
12:06 Spa heat change from 95 to 98
12:07 Spa OFF
12:08 Pool OFF
12:09 Pool ON
12:10 Pool temp from 85 to 83
12:11 Pool temp from 83 to 85

Existing air temp was showing 70, existing pool temp was showing 73.

If it's helpful, the spa and pool are "circuits" on the SunTouch (you go into the circuits menu and turn each one on or off there)...not sure how other controllers work.

Thanks for the assistance...I truly appreciate it!

jstewart99, 2019-05-09:
Just wanted to bump this issue...I know you've been busy with the new release.  If there's anything else I can do to assist getting this sorted out, please let me know.  I'm sure there's no small percentage of Suntouch users out there who'd appreciate having this work, and again, if we could just turn the Spa on and off remotely, that would be a huge benefit.  Thanks!

tagyoureit, 2019-05-09:
I just posted the 6.0 alpha release in the [forums](https://gitter.im/nodejs-poolController/Lobby).  So I'm getting closer to be ready to add new functionality.

tagyoureit, 2019-06-03:
Hi,
I got a chance to take a look at this finally.  Thanks for your patience.  The good news is that the app already understands all of the packets that are being sent.  We just don't have all of the configuration information yet.  The error happens when you try to change the spa we just don't know what the spa is yet.

I think if you tell the system you have an Intellitouch we will get all of the right configuration information (we request it upon app startup).

Looking at the Suntouch manuals it is a bit hard to tell exactly how many circuits there are.   By my count, there is pool/spa/aux 1-3/feature 1-4.  So possibly 9?  This should be  `numberOfCircuits` in your `config.json`.

It also looks like a max of 6 schedules.  `numberOfSchedules` should be 6.  

I don't see anything about custom names.

Can you try these settings in your `config.json` and see what happens?

```
"intellitouch": {
                "installed": 1,
                "friendlyName": "",
                "numberOfCircuits": 9,
                "numberOfPumps": 1,
                "numberOfCustomNames": 0,
                "numberOfSchedules": 6
            }
```

Hopefully, this will get you further.  What we can also do is have you turn off your pool (physically turn off power to the Suntouch), then start this app (with packet capture enabled), and then turn on the Suntouch.  The Suntouch should broadcast all of its configuration information and we can capture that in the logs.

lmk how it goes.

jstewart99, 2019-06-11:
Thanks for the suggestions!  I tried the config settings (your proposed modifications seem correct based on my limited knowledge of the SunTouch as well), but it's still not quite working.  The bootstrap web page slowly populates info (somewhat randomly...the first two times I ran it there was no pump info, but then the third time it showed the pump info (I think it may have switched to a different circuit because I'd lowered the pool temp, so perhaps that's why).  It seems like the app is crashing...after a bit, the "last update" counter just keeps climbing and nothing updates (nothing really ever updates after the initial populated data).  If I try to change the pool temperature through the webpage, it seems to log the request, but it never updates to the new temp (although it seems like the request may have actually gone through at least once, as I noted the pool temp went from 86 to 85 on the actual controller).  

Per your suggestion I turned power off to the controller and started the app and then powered on the controller.  The attached files should have that info (at about 1:49pm in the logs is when I turned on...the earlier stuff is from me trying a few things with the new config settings).  Interestingly, turning it on from scratch gave me the most output in the bootstrap web page (actually displaying some schedules this time), and trying to change the pool temp did change it (once) on the controller, but then I think it crashed again as subsequent attempts to lower the temp did nothing (and the temp change still never gets updated on the bootstrap web page (ie it still shows 85 even though the controller now shows 84 after I lowered the temp on the web page).

Capture files attached.

[Pool.zip](https://github.com/tagyoureit/nodejs-poolController/files/3278516/Pool.zip)

tagyoureit, 2020-05-13:
Hi,
I've done a lot of work on v6 (the `next` branch) and it is just about ready for release.  I ran your packet captures through the app and nothing breaks.  However, it also isn't seeing the circuits\features because we are missing the config data.  Pumps/chlorinator work fine.

If you want to test out the next branch I'd be curious to see if we are able to pull the configuration from the controller.
thx,
Tag

jstewart99, 2020-05-14:
Thanks for the update.  I’m happy to help test things out.  My only hesitation is that, while things aren’t working 100% correctly by any means, I’ve got it set up well enough to accommodate the few things I wanted to do by just sending the appropriate URL command through my home automation system, such as turn the Spa or an Aux circuit (pool lights) on and off, and set the pool temperature to trigger the solar heating.  Using the web interface to do things didn't seem to work reliably but if I just send something like http://192.168.0.131:3000/circuit/1/set/1 it works fine.  So it’s one of those “if it ain’t broke” situations.  I'm slightly reluctant to change anything since it's all basically meeting my needs and I'm no Raspberry Pi expert (if there's an easy way to backup my current setup, try anything new you'd like me to try, and then restore my old setup if need be, let me know).  I've attached a screenshot of what I'm getting now on the web interface and it has some notations on it with what's working and not working.  It's pretty close and I think whatever tweaks we made a while ago in the config file seem to have improved it (before it wasn't loading data reliably but now it all loads correctly, although it doesn't refresh automatically). I don't know if there's other SunTouch folks out there so it's up to you if you want to move this forward, and I'm happy to help (with a little guidance) if it's easy enough.
[image]

tagyoureit, 2020-05-16:
It's easy to setup a different directory.
`git clone -b next https://github.com/tagyoureit/nodejs-poolController.git nodejs-poolController-next`
This will clone the `next` branch into `nodejs-poolController-next` directory.  Then just go into that directory and run `npm start`.  You won't be able to have both apps utilizing the same RS485 port at the same time because they will both try to lock the port so you should only have one app running at a time.  

The webclient will also install in it's own directory. `git clone https://github.com/tagyoureit/nodejs-poolController-webClient.git`. Run `npm start` in this directory also.  

See what comes up here and we can then start to fix some of the items that are wrong.  I imagine it will likely have the same errors but we'll see.

tagyoureit, 2020-05-26:
I've updated the [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) instructions for 6.0.  Can you please pull both the latest -webclient and main app, the run `npm i` on each, and send me the capture?  I'll get this straightened out.

tagyoureit, 2021-01-22:
Closing due to lack of activity.

closed 2021-01-22 by tagyoureit