tagyoureit/nodejs-poolController #122: SWG delayed status readings / pool controller not detecting status changes?

johnny2678, 2019-04-24:
### Expected behavior and actual behavior

Low Flow warnings aren't shown as cleared even though the SWG (Pentair IC40) LEDs say flow is fine

### What is actually happening (screenshots are helpful)

PoolController still reads Low Flow even after SWG LEDs indicate flow has been restored

### Steps to reproduce the problem

1. Run Pump at low RPMs forcing a Low Flow state
2. Increase RPMs - SWG status will return to green but poolController still reads low flow

### Specifications

1. What version of the code are you using?

5.3

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))

[swg.zip](https://github.com/tagyoureit/nodejs-poolController/files/3112858/swg.zip)

3. Any errors that are in the console

no errors, but you can observe the occurance at the following time stamps

```
09:25:03.928 DEBUG Msg# 14602   Chlorinator status packet: 165,56,15,16,25,22,41,21,128,79,129,0,73,110,116,101,108,108,105,99,104,108,111,114,45,45,52,48,8,96
09:25:03.928 DEBUG Msg# 14602   Chlorinator status has not changed. 
09:26:03.508 DEBUG Msg# 14687   Chlorinator status packet: 165,56,15,16,25,22,41,21,129,79,129,0,73,110,116,101,108,108,105,99,104,108,111,114,45,45,52,48,8,97
09:26:03.509 DEBUG Msg# 14687   Chlorinator status has not changed. 
09:27:03.716 DEBUG Msg# 14764   Chlorinator status packet: 165,56,15,16,25,22,41,21,129,79,129,0,73,110,116,101,108,108,105,99,104,108,111,114,45,45,52,48,8,97
09:27:03.717 DEBUG Msg# 14764   Chlorinator status has not changed. 

****Changed RPMs to 2760, status goes from 129 -> 128. poolController reports Chlorinator status has not changed???

09:28:03.345 DEBUG Msg# 14957   Chlorinator status packet: 165,56,15,16,25,22,41,21,129,79,128,0,73,110,116,101,108,108,105,99,104,108,111,114,45,45,52,48,8,96
09:28:03.346 DEBUG Msg# 14957   Chlorinator status has not changed. 
09:29:03.349 DEBUG Msg# 15042   Chlorinator status packet: 165,56,15,16,25,22,41,21,129,79,128,0,73,110,116,101,108,108,105,99,104,108,111,114,45,45,52,48,8,96
09:29:03.350 DEBUG Msg# 15042   Chlorinator status has not changed. 
09:30:03.472 DEBUG Msg# 15125   Chlorinator status packet: 165,56,15,16,25,22,41,21,129,79,128,0,73,110,116,101,108,108,105,99,104,108,111,114,45,45,52,48,8,96
09:30:03.473 DEBUG Msg# 15125   Chlorinator status has not changed. 
```

4. Your config.json or in-use configuration file

*see files above

5. Your pool equipment
Pentair ScreenLogic / EasyTouch
Pentair VS Pump
Pentair IC40 SWG

tagyoureit, 2019-05-17:
I think I have this fixed in the [6.0-alpha](https://github.com/tagyoureit/nodejs-poolController/tree/6.0-DEV).  Can you give it a try?

tagyoureit, 2019-05-17:
Actually, doing some more testing I'm even seeing Screenlogic showing a delay.  Intellitouch only sends the chlorinator packet out every minute.  Screenlogic and app status correctly updated flow status after ~2 mins.  Not sure we can get this any lower, but test it out.

tagyoureit, 2020-05-13:
v6 does a much better job of monitoring chlorinator comms.

closed 2020-05-13 by tagyoureit