tagyoureit/nodejs-poolController #123: Create API to allow Pump RPM change

johnny2678, 2019-04-24:
***Enhancement Request, not a bug**

Currently, to change the Pump RPMs through poolController, you have to define circuits in your automation and then use poolController to turn on/off those circuits.

Would be awesome if you could set the circuit RPMs using poolController.

I only need API, but someday doing this through the UI would be cool too.

1. What version of the code are you using?

5.3

5. Your pool equipment

Pentair EasyTouch / Pentair VS Pump / Pentair SWG

bsileo, 2019-04-26:
Would these APIs provide the needed functionality @johnny2678 @MaxVonEvil ?

/pumpCommand/run/pump/{pump}/rpm/{rpm}
/pumpCommand/run/pump/{pump}/gpm/{gpm}
/pumpCommand/run/pump/{pump}/rpm/{rpm}/duration/{duration}

bsileo, 2019-04-26:
One question on this @tagyoureit - What is the possible range for Pump RPMs? I assume this might depend on the equipment and I am not sure there is a way to get it from the config. This is a challenge for a slider control in ST since the UI is static there - basically I have to predefine the "range" for the slider upfront.

tagyoureit, 2019-04-26:
Typically, 450-3450RPM is the range for the pumps.  It's usually in the manuals somewhere but that's the range I check for in the app.

Those `/pumpCommand/` API's are for the standalone pump.  The Intellitouch RPM's need to be set through the Configuration settings so they do need to be added.

bsileo, 2019-04-26:
Ok I have put together the basic structure for this...once APIs are in place it will be easy to hook them in.  Side benefit...adding the new Pentair Pump Controller child device now allows us to have a Power meter device in Smart Things so you get access to the watts reported by the pump!

MaxVonEvil, 2019-04-26:
Hi all, thanks for responding to this idea. 

@bsileo - I was playing around with @tagyoureit 's API's, manually firing commands off in a browser. They ought to suffice for this proposed change, However, there's a few things worth highlighting, if one (like me) does not have any pentair control equipment except the pump itself:

1) All pumps aren't created equally. For example when I moved in, the pool had an old, noisy and  powerhungry 1kW single-speed pump. It had no brains, i.e. no RS485 port, hence only way to control it, would be a simple on/off switch like the Aeotec Heavy Duty Switch (which btw has a power-meter and kWh counter!) Perhaps worth it to support such a switch as a pump? There's still many of them old pumps around.

2) As I learned, even VariSpeed pumps are different. I promptly replaced mentioned singlespeed clunker with a Pentair Intelliflo VS model 011018. Contrary to other pumps (VariFlo?) it seems only to deal in RPM's. It has no concept of GPM's.

3) About the min/max pump RPM speeds you asked about, yes the defaults are 450-3450 RPM as Tag pointed out. However they are (in my case) configurable on the pump's own panel.  Yet not owning any, I have no idea if Pentair's control hardware is able to override those min/max values.

Unless Brad beats me to the punch :) (likely, as my coding time is minimal these next few weeks) my own nefarious plans is to repurpose some DHT code used to control the API's on my security cameras, mash it up with some standard dimmer code, creating a stand-alone DHT for the forementioned dimmer-type controlling the pump, by talking to Tag's pool-controller API's.  
The UX would re-use the standard dimmer's on/off and slider controls. I would however create the min/max rpm's as configurable options with defaults as above, setting the custom range for the slider. One could later add more bells, converting RPM's to percentages, or to opt to use GPM's instead of RPM's. 

@bsileo - was the above something along the lines of what you had in mind for the DTH?

Thanks,
Max

bsileo, 2019-04-26:
I had some time this morning so there is a first pass at this device type in my repo [here](https://github.com/bsileo/SmartThings_Pentair/blob/master/devicetypes/bsileo/pentair-pump-control.src/pentair-pump-control.groovy).

This is built to run integrated with the overall PoolController family of devices (e.g. it is setup to be a child of the main Pentair Pool Controller device) so this is not quite the same as what you described.

So far, I have it setup with a slider for RPMs, and on/off toggle, and a report of the wattage. Other data values form the pump are available just not on the UI at this point. 

There are still a few gaps - most notably, the on/off is hardcoded to Circuit 6 so will only work for a single Pump on the Pool side right now. @tagyoureit is going to add an API in the future for this.  Also, the API call to set pump speed is disabled right now  (though it is in [the code](https://github.com/bsileo/SmartThings_Pentair/blob/master/devicetypes/bsileo/pentair-pool-controller.src/pentair-pool-controller.groovy#L730)) since it does not actually work for non standalone use cases.

MaxVonEvil, 2019-04-26:
Great, I'll see if I can get time to try it this weekend. Two quick ones, if you don't mind:

1. Can the pool-controller device actually be installed without child apps for stuff one doesn't  have, such as chlorinators, lights, etc? Instructions says to install them all. Just wanted to minimize the footprint if possible.

2. This business of circuit 6... how does that translate to the lone VS pump I have here? I know I can change it's address on the RS485 bus, yet something tells me said circuit is something else - if memory serves me right it's something defined in the Easytouch systems, right? I guess what I'm asking is how, if at all does this affect my setup?

Thanks,
Max

bsileo, 2019-04-26:
> 1. Can the pool-controller device actually be installed without child apps for stuff one doesn't  have, such as chlorinators, lights, etc? Instructions says to install them all. Just wanted to minimize the footprint if possible. Canthe childDeviceTitle() references just be knocked out?

It will auto create child apps for everything it thinks exists on the system. Some stuff it can not tell and it just creates it. After that, you can go in and delete any child devices you do not want to keep. This is done via the Things via in the Smart App or the devices list in the IDE.

> 2. This business of circuit 6... how does that translate to the lone VS pump I have here? I know I can change it's address on the RS485 bus, yet something tells me said circuit is something else - [...]

I am not sure about this but I suspect it will not match up for you as Circuit 6 is specific to Easy/Intellitouch.

MaxVonEvil, 2019-04-27:
Okay, looks like a separate DHT may be needed to fill the gap between those installations which have all the Pentair gear already and those of us who are trying to avoid buying it to begin with ;)

I'm going to have a stab at this when I get some time, but if someone makes something in the interim, I'm all ears.

rsissons, 2019-05-09:
@bsileo and @tagyoureit   - I am not getting any values for my pump (VS)

Parameter	Pump 1
Watts	wattsnotset
RPM	rpmnotset
Error	errnotset
Drive State	drivestatenotset
Run Mode	runnotset
 Am I missing something?

Thx

MaxVonEvil, 2019-05-09:
Initial thoughts: Are you getting any console messages of significance on
your RaspberryPi?

tagyoureit, 2019-05-09:
What do you get if you navigate to `http://your_pool_server:3000/all`?  You can also attach the [replay logs](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) here and I could have a quick look.

rsissons, 2019-05-15:
> What do you get if you navigate to `http://your_pool_server:3000/all`? You can also attach the [replay logs](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) here and I could have a quick look.

Here's what I get with /all

{"config":{"systemReady":1,"equipment":{"controller":{"intellicom":{"installed":0,"friendlyName":""},"intellitouch":{"installed":1,"friendlyName":"","numberOfCircuits":8,"numberOfPumps":2,"numberOfCustomNames":10,"numberOfSchedules":12},"virtual":{"pumpController":"never","chlorinatorController":"never"},"id":{"productName":"","productNumber":"","manufacturer":"","description":""}},"chlorinator":{"installed":0,"desiredOutput":-1,"friendlyName":"","id":{"productName":"","productNumber":"","manufacturer":"","description":""}},"pump":{"1":{"type":"VS","externalProgram":{"1":-1,"2":-1,"3":-1,"4":-1},"friendlyName":"","id":{"productName":"","productNumber":"","manufacturer":"","description":""}},"2":{"type":"None","externalProgram":{"1":-1,"2":-1,"3":-1,"4":-1},"friendlyName":"","id":{"productName":"","productNumber":"","manufacturer":"","description":""}}},"circuit":{"friendlyName":{"1":"","2":"","3":"","4":"","5":"","6":"","7":"","8":"","9":"","10":"","11":"","12":"","13":"","14":"","15":"","16":"","17":"","18":"","19":"","20":""},"hideAux":false,"nonLightCircuit":{"1":{"circuitName":"SPA"},"4":{"circuitName":"JETS"},"5":{"circuitName":"SPILLWAY"},"6":{"circuitName":"POOL"},"7":{"circuitName":"CLEANER"},"8":{"circuitName":"AUX 7"}},"lightCircuit":{"2":{"circuitName":"POOL LIGHT"},"3":{"circuitName":"SPA LIGHT"}}},"intellichem":{"installed":0},"spa":{"installed":1},"solar":{"installed":0}}},"circuit":{"1":{"number":1,"numberStr":"circuit1","name":"SPA","circuitFunction":"Spa","status":0,"freeze":0,"macro":0,"delay":0,"friendlyName":"SPA"},"2":{"number":2,"numberStr":"circuit2","name":"POOL LIGHT","circuitFunction":"SAM Light","status":0,"freeze":0,"macro":0,"delay":0,"friendlyName":"POOL LIGHT"},"3":{"number":3,"numberStr":"circuit3","name":"SPA LIGHT","circuitFunction":"SAL Light","status":0,"freeze":0,"macro":0,"delay":0,"friendlyName":"SPA LIGHT"},"4":{"number":4,"numberStr":"circuit4","name":"JETS","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"friendlyName":"JETS"},"5":{"number":5,"numberStr":"circuit5","name":"SPILLWAY","circuitFunction":"Spillway","status":0,"freeze":0,"macro":0,"delay":0,"friendlyName":"SPILLWAY"},"6":{"number":6,"numberStr":"circuit6","name":"POOL","circuitFunction":"Pool","status":1,"freeze":0,"macro":0,"delay":0,"friendlyName":"POOL"},"7":{"number":7,"numberStr":"circuit7","name":"CLEANER","circuitFunction":"Generic","status":1,"freeze":0,"macro":0,"delay":0,"friendlyName":"CLEANER"},"8":{"number":8,"numberStr":"circuit8","name":"AUX 7","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"friendlyName":"AUX 7"}},"pump":{"1":{"pump":1,"name":"Pump 1","type":"VS","time":"timenotset","run":"runnotset","mode":"modenotset","drivestate":"drivestatenotset","watts":"wattsnotset","rpm":"rpmnotset","gpm":"gpmnotset","ppc":"ppcnotset","err":"errnotset","timer":"timernotset","duration":"durationnotset","currentrunning":{"mode":"off","value":0,"remainingduration":-1},"externalProgram":{"1":-1,"2":-1,"3":-1,"4":-1},"remotecontrol":1,"power":1,"friendlyName":"Pump 1","virtualController":"disabled","currentprogram":4}},"schedule":{"1":{"ID":1,"CIRCUIT":"POOL","friendlyName":"POOL","CIRCUITNUM":6,"BYTES":[165,1,15,16,17,7,1,6,25,0,7,15,0,1,19],"MODE":"Egg Timer","DURATION":"7:15"},"2":{"ID":2,"CIRCUIT":"CLEANER","friendlyName":"CLEANER","CIRCUITNUM":7,"BYTES":[165,1,15,16,17,7,2,7,8,5,15,55,255,2,56],"MODE":"Schedule","DURATION":"n/a","START_TIME":"8:05","END_TIME":"15:55","DAYS":"Sunday Monday Tuesday Wednesday Thursday Friday Saturday "},"3":{"ID":3,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,3,0,0,0,0,0,0,0,224],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"4":{"ID":4,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,4,0,0,0,0,0,0,0,225],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"5":{"ID":5,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,5,0,0,0,0,0,0,0,226],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"6":{"ID":6,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,6,0,0,0,0,0,0,0,227],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"7":{"ID":7,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,7,0,0,0,0,0,0,0,228],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"8":{"ID":8,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,8,0,0,0,0,0,0,0,229],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"9":{"ID":9,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,9,0,0,0,0,0,0,0,230],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00:00","DAYS":"None"},"10":{"ID":10,"CIRCUIT":"NOT USED","friendlyName":"NOT USED","CIRCUITNUM":0,"BYTES":[165,1,15,16,17,7,10,0,0,0,0,0,0,0,231],"MODE":"Schedule","DURATION":"n/a","START_TIME":"0:00","END_TIME":"00: [... cut ...] ° Farenheit"},"valve":{},"chlorinator":{"installed":0,"saltPPM":-1,"currentOutput":75,"superChlorinate":0,"version":-1,"name":"","controlledBy":"none"},"intellichem":{"readings":{"PH":-1,"ORP":-1,"WATERFLOW":-1,"SI":-1,"SALT":-1},"settings":{"PH":-1,"ORP":-1,"CYA":-1,"CALCIUMHARDNESS":-1,"TOTALALKALINITY":-1},"tankLevels":{"1":-1,"2":-1},"mode":{"1":-1,"2":-1},"lastPacket":[]}}

Here's a link to the replay files. 
https://drive.google.com/file/d/1-l7fXMXrc3M9a-MFW40VN8p3yttnnRY1/view?usp=sharing

rsissons, 2019-05-15:
I am not sure if this has anything to do with it or not?

 ERROR Aborting controller packet Get Pump Config (Extended).  Tried 10 times to write 255,0,255,165,1,16,33,219,1,1,1,180

tagyoureit, 2020-05-13:
v6 is ready for testing.
[Chlorinator directions](https://github.com/tagyoureit/nodejs-poolController/wiki/Virtual-Chlorinator-Controller-v6)
[Pump directions](https://github.com/tagyoureit/nodejs-poolController/wiki/Virtual-Pump-Controller---v6)

There is native support for controlling pump RPM.  API's are in place and will be documented shortly.

closed 2020-05-13 by tagyoureit