tagyoureit/nodejs-poolController #124: IntelliPH - help decoding packets

johnny2678, 2019-04-24:
### Expected behavior and actual behavior

Would like to start the process of decoding packets of a Pentair IntelliPH acid dosing system that is installed alongside a Pentair SWG.

While the packet capture was running, I up the does by 2% (hit the up button 2x) and then hit enter to set the new dosing % (68%)

Packet Capture Files attached - can provide anything else useful here or over gitter.

### What is actually happening (screenshots are helpful) - NA

### Steps to reproduce the problem - NA

### Specifications

1. What version of the code are you using? - 5.3
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))

[IntelliPH.zip](https://github.com/tagyoureit/nodejs-poolController/files/3114074/IntelliPH.zip)

3. Any errors that are in the console - n/a
4. Your config.json or in-use configuration file - provided above
5. Your pool equipment

Pentair VS Pump / Pentair IntelliPH / Pentair SWG

johnny2678, 2019-04-24:
adding crude video recording I did...

[image]

tagyoureit, 2019-10-24:
Where did we leave this?  I think I looked at it but there were no packets from IntelliPH, is that correct?  I tried to download the zip file again but Github seems to have "lost" it.

johnny2678, 2019-10-24:
[IntelliPH (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/3767825/IntelliPH.1.zip)

attaching the file again.

if there are packets present, a better mind than mine will have to find them ;)

my plan now is to hijack the 24v DC power to the stenner pump on the IntelliPH, put it on a Solid State Relay, and run it when i detect the PH getting too high.

gw8674, 2020-05-03:
Any development on this front? Would be great to see if there was a way to listen in. If using an IntelliChlor with the IntellipH, then it (ipH controller) must be on the bus? Manual dosing or changing dosing levels remotely would be big breakthrough.

tagyoureit, 2020-05-06:
The IntelliPH only listens on the bus... it doesn't send any packets here.  Nothing can be done on this front, unfortunately.

closed 2020-05-06 by tagyoureit

johnny2678, 2020-05-06:
yea, would love to be proven wrong but I listened for packets best I could and couldn't find any.