tagyoureit/nodejs-poolController #125: [Enhancement idea] Include Circuit ID in response to /circuit/ID/set

bsileo, 2019-04-26:
It would be helpful to include the details on which circuit was changed in the API response to the /circuit set endpoints.  This is so that async callbacks can take action on this - in my case the ST endpoint. For example:

http://poolcontrol:30000/circuit/5/set/1

returns:
{
"text": "User request to set PATIO LTS to on",
"status": "on",
"value": 1
}

Add an additional key for the curciuit number:
{
"number": 5,
"text": "User request to set PATIO LTS to on",
"status": "on",
"value": 1
}

bsileo, 2019-04-28:
I worked on this a bit more - for consideration @tagyoureit , what about a response format like this:

```
{
"5": {
   "text": "User request to set PATIO LTS to off",
   "statusString": "off",
   "status": 0,
   "value": 0,
   "number": 5
   }
}
```

I got to this by looking at ways I can use the same code to parse a single circuit response as I use to parse the full response from the /circuits endpoint. May not be suitable for all use cases though???

tagyoureit, 2019-05-02:
My API responses are all over the place right now-no consistency at all. We should open up a whole other thread to discuss how they could be better/more useful.

closed 2020-05-13 by tagyoureit