tagyoureit/nodejs-poolController #126: Update API notes for Intellibrite calls

rerouted, 2019-04-26:
Just a issue reminder to update the README.md with the Intellibrite API calls.

* setLightMode(mode)
* setLightColor(circuit, color)
* setLightSwimDelay(circuit, delay)
* setLightPosition(circuit, position)

tagyoureit, 2020-07-05:
LightGroup data structures are documented [here](https://github.com/tagyoureit/nodejs-poolController/wiki/Version-6-Data-Structures#7-light-group).  Intellibrite is now always LightGroup[0] (with id:192) on *Touch systems.  

All available options can be retrieved from `http://localhost:4200/config/options/lightGroups`.
LightGroup variables can be updated with a PUT to `http://localhost:4200/config/lightGroup`.

We'll continue to work on the documentation but HTH.

tagyoureit, 2020-08-11:
Updated in new API documentation: https://tagyoureit.github.io/nodejs-poolcontroller-api/

closed 2020-08-11 by tagyoureit