tagyoureit/nodejs-poolController #129: setLightMode causes fatal error after systemReady, but just after, in 6.0-DEV

emes, 2019-05-20:
### Expected behavior and actual behavior

non-fatal error or be ignored

call to setLightMode(1) with two IntelliBrites configured results in the following error.

### What is actually happening (screenshots are helpful)

```
03:03:13.000 verbose: Msg# 79   Main --> Pump 1: Remote control - disable pump control panel: [165,0,96,16,4,1,255,2,25]
03:03:13.000 verbose: Msg# 80   Pump 1: Remote control -  disable pump control panel: [165,0,16,96,4,1,255,2,25]
03:03:13.000 verbose: Msg# 81   Main --> Pump 1: Pump power to on: [165,0,96,16,6,1,10,1,38]
03:03:13.000 debug: Msg# 85  Light all on/off and position packet is: 1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0

/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/equipment/circuit.ts:935
                    tempLightGroupPacketArr = lightGroupPackets[ 0 ].slice( 1 ).concat( lightGroupPackets[ 1 ].slice( 1 ) )
                                                                     ^
TypeError: Cannot read property 'slice' of undefined
    at Object.assignControllerLightGroup (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/equipment/circuit.ts:935:70)
    at Object.process (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/comms/inbound/controller/39.ts:31:13)
    at Object.processControllerPacket (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/comms/inbound/process-controller.ts:138:35)
    at decode (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/comms/inbound/decode-helper.ts:307:45)
    at Object.processChecksum (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/comms/inbound/decode-helper.ts:363:13)
    at iterateOverArrayOfArrays (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/comms/inbound/receive-buffer.ts:144:38)
    at Timeout.checkIterate [as _onTimeout] (/home/ms/docker/poolcontroller/dev/nodejs-poolController/src/lib/comms/inbound/receive-buffer.ts:257:17)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
nodejs-poolController has closed successfully.
```

### Steps to reproduce the problem

call setLightMode shortly after systemReady: 1

tagyoureit, 2019-05-22:
This is some of the hardest code in the whole app.  If you wait a few minutes and then try do you get the same error?  Hoping that it is just because we did not receive the light packets yet.  If not, I'll need to dig into this quite a bit further.

tagyoureit, 2019-05-23:
I just put in a check to make sure that we aren't processing the 2nd light packet file unless we have the 1st.  I also put a try/catch block around this so at least the app won't crash.  Let me know if you still see this issue.

closed 2019-05-23 by tagyoureit

emes, 2019-05-26:
Hey.  Getting an error on startup with this commit.

```
14:52:37.000 error: There was an error assigning light controller packets. Cannot read property '0' of undefined

15:10:15.000 warn: Set valve status skipped because circuit is not available yet.
15:10:15.000 info: 
  Custom Circuit Names retrieved from configuration: 
        Levolor
        Medium
        High
        Max
        Cl Boost
        Cl Daily
        TEST
        East Light
        West Light
        Bleach Pump
        Floor Jets
        USERNAME-12
        USERNAME-13
        USERNAME-14
        USERNAME-15
        USERNAME-16
        USERNAME-17
        USERNAME-18
        USERNAME-19
        USERNAME-20
15:10:15.000 verbose: Msg# 39  Light details added for circuit EAST LIGHT (3): 1
15:10:15.000 warn: Tried to retrieve circuit 4 which is not a valid circuit.
15:10:15.000 verbose: Msg# 39  Light details added for circuit No valid circuit (4) (4): 2
15:10:15.000 warn: Tried to retrieve circuit 4 which is not a valid circuit.
15:10:15.000 info: Msg# 39:  Intellibrite All On/Off and Light positions discovered: 
EAST LIGHT (3): Position 1 
No valid circuit (4) (4): Position 2 

15:10:15.000 warn: The Pool circuit cannot be renamed at this time.  Skipping.
15:10:15.000 info: 
  Circuit Array Discovered from configuration: 
Circuit 1: NOT USED Function: Spa Status: 0 Freeze Protection: off Macro: 0
Circuit 2: Levolor Function: Generic Status: 0 Freeze Protection: off Macro: 0
Circuit 3: East Light Function: Intellibrite Status: 0 Freeze Protection: off Macro: 0
Circuit 4: West Light Function: Intellibrite Status: 0 Freeze Protection: off Macro: 0
```

Let me know if you need me to do anything.

tagyoureit, 2019-05-27:
Can you send me a packet capture and I’ll get it fixed?

emes, 2019-06-03:
Yes will do. Have since noticed that it’s probably just timing, both circuits are discovered/configured correctly soon thereafter.

emes, 2019-06-03:
After this I added code to my node app to prevent it from connecting before the system was very systemReady :-) 

This just shows simple error, which seems to really be nothing.

I will go back and see if I can trigger the reported one exactly again.

https://www.dropbox.com/sh/fz9mpzxg2csi4zx/AACbxvLe8lhpwqg_5EX8KYH8a?dl=0