tagyoureit/nodejs-poolController #133: UOM issues

emes, 2019-06-03:
### Expected behavior and actual behavior

UOM should be 0 for Fahrenheit and 4 for Celsius, but it is always 0.  This is not a problem since most are probably not using Celsius and UOM isn't used much in the code.  

The Intellichem calculateTemperatureFactor function does use UOM, but here it defaults to Celsius and the conditional never matches Fahrenheit.

### What is actually happening (screenshots are helpful)

On startup:

```
UOM: {
    "UOM": 0,
    "UOMStr": "unknown"
}
```

Equipment message received with UOM properly decoded, in this case Celsius.

[image]

setUOM is called, but always sets UOM to 0.

```
UOM: {
    "UOM": 0,
    "UOMStr": "° Farenheit"
}
```

Other issue: 

Intellichem calculateTemperatureFactor() checks for Farenheit but UOMStr has "° " prepended so temperature factor is always based on Celsius.  I don't have an Intellichem, so did not test this, but I guess this is a more meaningful issue.

### Specifications

1. What version of the code are you using?

I did this on 5.3, but it's basically the same in 6.0-DEV branch.  I'll put it in a PR.  I got rid of the degree symbol because it seemed more useful without.  But that's just me.

tagyoureit, 2019-06-04:
This is done... I updated the results object so UOM has more attributes now.

closed 2019-06-04 by tagyoureit

emes, 2019-06-04:
Looks good.