tagyoureit/nodejs-poolController #137: Stand-alone Aqua Rite SWG responses not being received correctly.

raz0rf0x, 2019-06-30:
### Expected behavior and actual behavior
Status and chlorinator commands being sent and reflected by UI.
In fact, sends commands, SWG responds and changes modes but responses not being correctly interpreted by poolController
### What is actually happening (screenshots are helpful)
UI doesn't update.

If I use the packet sender debug console it does go through and it does update the UI. But on its own it does not.

Command is sent, SWG changes %, response is sent, but poolController is only framing them 1-2 bytes at a time and not correctly interpreting responses. 

`Error writing chlorinator packet 'Set Salt %' to serial bus.  Tried 5 times to write 16,2,80,17,55,170,16,3`
and eventually:
`Aborting chlorinator packet Set Salt %.  Tried 10 times to write 16,2,80,17,55,170,16,3`

`{"type":"packet","packet":[16,2,80,17,55,170,16,3],"counter":0,"equipment":"chlorinator","direction":"outbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.167Z"}
{"type":"packet","packet":[16],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.193Z"}
{"type":"packet","packet":[2],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.194Z"}
{"type":"packet","packet":[0,18],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.196Z"}
{"type":"packet","packet":[68],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.197Z"}
{"type":"packet","packet":[0],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.198Z"}
{"type":"packet","packet":[104],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.224Z"}
{"type":"packet","packet":[16,3],"direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.225Z"}
{"type":"url","counter":0,"url":"/device","direction":"inbound","level":"info","message":"","timestamp":"2019-06-30T18:33:21.347Z"}
`
### Steps to reproduce the problem
Start nodejs-poolController

### Specifications

1. What version of the code are you using?
6.0-DEV (Also happens on 5.3 stable(.
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment
Aqua-Rite T-9 AL-2 
Only thing in on the RS485
[debug.zip](https://github.com/tagyoureit/nodejs-poolController/files/3343232/debug.zip)

tagyoureit, 2019-07-02:
I took a quick look at this.  The app will typically assemble partial packets as they come in.  However, we will also eject packets that are thought to be noise (random bytes between packets).  For whatever reason, your chlorinator is sending packets slow enough that we are discarding them as we come in.

Here is a temporary fix... in [line 75](https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/comms/inbound/receive-buffer.js#L75) put the following code:
```
        if (bufferToProcess.length <= 9) {
            processingBuffer = false;
            return;
        }
```
This will wait until we have at least 9 packets to start processing.  Then navigate to http://localhost:3000/chlorinator.

If you want to see the messages in the console (logs) then also make the log level debug in `config.json`: `"logLevel": "debug",`

I'll incorporate a better fix in the 6.0.

raz0rf0x, 2019-07-07:
Will attempt this and get back to you on the results. Had to overhaul the plumbing on my pool and everything is in shambles. In the interim I had created a Python Scott script to handle the chlorinator through MQTT. I may stick with this until I decide to shift from relay control of the pump to RS485 like the chlorinator.

Also, a note:

With the Aqua Rite in stand-alone mode I had to change the chlorinator to AL-5(Jandy) mode to get status/error messages. In AL-0 through AL-4 it would accept control the system but not send back any feedback other than the salinity levels in response to the chlorinator control. No matter what the status would come back 0.

jarrah42, 2019-09-16:
I've been seeing the same message using an iChlor 40. I added the code above, and it seems to have fixed the problem.

closed 2019-09-17 by tagyoureit

timelery, 2020-03-06:
@tagyoureit Do you have any recommendations on trying to understand how to construct a message to a Jandy JXI heater. only a few simply things I would like to accomplish. Turn Spa Heat On/Off and read temperature from heater.

Testing on the command line I get this. I do not have a controller. RS485 is connected between Jxi heater and PC.

[image]

tagyoureit, 2020-03-06:
The 16,2...16,3 is the same protocol the Intellichlor (also Goldline) protocol.  Check out some of the links in the [Readme](https://github.com/tagyoureit/nodejs-poolController#credit) and also what I have documented for the [chlorinators](https://github.com/tagyoureit/nodejs-poolController/wiki/Chlorinator). 

If 68 hex (104 decimal) is the address of the heater it shouldn't be too hard to figure out the basics.

timelery, 2020-03-06:
> The 16,2...16,3 is the same protocol the Intellichlor (also Goldline) protocol. Check out some of the links in the [Readme](https://github.com/tagyoureit/nodejs-poolController#credit) and also what [...]

Any idea why I am getting that constant incrementing sequence back that is shown in the screenshot

commit 4ae88d50 (2019-09-17): Temporary chlor fix

#137