tagyoureit/nodejs-poolController #139: Expose version from http API

leftyfl1p, 2019-08-03:
Currently there is no way to retrieve the poolcontroller version from the http API. This is useful (and in my opinion necessary) for extensions to check against for version compatibility reasons. Especially with the upcoming 6.0 which may or may not include breaking changes this should be included in some non breaking 5.x update.

dhop90, 2019-08-08:
Have you tried http://<pool controller ip>:3000/device?

it should return an xml structure with the version with major and min numbers

leftyfl1p, 2019-08-08:
Interesting that @tagyoureit didn't mention this to me when I asked him about it although this is in xml and I would prefer to have it in json rather than using another parsing library just for 1 endpoint.

dhop90, 2019-08-09:
This is all the processing I needed:
 def msg = hubResponse.xml
    if (msg) {
      def body = msg
        if (!body) {
            logerror "body was nil, returning"
            return null
        }    
        def verMajor = body.specVersion.major
        def verMinor = body.specVersion.minor
        def verPatch = body.specVersion.patch
        def manufacturer = body.device.manufacturer
        def modelDescription = body.device.modelDescription
        def friendlyName = body.device.friendlyName

tagyoureit, 2020-05-26:
This is included with the `/config/all` endpoint in 6.0.

closed 2020-05-26 by tagyoureit