tagyoureit/nodejs-poolController #141: Pump display toggling

jarrah42, 2019-09-16:
Hi, I'm using version 5.3.1. I have a Pentair VS and iChlor 40 with no controller. Using the virtual pump controller, I've noticed that the pump display continuously toggles from enabled to disabled and back again. I also the following messages:

15:12:04.179 VERBOSE Msg# 437   Main --> Pump 1: Remote control - disable pump control panel: [165,0,96,16,4,1,255,2,25]
15:12:04.199 VERBOSE Msg# 438   Pump 1: Remote control -  disable pump control panel: [165,0,16,96,4,1,255,2,25]
15:12:04.241 VERBOSE Msg# 440   Pump 1 responded with acknowledgement: [165,0,16,96,1,2,0,32,1,56]
15:12:04.284 VERBOSE Msg# 441   Main --> Pump 1: Remote control - enable pump control panel: [165,0,96,16,4,1,0,1,26]
15:12:04.324 VERBOSE Msg# 442   Pump 1: Remote control -  enable pump control panel: [165,0,16,96,4,1,0,1,26]

Is there any way to prevent this from happening (i.e. to configure the panel to always be enabled or disabled)? My preference would be to keep in enabled when there is no controller.

tagyoureit, 2019-09-17:
I'm working on a brand new version of the code now (see next branch) and am rewriting everything to be much more stable.  I'll include this in that process but don't have the time to go back and fix this in the 5.3 branch.

jarrah42, 2019-09-17:
Is it worth trying next or is it too early yet?

tagyoureit, 2019-09-17:
There is no support for standalone controllers in it, so not yet.

timelery, 2020-03-10:
@tagyoureit  @jarrah42 Are either of you aware of issues with the pump not switching back to local control when the message is sent?

Setting the pump to remote control works however but I am unable to switch it back to local.

This is the message I am sending to set back to local and I get back an error from the pump

Message Sent to set pump to local
255,0,255,165,0,96,33,4,1,255,2,42

Response from pump - I believe the 255 indicates an error - Any ideas?
[165,0,33,96,255,1,0,2,38]

tagyoureit, 2020-05-13:
v6 is ready for testing.
[Chlorinator directions](https://github.com/tagyoureit/nodejs-poolController/wiki/Virtual-Chlorinator-Controller-v6)
[Pump directions](https://github.com/tagyoureit/nodejs-poolController/wiki/Virtual-Pump-Controller---v6)

closed 2020-05-13 by tagyoureit

jarrah42, 2020-05-13:
As soon as I open my pool (hopefully this weekend), I'll try it out!