tagyoureit/nodejs-poolController #142: Add support for IntelliChem

Timeteo, 2019-09-22:
Would like to integrate with IntelliChem without using EasyTouch/IntelliTouch.

tagyoureit, 2020-04-19:
Do you have an RS485 adapter and are you able to run the wires?  If so, I can give you some directions to try and get the app to talk to the IntelliChem.

tagyoureit, 2020-07-04:
There are now endpoints in the app so you can integrate with IntelliChem.  We still need help figuring out how to communicate with Intellichem, however.

happyiphone, 2020-08-09:
I have an IntelliChem and the RS485 to USB. Connected the cables to the IntelliChem but have 0 communication with the device (cables are connected correctly, triple checked, I can see the send led light every time I type something on a terminal but the device doesn't send a thing back), is there some type of startup command or something like that? I was expecting the device to send some type of update over serial ever X seconds but not yet (opened the port at 9600 8,N,1) if there's something special I need to do, please, let me know and I'll do it so you can add support to it.

Awesome project! Thanks

tagyoureit, 2020-08-09:
We already have the base framework in place to talk to an IntelliChem but have had trouble finding people with the right equipment to do testing.
If you open up your browser and navigate to `http://{yourserver}:4200/config/chemController/search` we send out a bunch of packets to try and search.  Right now it looks like there are a couple "masters" on the bus; the IntelliChem can be a master and so can the OCP (outdoor control panel).  
See https://gitter.im/nodejs-poolController/Lobby?at=5ec4af206773a13b23f0883e for a discussion on the topic.
Pull the latest `next` code and try running that command there.  Also, do use the [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and post the results back here so we can continue to iterate.

happyiphone, 2020-08-11:
After I posted this I think I understand that to be able to send out a bunch of packets I need the controller on the line, which I don't have :) Is my understanding correct? If not I can definitely do it and see what do I get back and send you all you need. Thanks!

tagyoureit, 2020-08-14:
You will need this app and the IntelliChem on the same line, yes.

happyiphone, 2020-08-18:
I think I've found why I was getting some of those errors, used the new JBKey and I see the Intellichem now, just not with nodePoolController :) I can see it with Node-Red.

When I run NPC I get a lot of logs like this:

[8/18/2020, 1:17:05 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:17:07 AM] warn: Message aborted after 4 attempt(s): 165,33,16,23,19,1,0,1,1
[8/18/2020, 1:17:07 AM] warn: No response from chem controller: src: 23, dest: 16, action: 19, payload: [0] 
[8/18/2020, 1:17:12 AM] warn: Message aborted after 4 attempt(s): 165,33,16,33,19,1,0,1,11
[8/18/2020, 1:17:12 AM] warn: No response from chem controller: src: [default], dest: 16, action: 19, payload: [0] 
[8/18/2020, 1:17:15 AM] info: Timeout is retry_timeout
[8/18/2020, 1:17:15 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:17:15 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:17:16 AM] warn: Message aborted after 4 attempt(s): 165,33,16,23,231,1,0,1,213
[8/18/2020, 1:17:16 AM] warn: No response from chem controller: src: 23, dest: 16, action: 231, payload: [0] 
[8/18/2020, 1:17:20 AM] warn: Message aborted after 4 attempt(s): 165,33,16,33,231,1,0,1,223
[8/18/2020, 1:17:20 AM] warn: No response from chem controller: src: [default], dest: 16, action: 231, payload: [0] 
[8/18/2020, 1:17:24 AM] warn: Message aborted after 4 attempt(s): 165,33,16,23,217,1,0,1,199
[8/18/2020, 1:17:24 AM] warn: No response from chem controller: src: 23, dest: 16, action: 217, payload: [0] 
[8/18/2020, 1:17:25 AM] info: Timeout is retry_timeout
[8/18/2020, 1:17:25 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:17:25 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:17:28 AM] warn: Message aborted after 4 attempt(s): 165,33,16,33,217,1,0,1,209
[8/18/2020, 1:17:28 AM] warn: No response from chem controller: src: [default], dest: 16, action: 217, payload: [0] 
[8/18/2020, 1:17:29 AM] info: FREAKING DATA IS COMING BACK************************���!�)����
Failed:{"bytesReceived":2144,"success":0,"failed":21,"bytesSent":3084}
[8/18/2020, 1:17:29 AM] info: FREAKING DATA IS COMING BACK************************EZPP
�<�
[8/18/2020, 1:17:30 AM] info: FREAKING DATA IS COMING BACK************************���!�)����
Failed:{"bytesReceived":2248,"success":0,"failed":22,"bytesSent":3096}
[8/18/2020, 1:17:30 AM] info: FREAKING DATA IS COMING BACK************************EZPP
�<�
[8/18/2020, 1:17:31 AM] info: FREAKING DATA IS COMING BACK************************���!�)����
Failed:{"bytesReceived":2352,"success":0,"failed":23,"bytesSent":3108}
[8/18/2020, 1:17:31 AM] info: FREAKING DATA IS COMING BACK************************EZPP
�<�
[8/18/2020, 1:17:32 AM] info: FREAKING DATA IS COMING BACK************************���!�)����
Failed:{"bytesReceived":2456,"success":0,"failed":24,"bytesSent":3120}
[8/18/2020, 1:17:32 AM] info: FREAKING DATA IS COMING BACK************************EZPP
�<�
[8/18/2020, 1:17:33 AM] warn: Message aborted after 4 attempt(s): 165,33,144,16,210,1,210,3,11
[8/18/2020, 1:17:33 AM] warn: No response from chem controller: src: 16, dest: 144, action: 210, payload: [210] 
[8/18/2020, 1:17:37 AM] warn: Message aborted after 4 attempt(s): 165,33,144,33,210,1,210,3,28
[8/18/2020, 1:17:37 AM] warn: No response from chem controller: src: [default], dest: 144, action: 210, payload: [210] 
[8/18/2020, 1:17:42 AM] info: Timeout is undefined
[8/18/2020, 1:17:42 AM] info: Serial port: /dev/tty.wchusbserial1440 opened
[8/18/2020, 1:17:42 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:17:52 AM] info: Timeout is retry_timeout
[8/18/2020, 1:17:52 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:17:52 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:18:02 AM] info: Timeout is retry_timeout
[8/18/2020, 1:18:02 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:18:02 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:18:12 AM] info: Timeout is retry_timeout
[8/18/2020, 1:18:12 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:18:12 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:18:22 AM] info: Timeout is retry_timeout
[8/18/2020, 1:18:22 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:18:22 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:18:32 AM] info: Timeout is retry_timeout
[8/18/2020, 1:18:32 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:18:32 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:18:42 AM] info: Timeout is retry_timeout
[8/18/2020, 1:18:42 AM] error: Serial port /dev/tty.wchusbserial1440 recovering from lost connection
[8/18/2020, 1:18:42 AM] info: Serial port: /dev/tty.wchusbserial1440 request to open succeeded without error
[8/18/2020, 1:18:52 AM] info: Timeout is retry_timeout
[... cut ...]
My Intellichem is also version 1.060 so it SHOULD have the same protocol as the one used for NPC so not sure what else to do. Could it be that my unit is setup with some other settings making it different than another unit and is sending something back in a different way? I don't think it can be because I resented the unit as soon as I've got it,.

If there's ANYTHING I can do to solve this issue I would love to do it, I would hate to not use all the work that has been done on this project already.

tagyoureit, 2020-08-18:
> If there's ANYTHING I can do to solve this issue I would love to do it, I would hate to not use all the work that has been done on this project already.

Absolutely... we all want the same thing.  To bring it up to the 10,000' level there have been different users throughout the years that have wanted Intellichem support.  I don't have one (nor does rstrouse) so trying to remote debug/decode it has been challenging.  But now we have a good critical mass and have the status packet [nearly] fully decoded.  And you are getting data back so we are super close.  Let me make a few tweaks on my side and we'll see what's actually coming back and work from there.  It will still be an iterative process but so long as you can continue testing we'll continue to work on the code so we can be a virtual controller.  I'll post something back later today.

happyiphone, 2020-08-18:
That'll be awesome! What I can do is open up a connection to my machine so you can connect to it and query the real device. I've been there before not having the real device and I would have loved to have it, it'll make things SO much easier so that' why I'm offering, let me know if this helps and we need to coordinate to be on at the same time, at least the first time, Again, thanks!

tagyoureit, 2020-08-19:
I spent a bunch of time on this today and think it's mostly dialed in.  Here's a whole lot of information for you to go through and try:

1. In `config.json` set `inactivityRetry` to some really high number and you won't get those `... recovering from lost connection` messages.  This is on the list to address but just haven't done it yet.
2. Just like before, start up the app and run http://localhost:4200/config/chemController/search to start the search for the chem controllers.  It should search for 4 and hopefully the first one will come back a hit at address 144.  
3. If you think you are still getting data back but it isn't being logged, I suggest you download @rstrouse's excellent dashPanel and load up `localhost:5150/messageManager.html`.  It is a great view of all the inbound/outbound packets.  You can also turn on extra logging.    Click the green icon to start/stop the logging.  Use the yellow buttons as a guide to turn on all the logging.  Hopefully, you'll not need this and we'll find the chem controller but if not this will show you what we are seeing.  [dashPanel link](https://github.com/rstrouse/nodejs-poolController-dashPanel)
[image]

Couple things to note:
1. This will only discover the intellichem right now.  I have to wire up the bits to continually ask it for status.  If you "re-search" it should just grab the newest data.
2. I need to update -webClient to match the new data structures I've been working on.  It mostly works but check the /data config and state files for the exact readings.

happyiphone, 2020-08-19:
I've got the latest Next branch but that one didn't register anything. I used the previous Next branch with the changes I've made to the code and I do get some activity with the Intellichem.

Any of this helps? Can I capture something or do something that I can send you to test?

[image]
[image]

happyiphone, 2020-08-19:
I just realized that I can send the "query" command with the nodePanel and the Next branch, updated a few hours ago and get back a result from the Intellichem like nothing happened (the same message says nothing about being invalid). Not sure why this one is different but the intellichem search (http://localhost:4200/config/chemController/search) is not finding it.

[image]

happyiphone, 2020-08-19:
Let me know if a connection to my equipment will help you and I can set that up for you.

tagyoureit, 2020-08-20:
Did you join the Gitter chat?  We can exchange PM there. 

Join the chat in nodejs-poolController/Lobby https://gitter.im/nodejs-poolController/Lobby?utm_source=ios&utm_medium=link&utm_campaign=ios-share-link