tagyoureit/nodejs-poolController #145: pm2 unable to start 5.2.0

afgoody, 2019-10-13:
Not sure when this started, but probably within the last week. pm2 is unable to start nodejs-poolController. I have not change anything with the RPi file configuration. When I noticed it wasn't updating, I checked pm2 for the and the process was stopped, so I tried to manually restart it, but it would not start. So, I power cycled the RPi and pm2 is continually trying to restart the process (as it should), but it's failing.

Here are the error lines:

0|poolRPiController  | npm
0|poolRPiController  |  ERR! code ELIFECYCLE
0|poolRPiController  | npm ERR! errno 1
0|poolRPiController  | npm
0|poolRPiController  |  ERR! nodejs-poolcontroller@5.2.0 start: `node src/index.js "pool.json"`
0|poolRPiController  | npm ERR! Exit status 1
0|poolRPiController  | npm ERR!
0|poolRPiController  | npm ERR! Failed at the nodejs-poolcontroller@5.2.0 start script.
0|poolRPiController  | npm
0|poolRPiController  | ERR! This is probably not a problem with npm. There is likely additional logging output above.

tagyoureit, 2019-10-14:
I had an issue where I ran out of disk space because PM2 was logging everything.  If that's your problem, try [logrotate](https://github.com/keymetrics/pm2-logrotate).  

If that isn't the problem, just try starting the pool app via command line and see if there is another error that pops up.

afgoody, 2019-10-14:
Thanks. I think both are the issue. When trying to install pm2 rotate, I got a disk full error, so I did these commands:
pm2 flush
pm2 install pm2-logrotate
pm2 set pm2-lograte:max_size 10M 
pm2 set pm2-lograte:retain 14
sudo reboot

Upon reboot, I expected pm2 to start up again, but it did not, so I tried your second fix.
npm start pool.json

It error'd out claiming some issue with the configuration file (which I have not changed). I'm attaching the log hoping you can see what I'm missing.
[2019-10-14T18_59_36_408Z-debug.log](https://github.com/tagyoureit/nodejs-poolController/files/3726023/2019-10-14T18_59_36_408Z-debug.log)

tagyoureit, 2019-10-17:
Sorry for the delay.  I did try to download this just now and was getting an error that the file doesn't exist.

Most likely, you need to reconfigure the config.json.  Because we write to that file, it probably became corrupt when your drive ran out of space.  There isn't anything else on 5.3 that is written to disk.

afgoody, 2019-10-18:
No worries. That fixed it. Thanks!

tagyoureit, 2019-10-18:
👍

closed 2019-10-18 by tagyoureit

GoodyGizmos, 2019-11-24:
I just had to flush pm2 and restore the config.json file again after only a month. I was running logrotate at 10M and 14 logs. I change to 5 logs and will see how stable it is from here.

GoodyGizmos, 2019-12-13:
I'm still having issues keeping pm2 running even at 5 logs. It appears the disk is getting full even after a flush still. How can I determine what's filling the disk?

@tagyoureit, can you re-open this issue?

tagyoureit, 2019-12-13:
You can turn down (or off) the logging levels with this app. I would also suggest to look to the PM2 team to see why you are running out of space Amd the logs aren’t rotating. 

Add this isn’t really an issue with this app there isn’t anything I can do to fix it and it doesn’t make sense to keep this issue open.

GoodyGizmos, 2019-12-14:
Ok. Thanks.