tagyoureit/nodejs-poolController #147: Cannot install, (Serial port fails to install)

LavaGlass, 2019-11-28:
Expected Behavior: Running "npm install" in the nodejs-poolcontroller directory would install all dependencies, etc.

Actual Behavior: Errors occur saying that serialport could not be installed and that it cannot write to a certain directory.

[image]

Steps to reproduce the problem: Run npm install.

1. What version of the code are you using?
5.2.0 (tried with 5.3.0 aswell)

3. Any errors that are in the console
Lots! See Here: https://pastebin.com/CF7K767n

4. Your config.json or in-use configuration file
N/A

5. Your pool equipment
N/A

Any help is appreciated as I have been searching on the serialport docs, issues here and even followed the steps in this issue to no avail. https://github.com/tagyoureit/nodejs-poolController/issues/113

Thanks.

tagyoureit, 2019-11-28:
Looks like those serialport docs got moved [here](https://serialport.io/docs/guide-installation).

99% of the time the command `sudo npm install serialport --unsafe-perm --build-from-source` resolves the issue.

I would also recommend dropping back to Node v8 (maybe v10).  Looks like you are on v12 which might not be supported by the version of serialport packaged with 5.2/5.3 of this app.  You could also try to upgrade the serialport dependency but usually running the latest Node version creates some kind of havoc.

LavaGlass, 2019-11-28:
FIXED!

Downgraded to Node 8.9 and it works like a charm.

Thanks @tagyoureit 

[image]

closed 2019-11-28 by LavaGlass