tagyoureit/nodejs-poolController #148: Working Versions ?

ThrowinSticks, 2019-12-04:
Doing a brand new install of poolController and having problems compiling (serial port does not compile during poolController install, although I can run "npm install serialport" and it works fine outside of the poolControler install).  

So I'm hoping to get known working versions so I don't troubleshoot something for nothing.

HW
Raspberry Pi 3 w/ 64GB SD card

OS 
Raspbian Buster Lite

Node JS
(Arm7 binaries from Node JS download page)
12.13.1

NPM
6.13.2

poolController 
5.3.0 (also tried 5.2.0, same error)

So just thinking I might have better luck by switching to Jessie or a different Node JS version.  Any advice would be appreciated.

Here is the error message FYI:
prebuild-install WARN install No prebuilt binaries found (target=12.13.1 runtime=node arch=arm platform=linux)
make: Entering directory '/usr/local/nodejs-poolController-5.3.0/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)

ThrowinSticks, 2019-12-04:
It looks like I solved my own problem.  Same hardware, but installed
latest Stretch Lite
Node JS v 10.17.0
NPM v 6.11.3
poolController v 5.3.0

Everything compiled correctly and appears to be running.  Now time to conect it up to my pool equipment.

tagyoureit, 2019-12-05:
The 5.3 dependencies aren’t compatible with Node v12. Glad you figured it out.

closed 2019-12-05 by tagyoureit