tagyoureit/nodejs-poolController #150: Virtual Pump Commands not working

timelery, 2020-03-05:
It doesn’t look like anything in pump.html actually works.

I also do not see this Socket method even existing.

setPumpCommand

All the buttons on pump.html emit this setPumpCommand but it does not exist.

Can you confirm or point me to it.

tagyoureit, 2020-03-06:
That file is old and should just be removed.  The API's are on this (and the following section) in the [Readme](https://github.com/tagyoureit/nodejs-poolcontroller#pumps).

What is your equipment setup?  I've been working on the 6.0 which is a complete re-write and the API's will be changing.

timelery, 2020-03-06:
That explains it. It is working from the bootstrap page you referenced. My setup is simply a linux system connected directly to my Intelliflow VSF pump. I am really looking forward to the 6.0 release. Hopefully it will soon include Virtual Controller support. Thanks!

tagyoureit, 2020-03-06:
Should be available shortly for testing.  Keep an eye on the Gitter forums.

closed 2020-03-06 by tagyoureit