tagyoureit/nodejs-poolController #152: Sequence and Timing of Pump Messages

timelery, 2020-03-13:
@tagyoureit I was hoping to ask for your insight base don all the work you have done here.

I am struggling to figure out why I get inconsistent results when powering off and on my Intelliflow pump. After spending a few days looking at your code I have the following. It works but not consistently and it makes me think my timing or sequence if off or more importantly I am missing a packet that needs to be included. 

Do you have any ideas?

**To Power On the Pump into Program 3**
Packet 1 (Set to Remote) 165,0,96,33,4,1,255
Packet 2 (Set Program 3) 165,0,96,33,1,4,3,33,0,24 -- 175ms after previous message
Packet 3 (Turn Pump On) 1165,0,96,33,6,1,10 -- 175ms after previous message

**To Power Off the Pump from Program 3**
165,0,96,33,4,1,255]
165,0,96,33,1,4,3,33,0,0 -- 175ms after previous message
165,0,96,33,6,1,4 -- 175ms after previous message

tagyoureit, 2020-03-15:
This is the repeating sequence when the pump is On:
```
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,7,0][][1,28]],"ts":"2020-03-14T16:31:42.878-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,4,1][255][2,25]],"ts":"2020-03-14T16:31:50.924-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,6,1][10][1,38]],"ts":"2020-03-14T16:31:50.925-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,1,4][2,196,5,200][2,173]],"ts":"2020-03-14T16:31:51.102-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,7,0][][1,28]],"ts":"2020-03-14T16:31:56.999-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,4,1][255][2,25]],"ts":"2020-03-14T16:32:05.035-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,6,1][10][1,38]],"ts":"2020-03-14T16:32:05.036-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,96,16,1,4][2,196,5,200][2,173]],"ts":"2020-03-14T16:32:05.210-0700"}
```

It asks for the status (7) and then waits 8 secs and then controls the pump and then repeats (over and over again).  The packets I put here are a bit different because it is controlling the pump directly (not telling the pump to run a stored program) but maybe the sequence will help.  

Here it is when it is Off:
```
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,97,16,7,0][][1,29]],"ts":"2020-03-14T16:43:22.014-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,97,16,4,1][255][2,26]],"ts":"2020-03-14T16:43:30.221-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,97,16,6,1][4][1,33]],"ts":"2020-03-14T16:43:30.222-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,97,16,7,0][][1,29]],"ts":"2020-03-14T16:43:36.088-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,97,16,4,1][255][2,26]],"ts":"2020-03-14T16:43:44.290-0700"}
info:   {"valid":true,"dir":"in","proto":"pump","pkt":[[][255,0,255][165,0,97,16,6,1][4][1,33]],"ts":"2020-03-14T16:43:44.291-0700"}
```

Again, it asks for status and waits 7 seconds and then tells the pump to power off and repeats.

I'm working on this bit of code currently but 
a) it's raining here (finally) so I'm no outside sitting with my pump and 
b) I'm not going to be supporting storing the programs on the pump with future versions.  It isn't really needed because it is much easier to control the pumps directly instead of trying to store the programs and then run them.  Also, with storing the programs on the pump you are limited to only 4.

I'll post here if I find anything else out.

TimAtSnow, 2020-04-01:
Thank you, this helps a bit. Is there a reason you continuously send the packet sequence over and over again as opposed to just one sequence for power one and one for power off?

TimAtSnow, 2020-04-01:
Also, what is the logic behind the 8 second wait between status check and sending power on/off. Thanks

tagyoureit, 2020-04-08:
I modeled my sequence after what I've observed the Pentair controller send.  I believe the pumps need these packets in this sequence or they will shut off.  The timing delay is also based on what I observe.

closed 2020-05-13 by tagyoureit