tagyoureit/nodejs-poolController #153: Setting Pump back to local Schedule Mode

timelery, 2020-03-13:
If I have a schedule set on the pump locally and I control it remotely by turning it off or setting a new speed and turning it on I am unable t set the pump back to local schedule mode and have to physically press it on the pump to switch the pump back to local schedule mode.

Are you aware of a command I can send the pump to set it back to local schedule mode?

Thanks,
Tim

tagyoureit, 2020-03-15:
I'm not.  I think there is a command I've seen to "initialize" the pump to work with the controller but generally once the pump is controlled by an external controller it isn't designed to also be controlled by a local schedule.  

The pumps do have their own timeout.  If they don't see anything from a controller in ~1 min I think they might reset on their own.  Have you waited to see if that happens?

timelery, 2020-03-19:
I figured this out. Sending a clean pump power on puts the pump back in local schedule mode.

[165,0,96,33,6,1,10] //Sets pump back to local schedule mode after a power off cycle

closed 2020-03-19 by timelery

tagyoureit, 2020-05-04:
Thanks, this was helpful!  I was incorporating this into my app and was able to get the pump to return to local control with only:
```
verbose: Wrote packet [255,0,255,165,0,96,33,6,1,4,1,49].  Retries remaining: 1
verbose: Wrote packet [255,0,255,165,0,96,33,4,1,0,1,43].  Retries remaining: 1
```

Would your pump not start running the local schedule again without the [5,1,0] and [6,1,10] packets?