tagyoureit/nodejs-poolController #157: iChlor water temp packets

barrelman70, 2020-04-19:
I have the iChlor connected to intelliconnect.  With intelliconnect I can see the water temp, so I tried to figure out where this data was coming from.  Here is what I found 

`10 0210 1600 0f49 0005 1085 1003`
I was at 15% and the temp was 73 F
`0f49` - 15 and 73 

I changed to 10% and got this
`10 0210 1600 0a49 0005 1080 1003`
`0a49` - 10 73 

I then waited for the temp to change, it changed to 74 and I got this
`10 0210 1600 0a4a 0005 1081 1003`
`0a4a` - 10 74

Hopefully this will help you grab water temp from the iChlor.  I don't have a heater or anything else that provides this, so I'm hoping you can grab it from here.

tagyoureit, 2020-04-19:
Interesting... haven't even seen the IntelliConnect before.  

The 16 (22 dec) packet is also new.  Wonder why the intelliconnect is sending the temp to the chlorinator?  Which specific version of the chlorinator do you have?

I can add this to the packet decoding.  Do you see other packets on the rs485 bus?  Any idea what the 05,10,8x parts are?

barrelman70, 2020-04-20:
This is a brand new install, so maybe IntelliConnect is new, not sure.  I have the [iChlor 30](https://www.pentair.com/en/products/pool-spa-equipment/pool-water-treatment/ichlor_salt_chlorinegenerator.html).  The iChlor has a built in temp sensor, so it is sending that to intelliconnect, not vice versa. I haven't figured out what the other parts are yet.  I'm at 3350 ppm and none of these seem to fit that.  However, the poolController already knows this `{"chlorinator":{"installed":1,"saltPPM":3350,"currentOutput":10,"outputPoolPercent":-1,"outputSpaPercent":-1,"superChlorinate":0,"version":-1,"name":"iChlor          ","status":"Ok","controlledBy":"intellicom"}}`

As far as the output number (as currently showing in poolController), it seems to always be the setpoint.  There is an indicator on the iChlor itself that turns on/off when it is actually generating. 
I'll look at actual packets and see if it is sending this data or not.

I also have a pump connected to IntelliConnect and poolController can read everything, but can't control it.  I also have lights installed, but poolController doesn't seem to know about those.

I'd be happy to try and decipher any other part of the protocol that you might need.

barrelman70, 2020-04-20:
using the console logging, these are the packets I'm seeing 

```
75 deg, 1%, 3350ppm
13:34:23.682 VERBOSE Msg# 765   Controller --> Salt cell: Set current output to 1.6 %: 16,2,80,21,0,119,16,3
13:34:23.707 DEBUG outputToHubitat sent event chlorinator
13:34:23.740 VERBOSE Msg# 766   Salt cell --> Controller: Other chlorinator packet?: 16,2,0,22,75,1,5,0,121,16,3
13:34:23.880 VERBOSE Msg# 767   Controller --> Salt cell: Other chlorinator packet?: 16,2,80,19,117,16,3
13:34:24.082 VERBOSE Msg# 768   Controller --> Salt cell: What is your version?: 16,2,80,20,0,118,16,3
13:34:24.283 VERBOSE Msg# 770   Controller --> Salt cell: Set current output to 1 %: 16,2,80,17,1,116,16,3
13:34:24.299 DEBUG outputToHubitat sent event chlorinator
13:34:24.329 DEBUG Msg# 771   Salt cell --> Controller: Current Salt level is 3350 PPM: 16,2,0,18,67,128,231,16,3

75 deg, 100%, 3350ppm
13:35:15.502 VERBOSE Msg# 851   Controller --> Salt cell: Set current output to Super Chlorinate %: 16,2,80,17,100,215,16,3
13:35:15.543 DEBUG outputToHubitat sent event chlorinator
13:35:15.583 DEBUG Msg# 852   Salt cell --> Controller: Current Salt level is 3350 PPM: 16,2,0,18,67,128,231,16,3
13:35:17.854 VERBOSE Msg# 853   Controller --> Salt cell: Set current output to 1.6 %: 16,2,80,21,0,119,16,3
13:35:17.875 DEBUG outputToHubitat sent event chlorinator
13:35:17.911 VERBOSE Msg# 854   Salt cell --> Controller: Other chlorinator packet?: 16,2,0,22,75,100,5,0,220,16,3
13:35:18.075 VERBOSE Msg# 855   Controller --> Salt cell: Other chlorinator packet?: 16,2,80,19,117,16,3
13:35:18.257 VERBOSE Msg# 856   Controller --> Salt cell: What is your version?: 16,2,80,20,0,118,16,3
13:35:18.459 VERBOSE Msg# 858   Controller --> Salt cell: Set current output to Super Chlorinate %: 16,2,80,17,100,215,16,3
13:35:18.475 DEBUG outputToHubitat sent event chlorinator
13:35:18.505 DEBUG Msg# 859   Salt cell --> Controller: Current Salt level is 3350 PPM: 16,2,0,18,67,128,231,16,3

75 deg, 95%, 3350ppm
13:36:01.211 VERBOSE Msg# 946   Controller --> Salt cell: Set current output to 95 %: 16,2,80,17,95,210,16,3
13:36:01.234 DEBUG outputToHubitat sent event chlorinator
13:36:01.309 DEBUG Msg# 947   Salt cell --> Controller: Current Salt level is 3350 PPM: 16,2,0,18,67,128,231,16,3
13:36:12.042 VERBOSE Msg# 952   Controller --> Salt cell: Set current output to 1.6 %: 16,2,80,21,0,119,16,3
13:36:12.063 DEBUG outputToHubitat sent event chlorinator
13:36:12.095 VERBOSE Msg# 953   Salt cell --> Controller: Other chlorinator packet?: 16,2,0,22,75,95,5,0,215,16,3
13:36:12.236 VERBOSE Msg# 954   Controller --> Salt cell: Other chlorinator packet?: 16,2,80,19,117,16,3
13:36:12.439 VERBOSE Msg# 955   Controller --> Salt cell: What is your version?: 16,2,80,20,0,118,16,3
13:36:12.640 VERBOSE Msg# 957   Controller --> Salt cell: Set current output to 95 %: 16,2,80,17,95,210,16,3
13:36:12.655 DEBUG outputToHubitat sent event chlorinator
13:36:12.685 DEBUG Msg# 958   Salt cell --> Controller: Current Salt level is 3350 PPM: 16,2,0,18,67,128,231,16,3

75 deg, 50%, 3350ppm
13:42:31.304 VERBOSE Msg# 1531   Controller --> Salt cell: Set current output to 1.6 %: 16,2,80,21,0,119,16,3
13:42:31.326 DEBUG outputToHubitat sent event chlorinator
13:42:31.362 VERBOSE Msg# 1532   Salt cell --> Controller: Other chlorinator packet?: 16,2,0,22,75,50,5,0,170,16,3
13:42:31.503 VERBOSE Msg# 1533   Controller --> Salt cell: Other chlorinator packet?: 16,2,80,19,117,16,3
13:42:31.704 VERBOSE Msg# 1534   Controller --> Salt cell: What is your version?: 16,2,80,20,0,118,16,3
13:42:31.906 VERBOSE Msg# 1536   Controller --> Salt cell: Set current output to 50 %: 16,2,80,17,50,165,16,3
13:42:31.922 DEBUG outputToHubitat sent event chlorinator
13:42:31.954 DEBUG Msg# 1537   Salt cell --> Controller: Current Salt level is 3350 PPM: 16,2,0,18,67,128,231,16,3
```

When it was on 50% I compared the packets when the cell generation light on the iChlor was on and when it was off and they were the same

tagyoureit, 2020-04-20:
Ah, the IC30 must be the only one that has a temp setting.  Good to know.

> 13:42:31.304 VERBOSE Msg# 1531   Controller --> Salt cell: Set current output to 1.6 %: 16,2,80,21,0,119,16,3
The format of this 21 message is different from others I've seen.   It shouldn't be 1.6%.

I've added these to the 6.0/next branch and will push it up soon.

If you want to start a new issue and attach an entire [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) I can look at the circuits/lights, too.  Right now, the app knows nothing about the IntelliConnect.

barrelman70, 2020-04-20:
#158

tagyoureit, 2020-05-13:
Please test out the next branch and let me know if you have any issues.

closed 2020-05-13 by tagyoureit