tagyoureit/nodejs-poolController #158: intelliconnect packet capture

barrelman70, 2020-04-20:
Here is the packet capture from my intelliconnect.  I have an intellipro vsf pump, the ichor 30 swg, and intellibrite 5g color lights.  The intelliconnect phone interface is very slow, 5-30 secs between button pushes so the times I list might be a bit off.  Here is what I did

18:35 - started capture and did nothing else
37 - went into my intelliconnect app
        viewed pump status
38 - changed pump to preset 2
        stopped pump
39 - set preset 1
        set to schedule (0800 - 2200 ET)
41 - changed ichlor to 30% (3350ppm 75 deg)
        put on boost for 24 hrs
42 - set to 10%
44 - turn lights ON and to BLUE
         switched to CARIBBEAN
46 - turned lights OFF

[packetCapture.log](https://github.com/tagyoureit/nodejs-poolController/files/4505575/packetCapture.log)

[packetCapture.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/4505578/packetCapture.json.txt)

[config.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/4505579/config.json.txt)

Let me know anything else you'd like me to grab

tagyoureit, 2020-04-21:
I see some interesting packets which I don't recognize:
`{"type":"packet","packet":[255,0,255,166,1,176,12,103,1,0,1,203],"direction":"inbound","level":"info","message":"","timestamp":"2020-04-20T18:35:28.825Z"}`
but these never seem to change.

Other than that, everything else either seems to be a pump or a chlorinator packet.  It looks like the only way to control the relays is through their app which is going to be network calls and not going over the RS485 bus.

Pretty much the only thing you can do with the Intelliconnect is listen to the pump and the IntelliChlor.  You won't be able to control them as it would be competing with what the IntelliConnect is telling them to do.

barrelman70, 2020-04-22:
I figured you wouldn't see anything with the lights as they are relay only and not connected to the rs485 bus

The pump and swg have their own dedicated power and aren't connected to a relay in intelliconnect.  They only connect to the intelliconnect via rs485, so I think they have to be controlled through that.  Here is the [install guide](https://www.pentair.com/content/dam/extranet/aquatics/residential-pool/owners-manuals/automation/intelliconnect/intelliconnect-manual-english.pdf) not sure if it helps at all.

tagyoureit, 2020-04-24:
I don't think there is anything much that can be done to control the IntelliConnect, unfortunately.

closed 2020-04-24 by tagyoureit