tagyoureit/nodejs-poolController #159: Not capturing all Easytouch2 4p (virtual) circuits

baudfather, 2020-04-23:
### Expected behavior and actual behavior
Expected to load all circuits in v6, in my example circuits 11 and 12.  V6 only loads the first 4 circuits

### What is actually happening (screenshots are helpful)
Unable to access circuts above 4

### Steps to reproduce the problem

### Specifications

1. What version of the code are you using? 
6.0, attached packet capture from 5.3

2. All output from the file log 
[packetCapture1.zip](https://github.com/tagyoureit/nodejs-poolController/files/4525594/packetCapture1.zip)

3. Any errors that are in the console
No errors.

4. Your config.json or in-use configuration file
from v6: [config.zip](https://github.com/tagyoureit/nodejs-poolController/files/4525600/config.zip)

5. Your pool equipment 
Pentair Easytouch 4p w/ intelliflo VSF pump

tagyoureit, 2020-04-26:
Can you pull the `next` code I just pushed up now?  You may need to delete your `poolConfig.json` and `poolState.json` to pick up the new changes.

baudfather, 2020-04-26:
Success!

tagyoureit, 2020-04-26:
Ok, great!  Closing for now.

closed 2020-04-26 by tagyoureit