tagyoureit/nodejs-poolController #160: Where is the API documentation ?

ivanvach, 2020-04-28:
### Expected behavior and actual behavior

### What is actually happening (screenshots are helpful)

### Steps to reproduce the problem

### Specifications

1. What version of the code are you using?
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

tagyoureit, 2020-05-02:
Assuming you are asking about `next` since the `5.3` is listed in the README.md.  `next` is coming soon.  For now, you can look in the `/web` directory for config & state endpoints.  The `get`s are straightforward.  The puts typically accept any object that you will get back in the get but we are still flushing this out.

rerouted, 2020-07-27:
@tagyoureit Now that NEXT branch is GA when you have a moment can you add the API endpoints to the Wiki?  Like what you had before in 5.3.  Thanks

tagyoureit, 2020-07-31:
Beta documentation posted on https://tagyoureit.github.io/nodejs-poolcontroller-api/.  I may merge this with this project.  Please do provide some feedback.  I'll also post a note for this in the Gitter forums.

closed 2020-08-11 by tagyoureit