tagyoureit/nodejs-poolController #161: Basic info

tomascrespo, 2020-05-01:
Hi  I do not know if this is the best place to make a question, but I can't find any other way.

I'm building a small pool, 40m3, and I would like to get the info of the chlorinator to publish it in my Home Assistant installation. I'm intereseted in water temp, salt concentration and ph.

As I can see in this project there is nothing about ph. Does this project get ph data from any commercial product?

For my specific case, what equiment could I get nowadays ?

Lots of thanks

tagyoureit, 2020-05-02:
The IntelliChem and IntelliPH are the two Pentair products that will give you pH.  The chlorinator has salt info but does not measure pH.  

A few of us have "rolled our own" dispensing systems using commercially available pH and/or ORP probes.  You could do as much or as little of these as you would require.

This app doesn't directly talk to Home Assistant.  There are a number of [integrations](https://gitter.im/nodejs-poolController/HomeAutomation) to other platforms but doesn't look like anyone has contributed a native Home Assistant connector yet.  There is work in progress to migrate Hubitat to the `next` version and the other integrations will also need to be updated for the upcoming release.

You may want to check out https://gitter.im/nodejs-poolController/IntelliPH_IntelliChem or https://gitter.im/nodejs-poolController/HomeAutomation for previous conversations or to ask further questions.

hth

closed 2020-05-02 by tagyoureit

johnny2678, 2020-05-02:
Just to clarify, IntelliPH doesn't read PH.  It just doses acid on a timer.  IntelliChem reads PH & ORP and can control downstream salt cells or stenner pumps.

tagyoureit, 2020-05-02:
Oh, thanks for the clarification.