tagyoureit/nodejs-poolController #163: V6 console log

jgaro, 2020-05-14:
v6 console log of errors & json files
[image]
[image]

poolConfig.json
{
  "lastUpdated": "5/14/2020, 12:33:57 PM",
  "pool": {
    "options": {
      "clockMode": 12,
      "clockSource": "manual"
    }
  },
  "equipment": {
    "maxBodies": 1,
    "maxCircuits": 0,
    "maxSchedules": 0,
    "maxPumps": 2,
    "maxValves": 0,
    "maxCircuitGroups": 0,
    "maxLightGroups": 0,
    "maxIntelliBrites": 0,
    "maxChlorinators": 1,
    "maxCustomNames": 10,
    "model": "Virtual Controller",
    "maxFeatures": 10,
    "equipmentIds": {
      "circuits": {
        "start": 6,
        "end": 6
      },
      "features": {
        "start": 7,
        "end": 17
      },
      "circuitGroups": {
        "start": 0,
        "end": 0
      },
      "virtualCircuits": {
        "start": 128,
        "end": 136
      },
      "invalidIds": []
    }
  },
  "configVersion": {},
  "bodies": [
    {
      "id": 1,
      "isActive": true,
      "name": "Pool"
    }
  ],
  "schedules": [],
  "circuits": [
    {
      "id": 6,
      "name": "Pool",
      "type": 6,
      "isActive": true
    }
  ],
  "features": [],
  "pumps": [
    {
      "id": 2,
      "isActive": true,
      "isVirtual": true,
      "type": 0,
      "circuits": []
    },
    {
      "id": 1,
      "isActive": true,
      "isVirtual": true,
      "type": 128,
      "circuits": [
        {
          "id": 1,
          "circuit": 6,
          "units": 0,
          "speed": 1190
        }
      ]
    }
  ],
  "chlorinators": [
    {
      "id": 1
    }
  ],
  "valves": [],
  "heaters": [],
  "covers": [],
  "circuitGroups": [],
  "lightGroups": [],
  "remotes": [],
  "security": {},
  "customNames": [
    {
      "id": 1,
      "name": "Generic",
      "isActive": true
    }
  ],
  "eggTimers": [],
  "intellichem": {},
  "appVersion": "6.0.0",
  "intellibrite": {
    "id": 0,
    "isActive": false,
    "type": 3
  },
  "general": {},
  "controllerType": "virtual"
}

poolState.json
{
  "mode": {
    "val": 0,
    "name": "auto",
    "desc": "Auto"
  },
  "temps": {
    "units": {
      "val": 0,
      "name": "F",
      "desc": "Fahrenheit"
    }
  },
  "status": {
    "val": 0,
    "name": "initializing",
    "percent": 0
  },
  "equipment": {
    "model": "Virtual Controller",
    "maxBodies": 1
  },
  "pumps": [
    {
      "id": 2,
      "virtualControllerStatus": {
        "val": 1,
        "name": "running",
        "desc": "Running"
      }
    },
    {
      "id": 1,
      "type": {
        "val": 128,
        "name": "vs",
        "desc": "Intelliflo VS",
        "maxPrimingTime": 6,
        "minSpeed": 450,
        "maxSpeed": 3450,
        "maxCircuits": 8,
        "hasAddress": true
      },
      "status": {
        "name": "off",
        "desc": "Off",
        "val": 0
      },
      "command": 4,
      "mode": 0,
      "driveState": 0,
      "watts": 0,
      "rpm": 0,
      "flow": 0,
      "ppc": 0,
      "time": 833,
      "virtualControllerStatus": {
        "val": 1,
        "name": "running",
        "desc": "Running"
      }
    }
  ],
  "valves": [],
  "heaters": [],
  "circuits": [
    {
      "id": 6,
      "name": "Pool",
      "type": {
        "val": 6,
        "name": "pool",
        "desc": "Pool"
      },
      "isOn": false
    }
  ],
  "features": [],
  "chlorinators": [
    {
      "id": 1
    }
  ],
  "schedules": [],
  "circuitGroups": [],
  "lightGroups": [],
  "virtualCircuits": [],
  "intellibrite": {},
  "intellichem": {},
  "covers": [],
  "general": {}
}

tagyoureit, 2020-05-14:
How are you starting the webclient?  With `npm start`? 
And what is in your `config.json` for the `...-webClient`?  It looks like you have localhost hardcoded.  

Can you try to leave the config.json with `"discovery": {"poolControllerURL": "*"},`
and enable the SSDP server in the poolController config.json?

You should see something like this on the command line when you start the webClient.
```
Got a response to an m-search:
200
{
  "ST": "urn:schemas-upnp-org:device:PoolController:1",
  "USN": "uuid:806f52f4-1f35-4e33-9299-b8:27:eb:73:ba:9b::urn:schemas-upnp-org:device:PoolController:1",
  "CACHE-CONTROL": "max-age=1800",
  "DATE": "Thu, 14 May 2020 18:47:31 GMT",
  "SERVER": "node.js/8.15.0 UPnP/1.1 node-ssdp/4.0.0",
  "EXT": "",
  "LOCATION": "http://10.0.0.70:4200/device"
}
{
  "address": "10.0.0.70",
  "family": "IPv4",
  "port": 1900,
  "size": 331
}
Found pool server at new address http://10.0.0.170:4200 (previously was *)
```

jgaro, 2020-05-14:
yes, starting the client with npm start, and I did see the connection as you described after changing the discovery URL as you suggested. The resources still fail to load though. I now see this kind of error on dev console
[image]

tagyoureit, 2020-05-14:
Pull the latest and try again.  Thx.

jgaro, 2020-05-14:
works!

closed 2020-05-14 by jgaro