tagyoureit/nodejs-poolController #165: Cannot connect to panel

esunder, 2020-05-17:
Hi all, I got this up and working back in March. And it has been running great. All of a sudden now, I’m getting the error below. When I run “od -x” on my /dev/ttyUSB0 (rs485 usb connector), it just hangs. 

Is this possibly a faulty wire connection all of a sudden? How can I debug this? Can I hook a multimeter up to the data+, data- and see if I’m getting some kind of signal?

Note: my writings from base unit to raspberry pi is probably between 60-100ft. It’s quite a good distance. But it was working fine for a few months. 

I doubt this is an issue with the code/settings, since it was working for a couple months and then stopped working all of a sudden.

Any help would be appreciated. 

Thanks!

### Expected behavior and actual behavior
Node connects to main panel

### What is actually happening (screenshots are helpful)
Getting “serial port recovering from lost connection” every minute on startup

### Steps to reproduce the problem
Starting the app

### Specifications

1. What version of the code are you using?
5.3.3
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

tagyoureit, 2020-05-17:
My first step would be to spend another $6.99 on a spare/replacement USB adapter.  But beyond that I'd try these basics:

1. Make sure no other apps are locking/using the port.  EG don't have this app and the `od -x` running at the same time.  Also, if you don't have a pool controller (eg you have standalone pumps/chlorinators) then you won't see anything on the bus anyway until this app talks to it.
1. Unplug/replug in the adapter.  
1. Try the adapter in a different computer.
1. Take the adapter out to your controller/pump and connect it there to see if you have any breaks in the wire between your equipment.

I'm not an expert on multimeters and not sure what would show up there with successful communications.  I think you'd need something more like a digital signal processor instead of measuring electricity but could be wrong.

esunder, 2020-05-17:
Thanks, I’ll try these ideas and see where I get!

tagyoureit, 2020-05-26:
Please re-open if you still need help.

closed 2020-05-26 by tagyoureit