tagyoureit/nodejs-poolController #167: Temp jumps to 100 on heater toggle

baudfather, 2020-05-17:
### Expected behavior and actual behavior
[packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/4639713/packetCapture.zip)
No change to setpoint when heat toggles on/off

### What is actually happening (screenshots are helpful)
Setpoint on controller changes to 100 (celcius) when heater "off" is toggled or "heater" is toggled

### Steps to reproduce the problem
Fresh install, change setpoint works, if heat is toggled on or off, setpoint jumps to 100 at the controller (slider changes to max avail setting)

### Specifications

1. What version of the code are you using? Next, latest build as of issue
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment Easytouch 2 4p

baudfather, 2020-05-17:
[poolConfig.zip](https://github.com/tagyoureit/nodejs-poolController/files/4639729/poolConfig.zip)

tagyoureit, 2020-05-17:
I pushed up a fix for this, but not sure if I am computing all the circuits correctly.  Are all of your circuits assigned/used?  Is this correct?

```
 "circuits": [
    {
      "id": 2,
      "name": "Pool Light",
      "nameId": 63,
      "type": 16,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 0
    },
    {
      "id": 4,
      "name": "Not Used",
      "nameId": 53,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 0
    },
    {
      "id": 5,
      "name": "AUX 4",
      "nameId": 6,
      "type": 0,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 0
    },
    {
      "id": 6,
      "name": "Pool",
      "nameId": 61,
      "type": 2,
      "isActive": true,
      "freeze": false,
      "showInFeatures": true,
      "eggTimer": 0
    }
  ],
```

And what about features?  All used?  Can you set more?  You have 2 set but you should be able to set 8 IIRC.

baudfather, 2020-05-17:
Only pool and pool light circuits are being used, and yes, I beleive it can have 8 features, I'm only using the 2.  It was all set up by the pool installer and I haven't really played around with it - it does seem a bit odd the way they programmed it, not sure if it's a factory default or if there was a reason they programmed it the way they did.

tagyoureit, 2020-05-17:
Did the code changes I pushed up work for you?

And if you get a chance I'd be curious to see which circuits/features get assigned to which ID's.  It's not terribly hard to add/remove them (especially if you have ScreenLogic) but understand if you don't want to mess with it.  If you do, and can send me a "fully loaded" set of configuration files that would be very helpful!

baudfather, 2020-05-17:
Yup, the latest is working properly now (with flame!) - though it's not picking up the intelliclor unless I make a setting change from the panel.   I'll have to RTFM to figure out why the current config is the way it is, and then try playing around with it :)

tagyoureit, 2020-05-18:
I was able to see your chlorinator last night.  Try deleting the `poolConfig.json` and `poolState.json` and restart the app.

baudfather, 2020-05-18:
That did the trick, thanks!

closed 2020-05-18 by tagyoureit