tagyoureit/nodejs-poolController #168: Change of pm2 ecosystem file on wiki

jarrah42, 2020-05-24:
The pm2 ecosystem file you provide is javascript, so it needs to end in `.js`, not `.json` or you will get an error when running the `pm2 start ...` command. See the Javascript section [here](https://pm2.keymetrics.io/docs/usage/application-declaration/) for reference.

tagyoureit, 2020-05-24:
I updated the [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/PM2) yesterday, as a matter of fact.  Are those the directions you were referring to?

jarrah42, 2020-05-26:
Yes. Items 2 & 3 refer to `ecosystem.config.json`, however if you give that file to pm2 it will complain about a ';' at the end. The file name should be `ecosystem.config.js`.

tagyoureit, 2020-05-26:
D'oh!  Fixed.

closed 2020-05-26 by tagyoureit

johnny2678, 2020-05-26:
changing to .js let's it start without throwing an error, but then it goes into a loop restarting every 5 seconds or so.  Hard to troubleshoot b/c nothing is output to console or logs.

Not a huge deal.  I can always start using the `pm2 start 'npm start' --name poolController` method.

tagyoureit, 2020-05-26:
One of two things is happening...
1. PM2 is picking some file changes that are not in the exclude directory and it is restarting because watch=true.  Solution - change watch=false or exclude the directory/file that might be triggering the watch.
1. There is an error starting the app.  Solution - use one of `pm2 logs N` (where N is the process as listed in `pm2 list`) or `pm2 logs all` or the monitoring tool `pm2 monit`.  If there is an starting the app it should appear in the logs and you can fix the configuration file depending on what the error is.

LMK what you find so we can update the ecosystem file if necessary.

johnny2678, 2020-05-27:
hmm, I switched `watch` to `false` but am still seeing a process restart every 3-5 seconds.  Unfortunately, nothing is written to the logs during this time so `pm2 logs [X]` doesn't show anything.  `PM2 monit` is what I used to see the process restarting every 3-5 seconds, but nothing in the log window.

If it's just me seeing this, I wouldn't worry about about.  Just posting in case others are seeing the same issue.

johnny2678, 2020-05-27:
ex. loading using the ecosystem file fails as above but loading with `pm2 start 'npm start' --name poolController --watch --ignore-watch="data config.json node_modules logs dist replay .cache" --listen-timeout=60000 --restart-delay=3000` works fine

tagyoureit, 2020-05-30:
Did you try looking at the `./.pm2/dump.pm2` file?  It should be a snapshot of what you save.  Maybe there are clues in there.  Strange it doesn't display any logs at all.  (You can also check the logs directly in that directory.)