tagyoureit/nodejs-poolController #169: UnhandledPromiseRejectionWarning causes controller to stop working

jarrah42, 2020-05-26:
I was able to successfully control the pump, however after a couple of minutes the pump stopped. I see the following error in the pm2 log:

1|poolController            | received back pump power packet.
1|poolController            | warn: Message aborted after 2 attempt(s): 255,0,255,165,0,96,33,1,4,2,196,12,28,2,25
1|poolController            | received back run rpm.
1|poolController            | (node:6897) UnhandledPromiseRejectionWarning: ApiError: Message aborted after 2 attempt(s)
1|poolController            |     at SendRecieveBuffer.writeMessage (/home/pi/nodejs-poolController/controller/comms/Comms.ts:213:31)
1|poolController            |     at SendRecieveBuffer.processWaitPacket (/home/pi/nodejs-poolController/controller/comms/Comms.ts:174:102)
1|poolController            |     at SendRecieveBuffer.processOutbound (/home/pi/nodejs-poolController/controller/comms/Comms.ts:181:30)
1|poolController            |     at Timeout.processPackets [as _onTimeout] (/home/pi/nodejs-poolController/controller/comms/Comms.ts:167:98)
1|poolController            |     at listOnTimeout (internal/timers.js:531:17)
1|poolController            |     at processTimers (internal/timers.js:475:7)
1|poolController            | (node:6897) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 7)

tagyoureit, 2020-05-26:
I just pushed up two small changes.  First one is where I think the error is happening.  Second one is better logging in case I'm wrong.  Please re-open if it still happens.  And if you can consistently reproduce it please to use the capture tool.

jarrah42, 2020-05-26:
controller/boards/EasyTouchBoard.ts:768:57 - error TS2304: Cannot find name 'LightGroup'.

768     public async setLightGroupAsync(data: any): Promise<LightGroup> {
                                                            ~~~~~~~~~~

controller/boards/EasyTouchBoard.ts:771:1 - error TS2304: Cannot find name 'aaa'.

771 aaa
    ~~~

Found 2 errors.

npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! nodejs-poolController@6.0.0 build: `tsc`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the nodejs-poolController@6.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

jarrah42, 2020-05-26:
I'm still seeing the same error after updating.

tagyoureit, 2020-05-26:
I had a bad push.  Do the pull again and do the packet capture and I'll investigate.  Thx.

jarrah42, 2020-05-26:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4684912/replay.zip)

jarrah42, 2020-05-28:
I'm no longer seeing this error.

closed 2020-06-07 by tagyoureit