tagyoureit/nodejs-poolController #171: Memory leak causes controller to exit

jarrah42, 2020-05-29:
After about 2 days of running, the controller exited as follows:

1|poolController            | <--- Last few GCs --->
1|poolController            | [378:0x31edd08] 70633447 ms: Mark-sweep 206.1 (231.6) -> 205.7 (231.1) MB, 2476.8 / 0.2 ms  (average mu = 0.170, current mu = 0.099) allocation failure scavenge might not succeed
1|poolController            | [378:0x31edd08] 70633569 ms: Scavenge 206.1 (231.1) -> 206.0 (231.3) MB, 21.2 / 0.1 ms  (average mu = 0.170, current mu = 0.099) allocation failure 
1|poolController            | [378:0x31edd08] 70636254 ms: Mark-sweep 206.5 (231.3) -> 205.8 (231.6) MB, 2635.5 / 0.3 ms  (average mu = 0.118, current mu = 0.061) allocation failure scavenge might not succeed
1|poolController            | <--- JS stacktrace --->
1|poolController            | ==== JS stack trace =========================================
1|poolController            |     0: ExitFrame [pc: 0xe19c00]
1|poolController            |     1: StubFrame [pc: 0xe1bfd8]
1|poolController            |     2: StubFrame [pc: 0xded35c]
1|poolController            | Security context: 0x526834f1 <JSObject>
1|poolController            |     3: match [0x21cf30e5](this=0x6e4d3989 <String[18]: 01-NLS: 1587987243>,0x462b4611 <JSRegExp <String[#17]: ^([^:]+):\s*(.*)$>>)
1|poolController            |     4: _getHeaders [0x5266a4cd] [/home/pi/nodejs-poolController/node_modules/node-ssdp/lib/index.js:~482] [pc=0x22e68afc](this=0x379aaaf5 <SsdpServer map = 0x5013c45d>,0x6e4d3...
1|poolController            | FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
1|poolController            | Writing Node.js report to file: report.20200529.155643.378.0.001.json
1|poolController            | Node.js report completed
1|poolController            | Aborted
1|poolController            | npm
1|poolController            |  ERR! code ELIFECYCLE
1|poolController            | npm
1|poolController            |  ERR! errno 134
1|poolController            | npm
1|poolController            |  ERR! nodejs-poolController@6.0.0 start: `npm run build && node dist/app.js`
1|poolController            | npm ERR! Exit status 134
1|poolController            | npm ERR! 
1|poolController            | npm ERR! Failed at the nodejs-poolController@6.0.0 start script.
1|poolController            | npm
1|poolController            |  ERR! This is probably not a problem with npm. There is likely additional logging output above.
1|poolController            | npm ERR! A complete log of this run can be found in:
1|poolController            | npm ERR!     /home/pi/.npm/_logs/2020-05-29T19_56_43_577Z-debug.log
PM2                         | App [poolController:1] exited with code [134] via signal [SIGINT]

tagyoureit, 2020-05-30:
Interesting... I haven't seen this.  Node_SSDP is a 3rd party module that's included.  Are you able to attach the referenced file `report.20200529.155643.378.0.001.json`?  

If it keeps happening, you can turn off SSDP in `config.json` and hardcode the IP in your client(s).

jarrah42, 2020-05-30:
[report.20200529.155643.378.0.001.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/4705919/report.20200529.155643.378.0.001.json.txt)

jarrah42, 2020-05-30:
It happened again, but this time in a different library.

|poolController            | <--- Last few GCs --->
1|poolController            | [9287:0x27c1d08] 72380557 ms: Mark-sweep 205.6 (232.6) -> 205.0 (232.6) MB, 2775.1 / 0.2 ms  (average mu = 0.170, current mu = 0.013) allocation failure scavenge might not succeed
1|poolController            | [9287:0x27c1d08] 72383403 ms: Mark-sweep 205.9 (232.6) -> 205.2 (234.6) MB, 2535.3 / 0.3 ms  (average mu = 0.141, current mu = 0.109) allocation failure scavenge might not succeed
1|poolController            | <--- JS stacktrace --->
1|poolController            | ==== JS stack trace =========================================
1|poolController            |     0: ExitFrame [pc: 0xe19c00]
1|poolController            | Security context: 0x58f834f1 <JSObject>
1|poolController            |     1: once [0x58fa0521] [events.js:~303] [pc=0x6d0cb374](this=0x2efb9265 <EventEmitter map = 0x563bc86d>,0x3eb2d0a1 <String[#8]: readable>,0x2f099ab5 <JSFunction (sfi = 0x4bc3d6bd)>)
1|poolController            |     2: once [0x57ee12a5] [/home/pi/nodejs-poolController/node_modules/@serialport/bindings/lib/poller.js:~48] [pc=0x6d0cb038](this=0x2efb9265 <EventEmitter map = 0x563bc86d>,0x3eb2d0a1 <...
1|poolController            | FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
1|poolController            | Writing Node.js report to file: report.20200530.151858.9287.0.001.json
1|poolController            | Node.js report completed
1|poolController            | Aborted
1|poolController            | npm
1|poolController            |  
1|poolController            | ERR!
1|poolController            |  
1|poolController            | code
1|poolController            |  ELIFECYCLE
1|poolController            | npm 
1|poolController            | ERR! errno 134
1|poolController            | npm
1|poolController            |  ERR! nodejs-poolController@6.0.0 start: `npm run build && node dist/app.js`
1|poolController            | npm 
1|poolController            | ERR! Exit status 134
1|poolController            | npm ERR! 
1|poolController            | npm ERR!
1|poolController            |  Failed at the nodejs-poolController@6.0.0 start script.
1|poolController            | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
1|poolController            | npm ERR!
1|poolController            |  A complete log of this run can be found in:
1|poolController            | npm ERR!     /home/pi/.npm/_logs/2020-05-30T19_18_58_261Z-debug.log
PM2                         | App [poolController:1] exited with code [134] via signal [SIGINT]

jarrah42, 2020-05-30:
If it's a memory leak, then I presume any component that requests memory may fail.

tagyoureit, 2020-06-30:
Assuming you aren't seeing this anymore.  If it happens again, please re-open the issue.

closed 2020-06-30 by tagyoureit