tagyoureit/nodejs-poolController #172: Pump status not accurate

gadget-monk, 2020-05-31:
### Expected behavior and actual behavior
Pump status show be On when pump is running. RPM and Watts are accurately displayed. Status is also accurate when priming. 

### What is actually happening (screenshots are helpful)
Pump status is shown as Off

### Steps to reproduce the problem
Press "On" under Bodies->Pool to turn on pool pump, observe Status in web UI

### Specifications

1. What version of the code are you using? 6.0.0 (Next branch, updated 5/30)
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment
[Pump-on-off capture.zip](https://github.com/tagyoureit/nodejs-poolController/files/4706937/Pump-on-off.capture.zip)

tagyoureit, 2020-05-31:
Hi,
I'm able to replicate this and sure enough your pump is reporting 0 as the status.  Every single other pump I've seen reports the status as 1-16.  It looks like you have a VSF pump, but exactly what model is it?  I'm also seeing some other modes/values that I haven't seen before.

And also, are you turning off your pool at the end of the packet capture?  I know you said you saw some issues with the pool button not being synced with the circuit button but wasn't sure if that was related here.  

And finally, when you turn your pool circuit off, does your pump still have power?  EG if you go to the control panel on the pump is it still displaying data?  Or does power to the pump get cutoff when you turn off the pool circuit.

thx.

gadget-monk, 2020-06-01:
I have a [Sta-rite Intellipro VSF pump](https://www.pentair.com/en/products/pool-spa-equipment/pool-pumps/intellipro_vsf/sku/013004.html) (SKU 013004 on Pentair website). Not sure if this is related, but I also see a recurring message: warn: Message aborted after 4 attempt(s): 16,2,80,0,0,98,16,3: 16,2,80,0,0,98,16,3. I used to see a similar message in the old version too, but it didn't get in the way of anything so I ignored it. 

I also see this message in the logs (this is new in the 6.0 version):
Failed:{"bytesReceived":497427,"success":13825,"failed":13,"bytesSent":8976}

Also, not sure if it matters, but I don't have the normal Easytouch panel, but its a Compool to Easytouch upgrade kit. 

I did turn off the pool towards the end of the packet capture. I just checked the pump and when even off there is a display on it that says "Not Active!"

tagyoureit, 2020-06-07:
> Message aborted after 4 attempt(s): 16,2,80,0,0,98,16,3: 16,2,80,0,0,98,16,3.

This shouldn't be recurring.  The message should try 4 times, not find the chlorinator and stop.  Is it happening more than that?

> Failed:{"bytesReceived":497427,"success":13825,"failed":13,"bytesSent":8976}

This is a normal part of RS485 communications - packet collisions.  It's nothing to worry about and you have a very low rate of failure.

> Also, not sure if it matters, but I don't have the normal Easytouch panel, but its a Compool to Easytouch upgrade kit.
Don't think it matters.

> I did turn off the pool towards the end of the packet capture. I just checked the pump and when even off there is a display on it that says "Not Active!"
Ok, that's good.  It means that it is still has power and is wired correctly.

I'll ask in the forums if anyone else has a Sta-Rite pump and is seeing the same thing.

gadget-monk, 2020-06-08:
Thanks, i keep restarting to pull the new code daily which is why I see it in the log, I will keep a closer watch to see if it comes up more than 4 times.

gadget-monk, 2020-06-08:
I checked and the 16,2,80... message doesn't come up often, but there is another message about Pump 1 rejecting the command  [165,0,16,96,255,1,8,2,29] that comes up a few times

tagyoureit, 2020-06-12:
I just put a quick fix in the code for this.  Please test it out and LMK.  It looks good per your packet capture you submitted.

closed 2020-06-12 by tagyoureit