tagyoureit/nodejs-poolController #173: Intellicenter i10ps being recognized as Intellitouch i5+3S

cmc0619, 2020-06-10:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4755974/replay.zip)
### Expected behavior and actual behavior
Intellicenter i10ps being recognized as Intellitouch i5+3S
### What is actually happening (screenshots are helpful)
Intellicenter i10ps being recognized as Intellitouch i5+3S
### Steps to reproduce the problem
npm start
### Specifications

1. What version of the code are you using?
6.0 latest code as of 6/9/2020 - 11:07pm Eastern
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

cmc0619, 2020-06-15:
[replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/4777629/replay.1.zip)

We removed the intellivalve from the bus since we don't know the protocol for that yet.  Shortened the length between the pi and the Intellicenter from 150 ft to 3ft and took another capture.  Still showing Intellitouch atm moment.  The capture is attached.

rstrouse, 2020-06-15:
Out of curiosity which USB to RS485 dongle are you using?

cmc0619, 2020-06-15:
https://www.amazon.com/GearMo-Converter-Terminals-Windows-Support/dp/B005CPI0JQ

It's FTDI based.

rstrouse, 2020-06-15:
Ok now that I have gone through the replay I see that the fw version is probably 1.047.  This was an unexpected signature and have added it to the list of expected signatures.  It should now detect IntelliCenter instead of IntelliTouch.

Thanks for the info on the RS485.  I have ordered an FTDI based adapter as the one I am using is CH341 based.
[USB RS485 Adapter](https://www.amazon.com/gp/product/B00NKAJGZM/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

cmc0619, 2020-06-17:
Success! Sort of... the startup message on console now shows Intellicenter.  But dashpanel still shows a blank page with the empty bubbles and webclient flashes the display for a second, just long enough to see it, and then the screen goes blank.  Tried it in Chrome and IE, same effect both times.

rstrouse, 2020-06-17:
That is good news.  I have added some additional features to the dashpanel.  I am sure there is a change with 1.047 that I don't know about so if I can get you to click the menu bars at the top left of the screen then choose the logging tab and press the capture replay button.  On the dialog simply press Begin Capture.  Let it run for a couple of minutes or until the status at the top right changes from Loading to Ready.  At when the message changes to Ready press the Cancel Capture button and a zip file will download to your browser.   

This will contain all the output from IntelliCenter and allow me to see what has changed with 1.047.  If you close the Settings window you can open it back up again to press the Cancel Capture.  Do not leave this running as the files will grow to be very big after a while.  Take the file that has downloaded and upload it here.
[image]

rstrouse, 2020-06-20:
Any luck getting the files.

cmc0619, 2020-06-20:
No.  Tomorrow for sure!  It's not actually my IC.  I'm just acting as a tech liaison for a friend who has one so we can get him going with node-red and nodejs-poolcontroller.  I was smart and rolled my own from the beginning! :) https://www.youtube.com/watch?v=JXHMt6ylWnM

Question though.  Will I be able to see that screen with the capture button?  To date we've only ever gotten empty bubbles in dashPanel.  When I update the code I do the following.  Sanity check?

git pull (usually see some changed files)
npm install
npm start

That's should be all there is to get a new version right?

rstrouse, 2020-06-20:
No worries.  You should it will be the bars on the top left of the header.

Update both poolController and dashPanel.  Start poolController first then dashPanel.  The steps for update sound about right.  Also, are you installing both poolController and dashPanel in the same machine.  Make sure you have the ip addresses set up for poolController and dashPanel server.

thumbnut, 2020-06-20:
I was out of town all day. Will try downloading in the morning.

Thanks

Steve

cmc0619, 2020-06-21:
[replay (2).zip](https://github.com/tagyoureit/nodejs-poolController/files/4808815/replay.2.zip)

I changed the IP address (was using loopback) to the ethernet IP in config.json and it all popped up!  Was able to see everything.  Here's the replay with firmware 1.047

cmc0619, 2020-06-21:
So far we're seeing Chemistry and Heater are empty.  Everything else looks good.

rstrouse, 2020-06-21:
Awesome.  The chemistry, heater, remotes, and covers configuration interface is still in the works.   If you have a heater or a chlorinator it should be an option in the main panel If it is configured in IntelliCenter u can control it.  To set the heat setpoints click or tap on the body heat area. To change the setpoints on the chlorinator tap on it.

tagyoureit, 2020-07-05:
Please reopen or start a new case.  Closing this as resolved.

closed 2020-07-05 by tagyoureit