tagyoureit/nodejs-poolController #175: Not detecting anything

baudfather, 2020-06-14:
### Expected behavior and actual behavior
Expected: auto-detect panel (easytouch) and all connected devices.  not detecting anything and going into error loop.  Last week's code detected everything with no errors. 

### What is actually happening (screenshots are helpful)

### Steps to reproduce the problem
clean install from latest build, deleted poolconfig files

### Specifications

1. What version of the code are you using? latest
[2020-06-13_17-30-47.zip](https://github.com/tagyoureit/nodejs-poolController/files/4775693/2020-06-13_17-30-47.zip)

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

baudfather, 2020-06-14:
After a few clean restarts, it picks up the easytouch, but nothing else that's attached (IC20 and VSF pump)
[2020-06-13_18-52-29.zip](https://github.com/tagyoureit/nodejs-poolController/files/4775756/2020-06-13_18-52-29.zip)

baudfather, 2020-06-14:
(side note: my chlorinator is on but not reporting - bad sensor waiting on replacement)

rstrouse, 2020-06-14:
The byte rejection mechanism for the serial port was too aggressive and was forcibly rejecting bytes between buffer flushes,  This keeps a runaway serial port from simply filling up the message buffer,  This has been modified so that it only tosses bytes that can never be the start of another message.

baudfather, 2020-06-15:
Still no dice - a few less errors, but now it crashes during the detection process.
[2020-06-14_16-54-47.zip](https://github.com/tagyoureit/nodejs-poolController/files/4777437/2020-06-14_16-54-47.zip)

baudfather, 2020-06-15:
No crash now, but still not picking up the pump.  

[2020-06-14_19-56-51.zip](https://github.com/tagyoureit/nodejs-poolController/files/4777848/2020-06-14_19-56-51.zip)

baudfather, 2020-06-16:
Finds everything but the IC20 now, but getting a ton of 
`info: We have an incoming message but the serial port hasn't given a complete header.` 
not sure if that's normal, don't remember seeing it before - unless that's just a new label for the regular collisions. 
[2020-06-15_20-14-15.zip](https://github.com/tagyoureit/nodejs-poolController/files/4784196/2020-06-15_20-14-15.zip)

rstrouse, 2020-06-16:
Actually, that is the port reading a partial buffer.  In most instances it is simply recovering from the adapter sending only a few bytes at a time.  We used to wait to process the messages until we had a string of 10 bytes, but it turns out that in doing that you can get messages that are shorter than 10 bytes and short periods of no traffic that can cause the timing to be off from the port.  

In the future, we will make that message not appear but it is normal as what it is saying is that the serial port is sending data in very short bursts.  This is what was happening where we would reject messages that were on their way but hadn't received enough information to verify that it was coming.

tagyoureit, 2020-07-05:
Quite a bit of work was done on the comms.  Let us know if you are still having issues.

commit 1b4e2d71 (2020-06-14): Moved the byte removal functions as these originally were too agressive for slower serial ports.  Fix #175
closed 2020-07-05 by tagyoureit