tagyoureit/nodejs-poolController #179: Pool config not accurate

gadget-monk, 2020-06-14:
### Expected behavior and actual behavior
pool config should show reflect configuration on the outdoor control panel in EasyTouch. In my case, I have a "Gas Heater" and regular lights, but in the poolController's config, Solar Heater and an intellibright light group are both showing up as active. (i.e "IsActive=true")

### What is actually happening (screenshots are helpful)
See above

### Steps to reproduce the problem
See attached packet capture (line 382 and line 397 in poolConfig.json). I also ran /app/config/reload to force reloading the config. 

### Specifications

1. What version of the code are you using? latest from the next branch
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution)) attached.
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

[heater replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4777399/heater.replay.zip)

tagyoureit, 2020-07-04:
I pushed up a change to remove the Intellibrites when the circuit array is empty. 

On the solar, I am seeing your system tell me there is solar installed.  (I was checking this code for another reason last night.). Can you check your system setup to see if either solar or heat pump is enabled and it shouldn't be?  See the [manual](https://www.pentair.com/content/dam/extranet/product-related/product-manuals/pool-and-spa-equipment/easytouch/EasyTouch%208%20and%204%20Pool%20and%20Spa%20Indoor%20Control%20Panel%20Installation-Users%20Guide_English.pdf) page 9.  You can check under Settings > Solar.

gadget-monk, 2020-07-05:
Updated to the new code and don't see Intellibrite any more.. thanks! On solar, I am sure I have it disabled on my outdoor panel, but will confirm again today.

gadget-monk, 2020-07-07:
Confirmed that I had disabled solar on the outdoor panel
[image]

tagyoureit, 2020-07-07:
Ok - interesting... I have always seen a value of 21 (decimal) or 10101 (binary) when there are no solar/no heat pumps.  You have a value of 5 as the first byte in the payload:
`[165,1,15,16,34,3],[5,0,70],[1,53]`
I see a value of 23 (10111) when I turn on the solar.  I pushed the code with the updates and hopefully that is generic enough for everyone (we'll have to see if others start complaining).

Can you do me a favor and enable your solar (it won't hurt anything on your system) and see what the above packet changes to?  I would expect it to change to `[6,0,70]` (although the last bit will change if it sets the start/stop temp deltas).  Once you see the packet you can paste it here and disable solar.  It should be almost immediately after you make the switch.

gadget-monk, 2020-07-08:
where would I see this message? I tried the messageManager view in the dash panel but I don't see a message with a payload of 5, 0, 70. Am I looking for these number as part of a larger payload? 

[image]

gadget-monk, 2020-07-08:
Attached is the packet log with solar enabled. I see [7, 0, 70]. 
[packetLog(2020-07-06_11-30-18).log](https://github.com/tagyoureit/nodejs-poolController/files/4888462/packetLog.2020-07-06_11-30-18.log)

gadget-monk, 2020-07-08:
And it switches back to 5, 0, 70 once I disable solar. 
[packetLog(2020-07-06_11-30-18).log](https://github.com/tagyoureit/nodejs-poolController/files/4888484/packetLog.2020-07-06_11-30-18.log)

gadget-monk, 2020-07-08:
I guess this means solar is indicated by the second bit. this will fit the pattern for both values you see (21 = 000101**0**1, 23 = 000101**1**1; 5 = 000001**0**1, 7 = 000001**1**1)

--

--

gadget-monk, 2020-07-10:
I installed this update, but now I don't see the gas heater also (which showed up as id=1 earlier)

tagyoureit, 2020-07-10:
This is just a refresh issue upon initial load.  If you reload the -webClient it will appear.  I also just uploaded a fix for this.

closed 2020-07-10 by tagyoureit