tagyoureit/nodejs-poolController #181: Where to connect the RS485 adapter to an EasyTouch?

jgrn307, 2020-06-19:
I'm eager to try this out, but I'm not super familiar with the EasyTouch circuitry -- I ordered the recommended RS485 adapter, and I have a Raspberry Pi I'm planning to hook it up to, but my question is: where do I physically hook it up to?  Does someone have some photos of the hookup?

gadget-monk, 2020-06-19:
I have a compool to easytouch upgrade board so it may look a bit different, but look for a COM port - usually a 4 wire connector (Yellow = Data+, Green = Data-). The COM port is also where you connect an indoor controller or Intelliflo pump. I used the other two wires have power on them (red is 15VDC and black is Ground). I connected [this](https://www.amazon.com/gp/product/B01NALDSJ0/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) buck converter to the red/blk wires to provide power to the Raspberry Pi.

tagyoureit, 2020-06-23:
It will be the same.  Pretty much anywhere you see the RS485 wires as gadget-monk indicated you can attach your D+/D-.

closed 2020-06-23 by tagyoureit