tagyoureit/nodejs-poolController #182: NEXT will not start (crashed with error message)

cs9p, 2020-07-03:
### Expected behavior and actual behavior
nodejs-poolController will start and run.  'npm start' fails with error message.

### What is actually happening (screenshots are helpful)
See attached file _shell_capture.txt_ It looks like _sys.equipment.model_ is not being defined before it is used, causing the program to terminate.
[shell_capture.txt](https://github.com/tagyoureit/nodejs-poolController/files/4867173/shell_capture.txt)

### Steps to reproduce the problem
Clean install of nodejs-poolController NEXT
npm i
npm start

### Specifications

1. What version of the code are you using?
NEXT 6.0.0 7/2/2020 at 10:11 - also tried older version 6/15/2020 at 17:32 with same result
Node.js 11.15.0 - also tried 10.x and 12.18.2 with same result

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
Attached - the program crashed before it could create the zip file, but I manually found log files
[consoleLog(2020-07-02_16-11-06).log](https://github.com/tagyoureit/nodejs-poolController/files/4867168/consoleLog.2020-07-02_16-11-06.log)
[packetLog(2020-07-02_16-11-06).log](https://github.com/tagyoureit/nodejs-poolController/files/4867169/packetLog.2020-07-02_16-11-06.log)

I wonder if it might be because I have an older pump.  The easytouch can control this pump, but not newer controllers. I don't have a chlorinator, even though there is some kind of chlorinator packet captured. I am a new user and have never had a previous version working.

3. Any errors that are in the console
See attached file _shell_capture.txt_

4. Your config.json or in-use configuration file
[config.json.txt](https://github.com/tagyoureit/nodejs-poolController/files/4867174/config.json.txt)

5. Your pool equipment
Easytouch 4 pool/spa with 2 valves
Intelliflow VF pump
Booster pump relay

cs9p, 2020-07-03:
Added a log statement and it seems that function _initTouch_ in  file _controller/comms/messages/status/EquipmentStateMessage.ts_ is being called with model1=6 and model2=13.  This explains the error message.  I will try to add a case statement to make this identical to model1=2 and model2=13 and see how that works.  

I don't understand the code or even node.js so please excuse my attempt to debug if it is off base.

tagyoureit, 2020-07-05:
I'm only seeing 3 packets in the packet capture.  Can you try again and start the [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) using the CLI method before you start the app?

I though I had captured all of the Easytouch 4/8 controller models but possibly one is missing.  Thx.

cs9p, 2020-07-05:
Packet capture attached.  Adding _case 6:_ for model1 makes things (mostly?) work.  I do have a repeatable error message on the console, which strangely doesn't show when I turn on replay logging:
`> nodejs-poolController@6.0.0 start /home/pi/nodejs-poolController
> npm run build && node dist/app.js

> nodejs-poolController@6.0.0 build /home/pi/nodejs-poolController
> tsc

Init state for Pool Controller
[32minfo[39m: Server is now listening on 0.0.0.0:4200
[32minfo[39m: Serial port: /dev/ttyUSB0 opened
[32minfo[39m: Serial port: /dev/ttyUSB0 request to open succeeded without error
[32minfo[39m: Found Controller Board EasyTouch2 4
[32minfo[39m: Requesting easytouch configuration
Failed:{"bytesReceived":1363,"success":48,"failed":1,"bytesSent":576}
[32minfo[39m: EasyTouch system config complete.
[32minfo[39m: Requesting easytouch configuration
Failed:{"bytesReceived":40326,"success":1243,"failed":2,"bytesSent":684}
Failed:{"bytesReceived":41059,"success":1270,"failed":3,"bytesSent":996}
[32minfo[39m: EasyTouch system config complete.`

I'm not sure if this is important.  I'm also getting _Unspecified Error 12_ for the Intelliflow VF pump status.  Perhaps a different issue. Craig

[2020-07-05_15-18-05.zip](https://github.com/tagyoureit/nodejs-poolController/files/4875843/2020-07-05_15-18-05.zip)

cs9p, 2020-07-09:
I managed to find the error with the replay logging on.  I turned on putty logging - the Failed message appears on line 196 of putty.log.  This is the same place as line 184 in the consoleLog file in the replay zip file, although the Failed message is missing from that file. Interesting part from putty.log below

verbose: Config Queue Completed... 84% (8 remaining)
info: {"id":88,"valid":true,"dir":"out","proto":"broadcast","pkt":[[],[255,0,255],[165,1,16,33,225,1],[0],[1,185]],"ts":"2020-07-09T15:11:38.780-0700"}
verbose: Wrote packet [255,0,255,165,1,16,33,225,1,0,1,185].  Retries remaining: 3
silly: Retrying outbound message after 1.037secs with 3 attempt(s) left. - 165,1,16,33,225,1,0,1,185
info: {"id":88,"valid":true,"dir":"out","proto":"broadcast","pkt":[[],[255,0,255],[165,1,16,33,225,1],[0],[1,185]],"ts":"2020-07-09T15:11:39.819-0700"}
**verbose: Wrote packet [255,0,255,165,1,16,33,225,1,0,1,185].  Retries remaining: 2
Failed:{"bytesReceived":1293,"success":46,"failed":1,"bytesSent":528}
warn: {"id":89,"valid":false,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,1,15,16,2,29],[15,18,0,0,0,0,0,16,240,9,47,127,215,244,255,253,2,0,0,0,0,0,106,96,6,13,3,36,255],[255,255]],"ts": "2020-07-09T15:11:39.867-0700"}**
info: {"id":90,"valid":true,"dir":"in","proto":"broadcast","pkt":[[255,255,255,255],[255,0,255],[165,1,15,16,33,4],[1,2,3,4],[0,244]],"ts": "2020-07-09T15:11:39.871-0700"}
verbose: Config Queue Completed... 86% (7 remaining)

[2020-07-09_15-11-28.zip](https://github.com/tagyoureit/nodejs-poolController/files/4899895/2020-07-09_15-11-28.zip)
[putty.log](https://github.com/tagyoureit/nodejs-poolController/files/4899899/putty.log)

rstrouse, 2020-07-10:
Did you guys figure this out?  The message you highlighted is likely a normal collision on the 485 bus.  From the packetLog it looks like it goes stable as soon as all the configuration commotion settles down.  It's pretty normal to get a collision exchanging the number of messages that occur when getting a configuration.  

The message that failed is because the OCP broadcast it's status at the same time your pump threw some bytes on the bus.  This happens when most often when said equipment is tardy in responding to a request.  There is no real moderator in the RS485 debate so sometimes things talk out of turn when trying to keep the conversation going.  poolController detected this condition and gave the equipment ~1sec to shut up before it retried its request.  A healthy bus has <.5% packet failure and it looks like you are at around .16%

I did add some code to the board detection so that it won't crash if it sees a board type that it doesn't understand.  The key is to make sure the configuration is correct maxCircuits, maxFeatures, maxValves... etc.

tagyoureit, 2020-07-10:
I just pushed up a quick fix for this.

> Easytouch 4 pool/spa with 2 valves

Can you tell me _exactly_ which model you have?  Your major model number comes under the "Easytouch2" series but I haven't seen the minor model (6) yet.  I already had an 'EasyTouch2 4' with minor model 2 so I just added this to that grouping.  

Pull the latest and it should all load up for you.

cs9p, 2020-07-10:
@tagyoureit I'll give it a try.  My model is: 522354 Pentair Easytouch PSL4 purchased 12/2019.  Thanks for the help.

tagyoureit, 2020-07-10:
Ahh, so the specs are different.  
I'm not sure what `Number of Colored Lights Supported: 3` means?  I think this would be only 3 intellibrites on the only 3 circuits you have free.

> Supports 4 operating speeds of 1 IntelliFlo® Variable Speed Pump, 3 colored lights and 2 feature circuits

Does this mean that you can only choose 4 speeds to run your variable speed pump?

I just pushed up a few more changes that shouldn't impact anything but limit the circuits/features/schedules to what the PSL4 can support.

cs9p, 2020-07-10:
I am only using 3 speeds, but I just checked the manual and it looks like 8 speeds, 2 pumps, 4 colored lights, 8 feature circuits

From the manual:
How many IntelliFlo pumps will EasyTouch support?
EasyTouch can support a total of two IntelliFlo (or IntelliPro) VF3050 or IntelliFlo VS3050 (or VSF+SVRS) pumps
in any combination with up to eight GPMs or RPMs per pump. For example pumps can be connected to
EasyTouch as follows:
• 1 IntelliFlo VF + 1 IntelliFlo VS (or VSF+SVRS)
• 2 IntelliFlo VF
• 2 IntelliFlo VS (or VSF+SVRS)

From the Lights screen you can manually switch all lights on or off,  and synchronize colored lights. Up to 4 lights can be independently controlled from the Lights screen. Each light requires a separate auxiliary relay circuit. Up to four lights can be assigned on each auxiliary circuit.

There are eight (8) “Feature Circuits” that can be used to control IntelliFlo pump speeds or valves actuators
for a spa spillway. Unlike an auxiliary relay circuit, a “Feature” circuit does not connect directly to a relay.
“Feature” are turned on and off from the control panel “Feature” circuit menu.

tagyoureit, 2020-07-10:
The [PSL4](https://www.pentair.com/en/products/pool-spa-equipment/pool-automation/easytouch_pl4_andpsl4poolandspacontrolsystems.html) brochure and manual have different specs.  There is the EasyTouch (4,4P,8,8P), EasyTouch2 (4,4P,8,8P) and now EasyTouch4 PS/PSL.  Those specs look like they are from the larger EasyTouch2 systems.

cs9p, 2020-07-10:
You're right. I was looking at the manual for the larger system.  Sorry

cs9p, 2020-07-11:
Preliminary testing is good.  I haven't been able to get the webclient to work (probably cockpit error on my end).  I can do a more complete test once I do.  I am concerned about line 209 in EquipmentStateMessage.ts

> if (sys.equipment.model.includes('p')){
            sys.equipment.maxBodies = 1; // All Ps are single body; exclude Spa
        }

Shouldn't that be a capital P and won't that break the spa for the PSL4?  Perhaps should be P and not S

tagyoureit, 2020-07-12:
Saw that, too.  I pushed up some new code.

cs9p, 2020-07-14:
Tested and works great! Thanks @tagyourit and @rstrouse for your help fixing this and to all for creating a very cool piece of code.  As far as I'm concerned, we can close this issue.

BTW
> **['S', 'P', 'D'].includes(sys.equipment.model.slice(-1))**

This works for all models anyone is currently using, but not for PL4

tagyoureit, 2020-07-14:
Good catch.  Added an extra couple lines of code for some nice spaghetti.  Glad it's all working now.

closed 2020-07-14 by tagyoureit