tagyoureit/nodejs-poolController #183: i5P Board Not Configuring Properly

Nomar383, 2020-07-03:
### Expected behavior and actual behavior
Installed nodejs-poolController and it pulls in the majority of the circuits and pumps, but appears as "Intellicenter undefined" and is missing at least 1 circuit.

### What is actually happening (screenshots are helpful)
https://imgur.com/a/IgGps6y

### Steps to reproduce the problem
Fresh install with a i5P board.

### Specifications

1. What version of the code are you using?
6.0.0 -- next branch -- latest

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))

3. Any errors that are in the console

N/A

4. Your config.json or in-use configuration file
{
  "controller": {
    "comms": {
      "rs485Port": "/dev/ttyUSB0",
      "mockPort": false,
      "netConnect": false,
      "netHost": "raspberrypi",
      "netPort": 9801,
      "inactivityRetry": 10,
      "portSettings": {
        "baudRate": 9600,
        "dataBits": 8,
        "parity": "none",
        "stopBits": 1,
        "flowControl": false,
        "autoOpen": false,
        "lock": false
      }
    }
  },
  "web": {
    "servers": {
      "http2": {
        "enabled": false
      },
      "http": {
        "enabled": true,
        "ip": "0.0.0.0",
        "port": 4200,
        "httpsRedirect": false,
        "authentication": "none",
        "authFile": "/users.htpasswd"
      },
      "https": {
        "enabled": true,
        "ip": "0.0.0.0",
        "port": 4201,
        "authentication": "none",
        "authFile": "/users.htpasswd",
        "sslKeyFile": "",
        "sslCertFile": ""
      },
      "mdns": {
        "enabled": false
      },
      "ssdp": {
        "enabled": false
      }
    },
    "services": {},
    "interfaces": {
      "smartThings": {
        "name": "SmartThings/Hubitat",
        "enabled": false,
        "fileName": "smartThings-Hubitat.json",
        "globals": {},
        "options": {
          "host": "0.0.0.0",
          "port": 39500
        }
      },
      "vera": {
        "name": "Vera",
        "enabled": false,
        "fileName": "vera.json",
        "vars": {
          "deviceId": 0
        },
        "options": {
          "host": "",
          "port": 3480
        }
      }
    }
  },
  "log": {
    "packet": {
      "enabled": false,
      "logToConsole": false,
      "logToFile": false,
      "filename": "packetLog",
      "invalid": true,
      "broadcast": {
        "enabled": true,
        "includeActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeActions": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "pump": {
        "enabled": true,
        "includeActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeActions": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "chlorinator": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "intellichem": {
        "enabled": true,
        "includeActions": [],
        "exclueActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "intellivalve": {
        "enabled": true,
        "includeActions": [],
        "exclueActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "unidentified": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "unknown": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      }
    },
    "app": {
      "enabled": true,
      "level": "info",
      "captureForReplay": false
    }
  },
  "appVersion": "6.0.0"

5. Your pool equipment
Intellicenter with i5P board. 1 pool, no spa. 1 waterfall feature. Popup cleaning heads.

rstrouse, 2020-07-03:
Ok so what I need you to do is click the three bars on the upper left corner of the dashPanel.  Choose the logging tab and press the Capture Replay button.  This will open a window.  Click the Capture Replay button.  After a couple of seconds the controller will begin loading information from the i5P when the indicator in the upper right goes from yellow to green press the Cancel Capture button.  This will download a file into your browser which I would like you to post back here.  

The problem is pretty simple in that we only have had access to i5PS, i8PS, and i10PS boards.  As a result we have not properly identified their signature.

Nomar383, 2020-07-03:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4867295/replay.zip)

Here you go!

rstrouse, 2020-07-03:
Thanks for that I have inspected the replay and reworked the code so that it detects the i5P.  Turns out Pentair simply identifies this by having no modules installed.  This also means that the circuit ids start at 2 and the first bit (which is normally spa) is ignored.  Update your poolController and delete the json files from the data directory. Then start poolController.  It should go through the detection process and pick up the i5P.

Nomar383, 2020-07-03:
So, this update removed the extra "bodies" and it looks closer to correct. However, it detected the board as a i5PS if that matters. Some of the circuits/schedules seem out of order and don't work. See the screenshot. I'll attach another replay as well.

https://imgur.com/a/yi8aDfc

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4870998/replay.zip)

rstrouse, 2020-07-04:
Thanks for that.  It appears the names were shifted 1 position.  Update your poolController and it should align the circuit names correctly.

Nomar383, 2020-07-04:
Just pulled the latest version and I'm getting an error on startup now:

`> nodejs-poolController@6.0.0 start /opt/pool/nodejs-poolController
> npm run build && node dist/app.js

> nodejs-poolController@6.0.0 build /opt/pool/nodejs-poolController
> tsc

web/services/state/State.ts:55:57 - error TS2339: Property 'setTempsAsync' does not exist on type 'SystemCommands'.

55                 let controller = await sys.board.system.setTempsAsync(req.body);
                                                           ~~~~~~~~~~~~~

Found 1 error.

npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! nodejs-poolController@6.0.0 build: `tsc`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the nodejs-poolController@6.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-04T18_53_11_147Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! nodejs-poolController@6.0.0 start: `npm run build && node dist/app.js`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the nodejs-poolController@6.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-04T18_53_11_210Z-debug.log`

rstrouse, 2020-07-04:
Pull again you got caught between two commits.

Nomar383, 2020-07-04:
Nice. Looks like everything is pretty much named correctly and is functioning when I use dashPanel. Only issue I see is that when I turn something on with dashPanel it turn green for a second, but doesn't stay lit up green. It works, but it doesn't reflect the status properly. Also, my IntelliChlor appears in dashPanel, but the name doesn't show up. See the attached picture.

[image]

rstrouse, 2020-07-05:
IntelliChlor will not report until IntelliCenter has cycled it on and it starts reporting its status.  This typically means that the pool circuit is turned on and IntelliCenter has had time to start interrogating the IntelliChlor.

I assume by your configuration you have no heater attached to the pool.  I think that when solar doesn't exist it always reports 100 degrees so I probably need to not show the solar temp in that condition.  It also means that there is no setpoint for the pool circuit or heatmode.

Please pull the latest nodjs-PoolController and dashPanel.  After you have updated it let's do a little troubleshooting.  When we get the status to report from the i5P correctly it will likely fix homebridge and dashPanel.

On the status I can't imagine that there is some other place where the circuit state is being reported but it is possible.  So to find this out, we need to do some troubleshooting.  Follow the steps below carefully so I don't have a ton of messages to evaluate.
### Step 1
Click on the menu bars on the top left, choose the logging tab and and capture a replay.  However, when you capture this replay I don't want you to include the configuration reload so after you press the Capture Replay button please uncheck the Capture Configuration Reload checkbox before you press begin capture.
[image]

### Step 2
After the capture has started please turn on the pool circuit.  You can do this from any controller including dashPanel since it is activating the circuits (clicking the big gray circle should turn it on).

### Step 3
Turn on the waterfall. 

### Step 4
Turn everything off again.

### Step 5
Return to the logging tab and press Cancel Capture.  This will download a zip file to your browser which I would like you to upload here.

Nomar383, 2020-07-05:
You are correct, I do not have a heater of any kind (here in AZ I would rather have a cooler on some days).

I have followed your instructions and attached the relevant replay. Thanks!

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4875728/replay.zip)

rstrouse, 2020-07-05:
In looking at the logs your chlorinator never responds back to the OCP.  In fact it has never responded to the OCP.  The OCP repeatedly asks for information from it and never gets a response.  Is it plugged into the bottom of the panel?  Also, I assume this to be an IntelliChlor not an iChlor.

In looking at the logs it is maintaining the state famously.  All information short of the Chlorinator is working like a champ.  Perhaps there is a browser error that we are not seeing.

I did notice one thing in your config.json.  change the https section so that it says enabled:false.  Let's make sure we are not getting traffic on server since you do not have a certificate authority assigned to it.  

Please provide the following information.
1. What browser are you using (e.g. Chrome, Safari...etc).  If you are using Safari please indicate whether it is iOS and what device it is on (e.g. iPad 2, iPad 3).
2. Post the config.json from dashPanel.  This looks like the socket that sets the data is not being initialized.  If this is not correct it will make homebridge not work properly either.
3. Do you have an ip reservation set up in your DHCP server for the raspberry Pi?  If you do not then you probably should we will try the ip address instead of the DNS name for the pi which i assume is RASPBERRYPI.

Also, post up the config.json from the dashPanel directory.  This l

Nomar383, 2020-07-06:
1. Using Chrome on both OS X and Windows. Haven't tried iOS. I don't see any errors in the Chrome Javascript Console when changing things or refreshing dashPanel.

2.  [config.txt](https://github.com/tagyoureit/nodejs-poolController/files/4876697/config.txt) It's a json file, just wouldn't let me upload it without changing the file extension.

3. Static IP (not using DNS or DHCP reservations) set on the raspberrypi = 10.0.0.101

4. It's an Intellichlor. I seems to be connected properly to the Intellicenter since I can go out to the physical panel and bump the output up/down and see the changes on the actual Intellichlor unit reflecting what I select on the Intellicenter panel. When I turn on pool mode, it seems like dashPanel can tell the Intellichlor is on and even what it is set to. Unless I'm missing something? It just isn't pulling in a name for it. It actually will show the salt output too after a little while in pool mode. Showing 2850ppm at the moment. See the picture below:
[image]

rstrouse, 2020-07-06:
I traced all the way through the messages in the log everything looks clean.  Do you see socket emits in the console?  There should be one at least every minute and one for each change to the screen  Temps, rpm, watts... etc.

rstrouse, 2020-07-07:
Out of curiosity, what version of node are you running.  Type the following command to get the answer.  
node -v

Unfortunately the 40% on the chlorinator happens to be us scraping the output messages from IntelliCenter OCP.  We should probably wait until IntelliChlor responds with a (18) message.  This is where we are getting the cell active status.  We are doing this because we have experienced some chlorinators not responding at all with a current level.  Some of these differences are differences between the age of the IntelliChlor cell itself.  I am running an old cell but have one at the ready when it stops producing.

The 2,850 is clearly one instance where the chlorinator responded with the status message (18).  However, IntelliChlor takes some time to respond to the requests for the model name.  This comes in a message that periodically gets sent from IntelliChlor and it will contain something like IntelliChlor--40 or IntelliChlor--60.  It will likely eventually fill in that name.  This happens after the pool circuit is turned on.  The flow stabilizes enough so that the cell thinks it has enough flow and it hasn't exceeded its 24 hour % of on time.  It can be a little weird but after we get the virtual chlorinator controller stabilized we may just bypass IntelliCenter's long wait time to ask for the name of the cell.  Pentair doesn't really show that name anywhere, it just tags it with IntelliChlor or generic Chlorinator when it runs into AquaPure.

Nomar383, 2020-07-07:
So, for the few circuits that worked before we started, it actually maintained the status in dashPanel. I would turn something on and it would stay green as expected. Since this problem seems to be the same in dashPanel and the Homebridge plugin, I would think this is related to the core nodejs-poolController.

When I turn a circuit on I see a Javascript ON event, then immediately I see an OFF event:

{evt: "circuit", data: {…}}
data: {id: 4, isOn: true, name: "Waterfall", showInFeatures: true, type: {…}}
evt: "circuit"
__proto__: Object

{evt: "circuit", data: {…}}
data: {id: 4, isOn: false, name: "Waterfall", showInFeatures: true, type: {…}}
evt: "circuit"
__proto__: Object

I do see events every minute in the Javascript console. I turned the Waterfall on about 20 minutes ago and I didn't see another circuit event for the waterfall after the 2 I posted above. I can see periodically it checks circuits 3 and 5, but not 4. Not sure what the expected behavior is here. 

Also, when a feature is on I can see the pump is on with a green ball on the left. It also properly shows schedules as active with a green ball correctly. So, this seems to be an issue with circuits and the pool mode.

My node version is v12.16.3

Here is an example of some of the console output after I turned the Waterfall on:
[image]

tagyoureit, 2020-07-07:
@Nomar383 
Can you try a hard reset on your OCP?  I have an IntelliTouch but have seen it where the OCP basically rejects any commands I send it.  In my case, I can’t even turn off the pool circuit from the panel. 

Once I do a reset, it behaves itself. Wondering if there is some leftover coffee or similar bug in the IntelliCenter.

rstrouse, 2020-07-07:
With what you are reporting there we are getting a circuit message from somewhere.  Can I get you to update dashPanel and poolController, then do the exact process that resulted in the console log above (I assume this is simply turning on the waterfall) with the capture running?  The only thing that can be happening here is that something is sending back an action(2) message with the 4th bit unset on payload(2).  Don't forget to uncheck the Capture Configuration Reload.

Just a little technical background.  Only equipment where the state has changed should be emitted on the websocket which is what you are seeing in the browser console.  So the temps message is probably because the Az sun is heating up and the temp changed.  The controller messages you see are when the clock clicks to the next minute or the status changes on the top bubble.   Finally, the pump messages are likely related to wattage changes since that bounces around the most.

So what shouldn't be happening is that circuit dance and this only comes from a single byte located on a message sent from the OCP.  The previous replay you provided verified that the pool is actually on bit 6 (just like PS models) but unlike PS models it sets bit 1 as well which is normally the spa.  This is what the OCP reports when you turn on the pool.
[image]

And when you turn on the waterfall bit 4 should also be set and remain set until it is turned off.  This message is sent every 2 seconds from the OCP onto the 485 bus as a broadcast.

Nomar383, 2020-07-07:
Okay, so the first capture I hit the Waterfall button and it actually didn't even turn on the Waterfall at all.

Replay: [replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4887548/replay.zip)

Screenshot:
[image]

On the second try it did turn on the Waterfall, but the status did not reflect in dashPanel (same problem I've had to this point.

Replay: [replay (1).zip](https://github.com/tagyoureit/nodejs-poolController/files/4887556/replay.1.zip)

Screenshot:
[image]

rstrouse, 2020-07-07:
Yeah IntelliCenter does sometimes throw away a command.  Not sure why, but I have experienced it from both the indoor panel and the wireless controller.  You click on it and it simply changes back.  If you click it again and it works.  There was a similar thing in IntelliTouch so I don't know if this is SOP in Pentair.  It is much better than the Jandy PDA though you had to peel a banana upside down and hold your leg at 26.3 degrees to get it to work consistently.

Thanks for that, I'll have a looksee.

rstrouse, 2020-07-08:
You know what.  I just noticed something odd about your setup.  I assume your waterfall is actually a waterfall not a spillway.  I know this because a spillway circuit is a special kind of circuit for shared pool/spa combinations where the intake and return valves are set such that water is drawn from the pool and returns to the spa.  Hence spilling over the spa and into the pool.  I think you need to set your waterfall circuit to generic.  This circuit type also coordinates fill/drain for champagne edge spas.  I don't know what IntelliCenter does with this type if the intake/return valves are not present like they are for pool/spa combos.

Still going through the message output though.  poolController shouldn't be sending that second emit where isOn=false.  The bit 4 is sticking when you select waterfall regardless of the circuit type.  Somewhere along the line poolController is emitting an off state even though it is on. 
[image]

rstrouse, 2020-07-08:
Alright pull the latest poolController.  I am confident that I found the issue.  Turns out it was really simple and when you stare at something long enough it will jump out at you.  The byte alignment started on byte 3 so the initial set responded with success which changed the state then the 2 message came along and set it to off.  The reason was that it was looking at byte 3 instead of 2 for single body systems.

Nomar383, 2020-07-08:
I updated everything and the first time it did the same ON/OFF sequence and didn't actually turn the Waterfall ON. However, on the second try it worked and now stays lit up green! This also means I can turn it OFF from dashPanel. The third try also worked and responded as expected.

Weird it seems the first try doesn't work, but I'm not sure how to address that. This is definitely better and works 99% correctly now. I'll try out the Homebridge plugin tomorrow and see if that also has an issue on the first command. I noticed this problem with the Homebridge plugin the past few days as well, so not sure it's a dashPanel issue.

rstrouse, 2020-07-08:
It's not a dashpanel issue or a poolController issue.  I can see it happening in the messages you sent and it is an IntelliCenter issue.  IntelliCenter simply ignores it sometimes.  Don't quite know what it is but the same thing happens on the indoor and mobile panels sometimes.  You likely won't find rhyme or reason to it.  It just acknowledges the command and does nothing with it.  I have thought about re-issuing the command if I check it later and it isn't set but that makes it feel sluggish.

Also with the dashpanel up and running lets see if we eventually get an IntelliChlor name.  This will be when your pool circuit has been running for a while.  It will first send the salt level then it will send the name.

Nomar383, 2020-07-08:
Do I need to have dashPanel open while the pool is running to get it to
pull in the name?

rstrouse, 2020-07-08:
Nope poolController should pick it up and dashPanel will load it up the next time you open it.  poolController does need to be running though because IntelliCenter doesn't broadcast this with the configuration.  We get it by listening in on the conversation between IntelliCenter and IntelliChlor.

Nomar383, 2020-07-08:
I left poolController running overnight while the pump was running for several hours and it never pulled in the name of the intellichlor. It's actually blank in dashPanel under chlorinators. If I manually enter the name there, it still doesn't show up in the main dashPanel display. Pretty minor issue overall. Is there anyway to totally disable/remove anything related to heaters?

rstrouse, 2020-07-08:
Yes I am currently working on making the heat options go away when there is no heater installed.  Stay tuned.  The solar temp should also go away when there is no solar when I am done.

I think there is some change with the chlorinators that newer versions might not emit their model through the channels that we have identified.  From the replay I can see IntelliCenter asking the IntelliChlor for information that it promptly ignores.

The manual name isn't currently being saved.  This option was previously, read only but I am going to make it save to the config and state.

tagyoureit, 2020-07-08:
I just wonder if IntelliCenter never uses the Chlorinator name maybe it won't ever ask for it (or only asks for it upon startup).  

Here are directions to manually ask for the Chlor name.  If this works we can easily add it into the code.

Load up @rstrouse's [messageManager](http://localhost:5150/messageManager.html).   Click the Hamburger menu (top left) and only enable chlorinator messages (so you don't see thousands of messages). Also click the 3rd button highlighted.  It should be purple so it turns on the logging.

[image]

Then click add Message.
[image]

Fill in the following values as seen in the pic.  Then click save message.
[image]

Finally, hit the "Send Queue" button.  I expect to see the outgoing message and name in the response.  
[image]

Let us know.

tagyoureit, 2020-07-08:
> Yes I am currently working on making the heat options go away when there is no heater installed. Stay tuned. The solar temp should also go away when there is no solar when I am done.

I was going to look at removing the appropriate virtual circuits that don't apply for solar or no heat, but if you were going to address that as part of your fixes I'll hold off.  Or if you want me to add it to my list LMK.

rstrouse, 2020-07-08:
Yes already in the throws of engagement.  I love the smell of codepalm in the morning.

Nomar383, 2020-07-08:
I don't know how many messages should be going back and forth between the Intellicenter and the IntelliChlor or poolController, but the chlorinator messages are basically continuous if that makes any difference. I don't even have any equipment on and there are tons and tons of chlorinator messages passing around.

It doesn't look like your example after I sent the command:
[image]

I tried both with the pump/equipment running and also without.

Nomar383, 2020-07-08:
Couple more odd things I noticed that are easily duplicated every time:

1) The button in dashPanel actually won't turn the "Pool" on or off. It will maintain status properly if I turn it on/off from the official Pentair app or the physical panel.

2) If the "Pool" is ON and I subsequently turn ON the Lights in dashPanel, the Lights will turn on as expected (and maintain status as expected), but the "Pool" actually turns off and dashPanel shows it as OFF too. This happens 100% of the time.

rstrouse, 2020-07-08:
The key is that the filter circuit must be running so the pool circuit must be on for the IntelliChlor to respond correctly.  In fact, if the load center is wired as recommended by Pentair, the IntelliChlor transformer has no power until the pool filter circuit is turned on.  

The way this is done is that the power for it is tapped off the switched side of the filter pump relay.
 
Another thing to double check is that in IntelliCenter you have the body selected to Pool and you have a % set for the output.  In that message chain IntelliCenter is definitely telling IntelliChlor that it is in control and if you are awake you need to set your output to 0%.  There are no responses coming from the IntelliChlor the source is always OCP and the destination is always the chlorinator.  This is what you get when the chlorinator is not responding.

rstrouse, 2020-07-08:
Interesting on the pool state.  I have the original capture you sent where you turned on the pool and the waterfall so I'll have a look at that.

rstrouse, 2020-07-08:
I found the reason why the circuit state is turning off.  There was one other place where the bytes are shifted 1 for single body systems.  It happens to be on the message where we send the on/off command.  I'll have a fix shortly.

tagyoureit, 2020-07-08:
@Nomar383 To reiterate what @rstrouse said about the chlor... you have 100% packets going outbound to the chlorinator but nothing being returned.  The source/dest should be alternating every packet.  Can you make sure the pool is really on when you are watching the messages for the chlor?  Also, try changing the values in IntelliCenter and see if that helps.  And finally, go out and see what the lights on your IntelliChlor look like.  Can you change the value on the physical chlorinator by pressing the ^ and v buttons?

Nomar383, 2020-07-08:
The screenshot was with the pump and equipment all off. The response was similar when I turn the equipment on earlier and sent the same message. I will try again tomorrow morning when I'm home and the equipment has all been running for a while. Makes sense there is no response if the IntelliCholr is literally not even powered on lol

Nomar383, 2020-07-09:
Here is the screenshot after sending the message with all the equipment on and running:
[image]

You can see the messages are going back and forth between the Intellicenter and the IntelliChlor now that I am testing with the equipment actually running. The manual message you had me send didn't appear to make any difference to anything; still no name pulled in.

I updated poolController again and now the Lights can be turned on and off without impacting anything else. Sometimes it will look like the Popup Cleaners or the Pool will change status for a moment when I turn the Lights on/off, but they correct right away and it doesn't look like any equipment is actually turning on/off that shouldn't (just a graphical glitch I guess).

tagyoureit, 2020-07-09:
Strange.  The chlorinator should be responding with Action=3 but it is responding with Action=20.  Exactly what brand/model/make is your chlorinator?  

Have you ever seen the chlorinator correctly report its name?

rstrouse, 2020-07-09:
Actually, the OCP is sending action 20 over and over again but the chlorinator blows it off.  It looks like it only ever responds with an 18.  Clearly IntelliCenter isn't waiting on a response from IntelliChlor or the bus would be bound up like it just got back from a cheese eating contest.  It does consistently respond to a 17 request with an 18 response though.  Perhaps on newer models they fixed the comm lost alert.  That would explain the ```17 [57, 128]```.  This means 2,850 salt level and communication lost.  The message that is being sent from the OCP ```17 [40]``` means that it is setting the chlorinator to 40%.  I assume that is the current setting for the pool side of the chlorinator.

For this panel we have never seen it report its name.

Nomar383, 2020-07-09:
Are there different types of IntelliChlor? Looks like that all it says on the unit itself. 

It's never once reported the name in dashPanel properly. The salt level and percentage are totally correct though and I can adjust it from the Intellicenter panel and the physical IntelliChlor unit reflects the new changes right away.

rstrouse, 2020-07-09:
Yes there are.  In fact the model numbers remained the same over the years but the equipment has also changed.  Typically the model will be IC40 or IC60 if it is IntelliChlor.  iChlor has different model numbers.

There are also some generics out there.  When was the cell purchased?  Did it come with your IntelliCenter or was it added later?  Can you take a picture of the cell and post it here?  Usually, there are subtle differences on the control panel when the version changes.

rstrouse, 2020-07-09:
One other thing there is typically a sticker on Pentair gear that indicates its manufacturing date.  This will likely be on the bottom of the cell.

Nomar383, 2020-07-09:
Here are the pictures from the unit:

[image]
[image]
[image]

rstrouse, 2020-07-09:
When you turn it on, do you see any red or flashing lights?

Nomar383, 2020-07-09:
For the first couple minutes there is alternating Red/Green flashing lights. I assume that's a bootup sequence of some kind because it goes away pretty quick and I see the following:
[image]

rstrouse, 2020-07-09:
If when in this state can you change the output % or does IntelliCenter have a lasso on it.  It may flash the increase then set it back to 40% but IntelliChlor isn't responding even to the action 0 message which tells it to disable its control panel.  It's just very odd that it only ever responds to 17.  Bear in mind the only message sent by poolController in the listing from the messageManager was the one you sent by queuing it.  That is just a listing where we are listening in on the conversation.  Below is a typical conversation between IntelliCenter an the Chlorinator.  Notice how IntelliCenter sends an action and IntelliChlor responds with the corresponding answer.  

Your IntelliCenter is asking all the same questions it's just that IntelliChlor is not responding with the answer.  Perhaps these are getting clobbered by something.  Can I get you do do the same steps you did for @tagyoureit above but include the Invalid messages as well.  

If it is being clobbered you may have a loose wire in the connection from the port to the IntelliChlor control board.  If the wires aren't making great connection you can get instances were the communication on one side of the data line are scrambled.  This often happens because the pb stuffs all the comms into a single port or worse, wire nuts it all together then stuffs it into a single port.  The problem with that is that some of the wiring ends up loose in there or you get reflection from the difference in the wire gauges.  There are 2 ports on the main board and 1 on the IntelliChlor board so it is shocking to see pumps, IntelliChlor, and Control Panels all stuffed into the same J4.  Are you comforable enough to shut it all down and inspect the J4 and J5 port wiring?  If your pump is in the same port at the IntelliChlor this may work but it can cause weird comms problems.
[image]

tagyoureit, 2020-07-10:
It should never be flashing red/green for "a few minutes," a few seconds maybe but definitely not in the minutes range.  You may want to check the [manual](https://www.pentair.com/content/dam/extranet/product-related/product-manuals/pool-and-spa-equipment/intellichlor/IntelliChlor_Owners_Manual_Version_3_units_after_Nov_2011_English.pdf) to see exactly which sequence you have.  Maybe your cell needs to be cleaned?  Have you inspected it in a while?

tagyoureit, 2020-08-11:
Closing due to lack of activity.

closed 2020-08-11 by tagyoureit