tagyoureit/nodejs-poolController #184: Intellibrite control not loading into webclient

baudfather, 2020-07-06:
### Expected behavior and actual behavior
Unable to control intellibrite through webclient, previous builds worked.

### What is actually happening (screenshots are helpful) 
Intellibrite control displays in webclient (appears to be in poolconfig.json)

### Steps to reproduce the problem
Full wipe/install, delete data/pool*.json files

### Specifications

1. What version of the code are you using? latest as of issue date submitted
2. All output from the file log 
[2020-07-05_21-38-26.zip](https://github.com/tagyoureit/nodejs-poolController/files/4876800/2020-07-05_21-38-26.zip)

3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment Easytouch 4p, VSF pump, IC20, Intellibrite

tagyoureit, 2020-07-07:
Fixed.  I have one small issue I need to look at when the controller loads for the very first time (it will show no features) but just refresh the webclient and it will come back.

closed 2020-07-07 by tagyoureit

baudfather, 2020-07-09:
Awesome - working again!  Thanks! :)