tagyoureit/nodejs-poolController #187: Make config.json immutable

wurmr, 2020-07-08:
### Expected behavior and actual behavior

Application config should be user preferences and different from discovered state.

### Steps to reproduce the problem

Make config.json read-only to reproduce the problem.

### Specifications

1. What version of the code are you using?
`@Next`
3. Any errors that are in the console
```
 Error writing configuration file Error: EROFS: read-only file system, open '/usr/local/nodejs-poolController/config.json'
    at Object.openSync (fs.js:458:3)
    at Object.writeFileSync (fs.js:1355:35)
    at Config.update (/usr/local/nodejs-poolController/config/Config.ts:28:16)
    at Timeout._onTimeout (/usr/local/nodejs-poolController/config/Config.ts:70:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7) {
  errno: -30,
  syscall: 'open',
  code: 'EROFS',
  path: '/usr/local/nodejs-poolController/config.json'
```

### Why?

When trying to run the controller inside of a docker compose setting the config file passed in will be read only, this allows for easily reproducible containers.

### Alternate/Preferred Proposal

Provide a way for the application to start up with "suggestions" from the user that are used only as input, if the app still needs to write those to disk that is fine. Ideally this would be through environment variables.

Environment variables would make containerizing this application much simpler.

tagyoureit, 2020-07-08:
This would be a big lift/shift on our side.  We dynamically write to that file from the app and not sure we want to tackle changing that right now.

As an alternative, what you can try is this:
1. In docker image, delete the `config.json` and `defaultConfig.json`.
2. Copy the `config.json` to `defaultConfig.json` in the docker.
3. When the app starts, it will copy `defaultConfig.json` to `config.json`.

However, this may break if we add/remove new items to the `defaultConfig.json`.

emes, 2020-07-23:
Agree that it would be nicer if all of the "user" stuff were in one place, or via ENVs.  So maybe not ideal, but why not just use volumes in the compose file for those things which need frequent access?
```
volumes:
   - ${USERDIR}/poolcontroller/config.json:/usr/local/nodejs-poolController/config.json
   - ${USERDIR}/poolcontroller/data:/usr/local/nodejs-poolController/data
   - ${USERDIR}/poolcontroller/logs:/usr/local/nodejs-poolController/logs
```

rstrouse, 2020-07-27:
You know in this case it could be as simple as not persisting the config.json if it is read-only.  We could also allow a config-lock.json that you to config parameters in and we would simply merge them.  This would be a file that does not exist in the distro that contains attributes that are specific to the image.  The values defined in that file would overwrite any other data.

tagyoureit, 2020-11-02:
I think some subsequent threads have resolved this.

closed 2020-11-02 by tagyoureit