tagyoureit/nodejs-poolController #189: Add support for IntelliValve

theRealMCQwerty, 2020-07-18:
Continuation of closed issue https://github.com/tagyoureit/nodejs-poolController/issues/116

The IntelliCenter is supported but the IntelliValve is not. Or at least not supported via RS485 for incremental control of the valve actuator position.

@rstrouse has made some good progress but with the previous issue being closed there is nowhere to continue the discussion on this particular issue.

@rstrouse - do you have an update on current state? Is there anything I can do to help?

rstrouse, 2020-07-18:
Here is where we are up to with this and what we have learned thus far.
1. The Pentair IntelliValve messaging is completely unknown.  However, we do know that IntelliValve broadcasts an action 82 message.  The suspicion is that this message contains an address in at least the last 5 bytes of the payload.
```[255, 0, 255][165, 1, 16, 12, 82, 8][0, 128, 216, 128, 57, 64, 25, 166][4, 44]```.  There is one other known piece of equipment in the wild that broadcasts on the 12 address and this is the new iChlor cell.  While the message structure is different we suspect this is also a hail indicating to the OCP that it is available to be commanded.
2.  I have dubbed this message "I am Groot" since no matter what you try to say to the valve it responds with the identical message above.  The message has a unique set of values that are always the same and so far the payload is unique for every valve we have encountered.  The first 2 values always appear to be 0, 128 on the payload and it is always broadcast from address 12 to address 16.  Valves manufactured about the same time seem to have similar byte patterns in the payload which seem to indicate that this is a unique valve address.  How funny will that be when I am proven wrong after the fw is released.
3.  I have written a module in messageManager that responds to these incoming messages at proper intervals.  If you respond to the IntelliValve too often it will stop broadcasting it's "I am Groot" message mentioned above.  I am not sure why it does this but I am sure that it is not an indicator that it acknowledged a message that it was sent.  The reason is that the message patterns being sent do not matter in making it stop.  Using this module I have sent thousands of messages to the valve with varying payloads but only ever get "I am Groot", however, if I send an action 81 from 16 (OCP) to 12 with 0 payload it appears to make IntelliValve send its "I am Groot" immediately without the steady delay.  Continuing to send this message repeatedly will make IntelliValve go silent.
4.  You can view the chatter from messageManager by setting the settings on the logging tab to below and clicking on the Start/Stop log icon at the top of the list.  You will see "I am Groot" for every IntelliValve you have attached to the RS485 bus.
[image]
5.  If you want to send messages to your valves you can use the Send Message Queue box to either create a queue and send it or create your own testModule to send messages out to the valve.  I would suggest using only one valve and an example of the code is in scripts/messages/testModules/sendValveMessage.js.  The operative method there is sendNextMessage which is called at the appropriate interval.  Setting the msgToSend variable will send the message you constructed to the RS485 bus.  The hope is that the valve picks it up.  If there is a change to "I am Groot" it will automatically stop to allow you to investigate.

If anybody has any ideas or wants to help coding on this project let me know.  Given the sheer number of possible messages out there, sending random messages to the valves is probably an exercise in futility.  I really want to see an iChlor attached to IntelliCenter to see that communication.  Perhaps it will yield a nugget that will break this wide open.

gw8674, 2020-07-19:
r.,
So based on your outstanding write up of the current status of this project, would it be helpful (and feasible), for one to get a hold of an iChlor temporarily, connect it up to the IntelliCenter and grab a capture. I wonder if it would yield the necessary messaging without actually having to actually plumb it in.  Just thinking out loud. I'm certainly willing to help if I can.

gw8674, 2020-07-19:
r.,
Another question on this issue.
If you could get anything you needed from Pentair to make this work, what would it be?
In other words, what specific question(s) would you ask the Pentair Engineer and what information/data would you be asking for exactly. 
Thanks!!

rstrouse, 2020-07-20:
I am not even certain iChlor would yield any results.  We do know that iChlor still communicates using the same protocol as IntelliChlor but the unit has additional features that are not broadcast through that protocol.  And it periodically broadcasts a message on the 12 address.  

However, I don't think an unplumbed test bed would yield the results we are looking for.  The idea here would be be go all out Jane Goodall and observe how it behaves within the troop.  A test bed that doesn't include activating the flow sensor or temperature sensor would be like throwing bananas at it through a chute.  In the end, it may just act like any chlorinator member but who knows maybe just maybe the silverback acknowledges it differently within the IntelliCenter troop.

If the Pentair engineers were to cough up the protocol for IntelliValve that would be spectacular.  However, I wouldn't hold my breath on that one.  That would be way above their pay grade.  While I would be surprised to get a response, I would ask Pentair for documentation on what messages are requested to IntelliValve and what messages are responded from IntelliValve.  Simple as that.  

I can't imagine them giving the information because innovation requires investment and without it we simply have cool ideas.  Cool ideas are driven by capital.  Those that can do the technical work have to be given that capital so that they can work on the idea and still fill their bellies.  Without it we have cool ideas and hunter-gathering.

While I can imagine that the engineering team is under intense pressure from the front side and the back side of the house to stabilize everything that has been released, IntelliValve doesn't represent pressure relief within my field of view.  If you think the pressure from the front side of the house is intense.  Imagine the investment, required to get IntelliCenter packaged into a cardboard box with a black and white label.  The life of the entire division depends on its success.  Perhaps there is some pressure from the back side of the house that we cannot see.  Then again looking up someone's backside doesn't make any sense.

Given the current state of the IoT space It is completely normal for them to be in a one legged butt kicking contest to reach some stasis with things like Alexa, mobile apps, and platforms.  All of it is currently very fragile and is very complex.   A good indicator of this is that failure is often not complete, rather it is some form of annoying unusable state when it occurs.

Pentair will get it figured out but only if they don't run out of money.  Right now service based software is a capital suck.  The front lines are so far away from the supply zone that a washed out foot bridge can bring everything into chaos.  The only folks really succeeding in this right now are platform host companies.  I wonder why everything always gets blamed on capacity when latency is the true problem.

tagyoureit, 2020-07-20:
> If the Pentair engineers were to cough up the protocol for IntelliValve that would be spectacular.

Heck, just ask them for the whole kit-n-caboodle for the specs.  There's still stuff we don't understand about the basic 2-status packet.  ;-)

gw8674, 2020-07-31:
I did ask (again).... They are "discussing" it.

gw8674, 2020-08-03:
Hi all...
On Friday, I received a message on TFP from a gentleman, that provided some new information regarding the IntelliValve.

I provided him with this link and suggested he come here to share the information.  I can only assume that he is busy these last few days.  I messaged him back this afternoon and advised that I would share the following information in order to get this out to the rest of the applicable folks here.  I'm sure he will be along shortly...  I'm hoping that the following information can provide some additional pieces to this puzzle.  Thoughts?

**Any luck decoding the intellivalve, I have three now and just ordered a fourth one to take apart and see if i can download the program directly form the controller chip. I am a computer and electrical engineer. I have hooked one up and here is what i got.

FF 00 FF A5 01 10 0C 52 08 00 80 D8 80 39 E7 06 64 04 7E

I have decoded the following
FF 00 FF A5 = preamable
01 = version
10 = destination
0C = source
52 = command
08 00 80 D8 80 39 E7 06 64 = data packet
04 7E = Checksum

as for the data packet
08 =home value of 0
D8 = xxxx value of 80
39 = xxxx vale of E7
06 = diverter end position value of 64 = 100%**

rstrouse, 2020-08-04:
He got "I am Groot".
```[255,0,255][165,1,16,12,82,8][0,128,216,128,57,231,6,100][4,126]```.  Here it is in decimal, the first 8 is actually the data length not part of the data itself.

I am building a purpose built app and have hooked another pi to see if I can send more messages faster to the valve.  As it stands I have tried several hundred thousand combinations and that damn tree keeps saying "I am Groot".  I will learn the language.

gw8674, 2020-08-04:
R.,
Ah, gotcha...
Hopefully you have some success with the app! 
Thanks!!

thumbnut, 2020-08-04:
I was wondering if Pentair designed the Intellivalve to operate similar to how the intellibrite and globrite lights work. They pulse the power and the lights react based on the number of pulses. It could be that the only purpose of the RS-485 is identification and feedback of position. So pulse the line that powers counterclockwise rotation 5 times and the valve moves 5 positions, or pulse the line 10 times in the clockwise direction and the valve moves 10 positions in that direction. Solid power in either direction moves the valve to the preset.
Just thinking out loud.  Pentair seems to go for the simplest approach. Even simpler than Morse code.

rstrouse, 2020-08-04:
@thumbnut I would highly doubt that one.  In fact I don't think the brain-dead on/off protocol was even a Pentair creation as many of the lighting products out there use the on/off sequencing (gawd why not dmx).  This kinda smacks of the Fiberstars stuff that was going around at the time it was all created.  This was back when colors were selected on a color wheel and Spectrum Amerlite was the new tech.  On/off was the way to tell the stepper motor to stop after a specific period of time.

Here are the things that I am pretty certain of.
1. There is only one set of endpoints stored on the valve.
2. The address that we need to communicate with the valve is included in "I am Groot".
3. The 12 message destination which all valves communicate on is a general hail address.  We have seen iChlor broadcasting on this destination address.
4. When connected to RS485 the last 6 bytes of "I am Groot" are unique.  However, valves manufactured around the same time have the first 3 of these bytes in common and the 4th byte seems to be sequential as if the first 3 indicate the batch and the 4th is incremented as each valve is programmed.  The valve above was likely manufactured after or on 4/19/2016 and before 08/19/2019.
5. I'm pretty sure we need to first address it then it will start calling us dad.  My worry is that the addressing function only occurs when the valve is in one of the modes other than auto... Off, Set, or Service.  If it is, then the huge number of potential messages are then multiplied by 4 and everything that has currently been sent in the Auto mode will need to be repeated for the 3 other modes.
6. When setting the address of the valve I am pretty sure we need to send at least some subset of the 6 bytes to it and perhaps a single byte identifier that will be used to address the valve.  This will likely be 1-26 or 65-90.

Amigaman68, 2020-08-07:
Actually I am going to dis assemble on of my valves. hopefully it isn't potted so i can get at the control board. The patent application shows a fairly detailed assembly. Will let you know when i get it apart. hopefully they used a standard chip.

Amigaman68, 2020-08-07:
[intelivalve patent application.pdf](https://github.com/tagyoureit/nodejs-poolController/files/5039493/intelivalve.patent.application.pdf)

Amigaman68, 2020-08-07:
Also the documentation on the pentair controller says a max of four valves on a rs485 channel. Not sure that will help

rstrouse, 2020-08-07:
@Amigaman68,

Could you point me to the documentation where you found that limit?

Are you referring to the 4 valve limit as the Intake, Return, ValveA, and ValveB?  If so that is the limit as it relates to the number of stock valve relays found on IntelliTouch, EasyTouch, and IntelliCenter.  For IntelliTouch you can add 3 additional relays and in IntelliCenter you can add 6 additional per personality board.  There is an apparent limit of 1 valve expansions per load center but I think this is because of the power draw potential of moving that many valves at once.  Each valve can draw 750mA continuous and it is an inductive load.  While the limit appears to be 7 total for all configurations in IntelliTouch the limit for IntelliCenter is supposed to be 26 yet mathematically you come up with 28 if you add up the expansion centers and relays.  In order to get to 26 you would have to remove the Intake/Return out of the equation for a shared body.  Btw this also coincides with the lettering scheme found on the valves A-Z.  

I would be very interested to hear your findings on your valve deconstruction project.

Amigaman68, 2020-08-07:
[image]

Amigaman68, 2020-08-07:
I based it upon the IntelliCenter Control System Installation Guide, but i agree if we can address it, then that may just be a limit of the intellicenter

thumbnut, 2020-08-07:
@rstrouse,

I have 13 valves that I need to hook up. One is a standard Intermatic and 12 are IntelliValves. I have been told by numerous Pentair techs and engineers that we cannot break the 10 valve total barrier on RS485, which includes the Intake, Return, ValveA, and ValveB. The only way around as I understand is to use relays, which bypass the RS-485.
I have the Intellicenter with i10PS personality module and valve expansion board. I had also purchased an expansion center with i5X and a second valve expansion board, but was told the second valve expansion board would not be seen by the automation. The best I could do is make use of the extra AUX circuits on the expansion panel to run valves. I ended up returning the expansion panel and boards.
 Do you know of any example where more than 10 IntelliValves have been operated on RS485?

Amigaman68, 2020-08-07:
As far as i know, no one has controlled the valves via rs485 yet except pentair. all anyone has done has gotten i am grout message. They wont release the protocols for communication. which is why i am taking one apart and hoping i can identify the controller chip, and download the software from it and disassemble the code, and figure out the protocols. Which should be fun because i haven't done assembly language in a long time, but i think i am up for the challenge.

Amigaman68, 2020-08-07:
the patent application contains a lot of useful information.

rstrouse, 2020-08-07:
@thumbnut,

Actually, I think the 10 valve limit is due to the traces on the IntelliCenter system.  I haven't spent any time looking too deeply at it, but the I/O channels are probably the limit.  If you add an XP this gives you additional I/O channels.  In fact the standard names for all valves > 10 start with (XP1), (XP2)... etc.  I think this is a hardware not a programming limit.  To that end I suspect there are some power limits with the transformer and traces as well.  Wanna see magic smoke draw 26 amps on that transformer.  Good thing there is a breaker on it.  

I agree that the way around it is using relays but if one were inclined that doesn't actually mean direct circuit relays for the valves.   These could be triggered from the circuit but emitted to another controller or even relays hardwired to the load side of the circuit.  My design could benefit from some variable valving but I have gotten around most of that with variable speed pumps.

So far I have run several million messages to the valve to see if I can get it to respond.  My fear is that there is some sort of user interaction required to get it to address.  So far everything is being done in auto mode but it may require the valve to be in off or set to get it to respond.  I know it's a fools errand.

Amigaman68, 2020-08-07:
So, i us a modicon plc to run my setup, so i am using relays until i can get at the protocols. what makes me mad is pentair wont release the protocols, i already bought the valves from them.

Amigaman68, 2020-08-07:
i opened it up
[image]
[image]
[PIC16F1717_8_9-data-sheet-40001740C.pdf](https://github.com/tagyoureit/nodejs-poolController/files/5043465/PIC16F1717_8_9-data-sheet-40001740C.pdf)
what i have learned so far

rstrouse, 2020-08-07:
As the French would say moi aussi.  I don't have any doubt that the protocols will never be released.  In fact the so called API rumor will probably only see the light of day to peeps like Control4 which by the way is such a small subset of the product.  I might be pleasantly surprised but that old-dog spirit of obscurity is so hard for engineers to give up.  What they fail to realize is that opening it up would put a death nail in competing platforms.  These systems are often sold by a network of pool builders that don't understand all the tech.  However, customers tend to drive that with the "is it possible to" statement.  At some point getting the deal takes precedence over incentives to the dealer.

My approach is depending on things that I have learned from the RS485 protocols.  Groot fits the mold but there are other protocols on this bus.  While theoretically it could communicate on another channel or even using a start/stop marker it is unlikely.

Amigaman68, 2020-08-07:
we are about to find out i think i see a serial port on jp2

Amigaman68, 2020-08-07:
the five pins are a serial port startin on the bottom pins bein 1,2,3,4,5 and 5 being at the top. pin 4 connects to pin 28 which is ICSPDAT, pin 5 connects to pin 27 which is ICSPClK,  and i only need to connect to pin 4&5 to be able to access the program. will need to order pic programming software and serial converter

Amigaman68, 2020-08-07:
[image]

Amigaman68, 2020-08-07:
I just ordered the programmer, but i am going away so I wont be able to continue till next week

thumbnut, 2020-08-07:
If and when we do crack the IntelliValve protocol, will it be possible to add code to run them directly from poolController?
If we don't, will it be possible to add a rpi and some solenoids to extend the number of valves we can control? I would much rather buy a cheap rpi, a 24v transformer, and some relays, so I could at least have my own direct control of any amount of valves I have, rather than paying a thousand to Pentair for another fairly useless motherboard with a display attached so I can control some relays.

rstrouse, 2020-08-07:
Once we see the level of control we can make a determination as to which way to go with it.  Perhaps we end up with valves AA-ZZ as a 3rd type of valve.  Either way the valve relay will be needed as the protocol won't actually spin the motor.  Currently you must energize one or the other load to get the valve to move to an endpoint but maybe for light duty you could simply flip the reverse direction bit.  Even so, today you can control a valve relay from PoolController through the interface bindings when a circuit is activated.  What we are looking to do here is control those endpoints.

theRealMCQwerty, 2020-08-09:
> I think the 10 valve limit is due to the traces on the IntelliCenter system.

I think I accidentally, (well ignorantly), tested this.
I currently have 2 of my 3 valve expansion modules installed. If use 5 of the unaddressable connection to provide power to 5 intellivalves that just sit in service mode and make me grumble every time I walk over to manually push their buttons. The 6th connection on the second expansion board is connected to a cable that runs to my junction box where I use it as a power feed for my other 9 intellivalves. When I first hooked it up at least 6 if not all 9 of them started moving. So one connection can power at least 6 valves all at the same time. Not sure what would happen if I tried more than that, but I don't have that many hands to hold down buttons on valves in service mode (grumble!).
Also, interesting that a valve expansion board that is unaddressable (although detected) by the main system provides power to the valve connectors.
I have a spare 24VAC transformer so I think I need put that in my junction box to run the extra valves rather than pushing my luck.
I have been delaying all of this since I really don't want to cut my cables and wire them into an ESP8266 controlled relay array. RS485 control woudl solve so many of my problems. Still holding out that @rstrouse will break the code and make that groot talk sense.
@Amigaman68 - good luck with your tear-down operation. I love the Blitz spirit on this one!

rstrouse, 2020-08-09:
Yep the message generator has been chumming the RS485 waters and has dumped over 2 million rotting fishes in the vicinity of the valve.  Narry a peep out of it yet.  It will continue to run... lets hope we need a bigger boat.

On a side note I have added some code that may give some relief to the extra valves and keep you from wearing a path to the pad.  This involves adding another controller that is set up to get http requests from pool controller.  A description of its operation can be found [here](https://github.com/tagyoureit/nodejs-poolController/wiki/Integrations-in-2.0#valve-relays).  

This should allow you with some minor programming and cobbling of existing components to add as many valves as you want (actually I set an artificial limit of 100) to your pool definition.  poolController will emit the commands regardless of where the circuit is triggered.  It will act just like a valve defined on IntelliCenter except the control will come from poolController.

tagyoureit, 2020-08-09:
This all reminds me of The Imitation Game and Alan Turing trying to break the unbreakable Enigma code.

rstrouse, 2020-08-09:
Yeah, we have eyes on.  While, @Amigaman68 is waterboarding, I continue to try to get one drunk.  How messed up would that be if the only thing it knows is "I am Groot."

Two valves walk into a bar.  One of them looks at the bartender and says "I am Groot."

rstrouse, 2020-08-11:
So I bought a pi compatible relay and will hook it up to a test bed pi probably next weekend.  That way I can follow along with you guys while grooting.  Maybe I'll create a valve relay controller server that at least works with a pi and can be easily configured.

Below is a user defined valve that is set up to operate on gpio pin 25.  Currently it sends an http request off into the ether but when I get the relay hooked up, there should be someone out in center field to catch that ball.

[image]

thumbnut, 2020-08-11:
Which relay did you get? I am looking to do something similar. I was hoping to bypass the pi and go straight to a web controlled relay, but I could go with the pi. I need 9 or more relays and there is a 16 channel relay that would work, but I have read from reviews on Amazon that it does not play nice with rpi.
https://www.sainsmart.com/products/imatic-rj45-ethernet-wi-fi-control-board-with-integrated-16-ch-dc-12v-relay

rstrouse, 2020-08-11:
I simply bought a hat for now with only 3 [relays](https://www.amazon.com/Raspberry-Pi-Expansion-Module-XYGStudy/dp/B01G05KLIE?ref_=fsclp_pl_dp_5) and it is stackable.  You can go directly to a web controlled relay as it should work.  I figured I would fart around with a board that is known to use 3v3 signal.  The problem with most of these is that they are 5v signal and are not well documented but there are a whole lot of tutorials out there to get you going.  I may add i2c to the project to be able to read temp sensors and the like at a later date.  For now I will be coding to straight GPIO.

Amigaman68, 2020-08-12:
So, Have not been able to download the program yet. it code locked, amd going to have to try voltage glitch the pic controller. its going to take a while.

Amigaman68, 2020-08-12:
Also i thing they may have put a false marking on the chip.

rstrouse, 2020-08-12:
I figured they probably flipped the copy protection bit.

Amigaman68, 2020-08-12:
I have to build a setup to voltage glitch it while its booting

thumbnut, 2020-08-12:
I like [this one](https://www.robotshop.com/en/sequent-microsystems-8-relay-expansion-hat-raspberry-pi.html). I am trying to keep everything in the low voltage side of the IC Load Center. It looks like this will work.

rstrouse, 2020-08-12:
@thumbnut I saw that one but never saw a place where I could actually get it before the sun starts making carbon.  That does look like a compact little unit.

thumbnut, 2020-08-12:
I ordered the last one they have but they will get more in a few weeks. It has a neat little LED hat that goes with it. I was surprised to see that it allows you to stack up to eight high. It must be great for the extreme holiday light enthusiasts.

theRealMCQwerty, 2020-08-12:
> user defined valve that is set up to operate on gpio pin 25

This is cool. I hooked up an 8-channel relay I have been meaning to connect to an ESP8266 and control via ESPHome but decided to give this a shot first since my pi is already out on the pad.

Pulled down the latest dashpanel and also the latest `next` branch of pool controller. Added a virtual valve on GPIO17 and got this:

```
ApiError: Valve Id has not been defined
    at /home/openhabian/nodejs-poolController/controller/boards/SystemBoard.ts:2389:35
    at new Promise ()
    at ValveCommands.setValveAsync (/home/openhabian/nodejs-poolController/controller/boards/SystemBoard.ts:2388:16)
    at /home/openhabian/nodejs-poolController/web/services/config/Config.ts:222:52
    at Layer.handle [as handle_request] (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at /home/openhabian/nodejs-poolController/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/index.js:335:12)
    at next (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/index.js:275:10)
    at /home/openhabian/nodejs-poolController/web/Server.ts:292:21
    at Layer.handle [as handle_request] (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/index.js:317:13)
    at /home/openhabian/nodejs-poolController/node_modules/express/lib/router/index.js:284:7
    at Function.process_params (/home/openhabian/nodejs-poolController/node_modules/express/lib/router/index.js:335:12)

```

What did I mess up?

rstrouse, 2020-08-12:
@theRealMCQwerty you are just getting a little bit ahead of me.  Don't get me wrong I really appreciate you grabbing this by the horns.  The IntelliCenter code is just grabbing the configuration before it gets added and I will be posting up some final changes for the config shortly.

theRealMCQwerty, 2020-08-12:
@rstrouse got it. 👍 
Don't let me disturb you from your grooting. That is the holy grail, raspberry pi GPIOs are pretty nice in the interim though 😄 

Also, (and this is certainly a distraction), any chance you could add support for inputs? The DHT22 (temp and humidity) would be fantastic. When I move the pi into a box with the relays this thing is going to get pretty hot and I would like to activate a 5v case fan based on a temp reading inside the box. It would be cool to also have a DHT22 outside the box to get a second opinion on air temp also as well as humidity.

I could of course do all of this with an ESP8266 but since we have the pi just sitting there and it has wired ethernet vs wifi on the ESP then it would be a neat solution. 😃 Would also convince me to finally spend the $ on a poe hat for the pi too 😉

rstrouse, 2020-08-12:
> Also, (and this is certainly a distraction), any chance you could add support for inputs? The DHT22 (temp and humidity) would be fantastic. When I move the pi into a box with the relays this thing [...]

How fun is that.  I'll see what all that takes.  My foray into pi GPIO will be a learning experience.  About to hit the 3M message mark on grooting.  Bear in mind the total possibilities are psychotically high and 3M is a drop in an ocean.  The approach however is to groot potential likely suspects so it's not entirely random.   I'd really like to see @Amigaman68 have some success so at least we can limit the incoming byte signatures.

theRealMCQwerty, 2020-08-12:
> About to hit the 3M message mark on grooting.

I think you said you are trying first with the ItelliValve in service mode correct?
Is this setup something us kids can try at home? Maybe I could run the same message list but with my valve in Auto mode? At least we could shorten the timelines a bit and if we had enough volunteers maybe message batches could be assigned each grooter. 
Bit like a bitcoin mining pool. You could call it Mt.Groot..... ok maybe _that's_ not such a great idea 😄

rstrouse, 2020-08-12:
@theRealMCQwerty I am hitting it in auto mode.

Here is the setup that it needs:
1. Exactly 1 valve hooked up directly to an RS485 dongle on a pi.  You can have no other devices on that bus.  If you do we won't be able to see stray messages being bantied about.
2. 24vac connected to one side of the valve.  This should power the valve and have it rotated to at least one of the stops.  I changed the configuration on the valve so that both of the stops have been changed.  This way if the valve reverses or an endpoint changes I can easily see it.

If you have such a setup I can package it up but it will take some time.  Git makes me want to sky scream.

Do you have such a setup?

thumbnut, 2020-08-12:
@rstrouse I have all the needed hardware an am totally up to "crowd-grooting".

rstrouse, 2020-08-12:
Later tonight I will package up the code.  We should make sure we are not duplicating efforts so each one of us should declare which mode they are running on.  I claim Auto.
There are 3 other modes.
1. Set
2. Off...  Don't discount this one as off may simply mean that the local valve interface is off.
3. Service... This one is probably the least likely since regardless of whether it is being controlled from the panel the service mode should keep the valve at the manual position.  This would be true for instance where you winterize a heater or solar.  Don't want that valve moving.

Amigaman68, 2020-08-12:
I have all the needed hardware an am totally up to "crowd-grooting".  as well

rstrouse, 2020-08-12:
Well aren't we getting all SETI on this.

theRealMCQwerty, 2020-08-12:
> Do you have such a setup?

That is exactly the setup I had back when we first started testing this out. I still have all the parts around and can easily set it back up again to commence Grooting-at-home!

theRealMCQwerty, 2020-08-12:
@rstrouse not sure how much work this would be but since there are a few of us, could we also divide up the message space to maybe exhaust the most likely mode/message sets first then we all switch to another mode and run our own respective message group, or we all move to another message group based on perceived probability of success?

rstrouse, 2020-08-12:
Good idea I have set up two distinct patterns.  The key would be to take the assignments and break out the address ranges.  When I get the code posted up we will divvy up the assignments.  As it stands right now I probably need to add a page to check on the status as right now I am watching a file.  It'll be ugly but it will display the underlying data.

rstrouse, 2020-08-13:
Ok guys so I posted the [valve-Grooter](https://github.com/rstrouse/valve-Grooter) up on github.  You can pull it install it then check to see if you have your environment set up.  Then let's take a role call to see where we stand,

gw8674, 2020-08-13:
r.,
Would it be okay to leave an existing IntelliValve (not a bench test actuator) plugged in to its 3-pin socket on the valve expansion module to get power and then just connect the two RS-485 wires (grn/yel) to the RS-485 adapter connected to the rp?

rstrouse, 2020-08-13:
@gw8674 Yes that should work like a charm.  As far as connectors I bought some male jst pigtails for the power side so I don't destroy the connector and some female jst for the RS485 pins.  They work great and will be what I hook the valve up to the relay controller with.

[211 more replies]

closed 2020-11-02 by tagyoureit