tagyoureit/nodejs-poolController #190: state/all reports 55 GPM for Intelliflo VS when pump is off (next branch)

dkossman, 2020-07-18:
Intelliflo VS pump is off, but flow is reported as 55 GPM in state/all JSON

### Expected behavior and actual behavior
http://myserver:4200/state/all
expected: GPM is 0
actual: GPM is 55

`
{"time":"2020-07-12T15:46:00.411-0700","valve":0,"batteryVoltage":0,"status":{"val":1,"name":"ready","desc":"Ready","percent":100},"mode":{"val":0,"name":"auto","desc":"Auto"},"appVersion":"6.0.0","circuits":[{"id":6,"name":"Pool","type":{"val":6,"name":"generic","desc":"Generic"},"isOn":true,"equipmentType":"circuit"}],"temps":{"units":{"val":0,"name":"F","desc":"Fahrenheit"},"bodies":[{"id":1,"heatMode":{"val":0},"name":"Pool","circuit":6,"heatStatus":{"val":0,"name":"off","desc":"Off"},"isOn":true}],"equipmentType":"temps"},"equipment":{"model":"Virtual Controller","maxBodies":1,"equipmentType":"equipment","softwareVersion":"","bootLoaderVersion":""},"pumps":[{"id":1,"type":{"val":128,"name":"vs","desc":"Intelliflo VS","maxPrimingTime":6,"minSpeed":450,"maxSpeed":3450,"speedStepSize":10,"maxCircuits":8,"hasAddress":true},"status":{"name":"off","desc":"Off","val":0},"targetSpeed":0,"virtualControllerStatus":{"val":1,"name":"running","desc":"Running"},"command":4,"mode":0,"driveState":255,"watts":0,"rpm":0,"flow":55,"ppc":44,"name":"Intelliflo VS","time":491,"equipmentType":"pump","minSpeed":450,"maxSpeed":3450,"speedStepSize":10,"circuits":[{"id":1,"circuit":{"id":7,"isOn":false,"name":"Pump 750 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},"units":{"val":0,"name":"rpm","desc":"RPM"},"speed":750},{"id":2,"circuit":{"id":8,"isOn":false,"name":"Pump 1500 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},"units":{"val":0,"name":"rpm","desc":"RPM"},"speed":1500},{"id":3,"circuit":{"id":9,"isOn":false,"name":"Pump 2500 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},"units":{"val":0,"name":"rpm","desc":"RPM"},"speed":2500},{"id":4,"circuit":{"id":10,"isOn":false,"name":"Pump 3000 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},"units":{"val":0,"name":"rpm","desc":"RPM"},"speed":3000},{"id":5,"circuit":{"equipmentType":"circuit","id":0,"name":"Not Used","speed":0},"units":{"val":0,"name":"rpm","desc":"RPM"}},{"id":6,"circuit":{"equipmentType":"circuit","id":0,"name":"Not Used","speed":0},"units":{"val":0,"name":"rpm","desc":"RPM"}},{"id":7,"circuit":{"equipmentType":"circuit","id":0,"name":"Not Used","speed":0},"units":{"val":0,"name":"rpm","desc":"RPM"}},{"id":8,"circuit":{"equipmentType":"circuit","id":0,"name":"Not Used","speed":0},"units":{"val":0,"name":"rpm","desc":"RPM"}}]}],"valves":[],"heaters":[],"chlorinators":[],"features":[{"id":7,"isOn":false,"name":"Pump 750 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},{"id":8,"isOn":false,"name":"Pump 1500 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},{"id":9,"isOn":false,"name":"Pump 2500 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"},{"id":10,"isOn":false,"name":"Pump 3000 RPM","type":{"val":0,"name":"generic","desc":"Generic"},"showInFeatures":true,"equipmentType":"feature"}],"circuitGroups":[],"lightGroups":[],"virtualCircuits":[],"covers":[],"schedules":[],"chemControllers":[]}
`
### Specifications
1. What version of the code are you using?
 latest next (git pull, npm start)
2. All output from the file log 
 see above
3. Any errors that are in the console
 don't see any errors
4. Your config.json or in-use configuration file

```
{
  "controller": {
    "comms": {
      "rs485Port": "/dev/ttyUSB0",
      "mockPort": false,
      "netConnect": false,
      "netHost": "ospi",
      "netPort": 9801,
      "inactivityRetry": 10,
      "portSettings": {
        "baudRate": 9600,
        "dataBits": 8,
        "parity": "none",
        "stopBits": 1,
        "flowControl": false,
        "autoOpen": false,
        "lock": false
      }
    }
  },
  "web": {
    "servers": {
      "http2": {
        "enabled": false
      },
      "http": {
        "enabled": true,
        "ip": "0.0.0.0",
        "port": 4200,
        "httpsRedirect": false,
        "authentication": "none",
        "authFile": "/users.htpasswd"
      },
      "https": {
        "enabled": true,
        "ip": "0.0.0.0",
        "port": 4201,
        "authentication": "none",
        "authFile": "/users.htpasswd",
        "sslKeyFile": "",
        "sslCertFile": ""
      },
      "mdns": {
        "enabled": false
      },
      "ssdp": {
        "enabled": true
      }
    },
    "services": {},
    "interfaces": {
      "smartThings": {
        "name": "SmartThings/Hubitat",
        "enabled": false,
        "fileName": "smartThings-Hubitat.json",
        "globals": {},
        "options": {
          "host": "0.0.0.0",
          "port": 39500
        }
      },
      "vera": {
        "name": "Vera",
        "enabled": false,
        "fileName": "vera.json",
        "vars": {
          "deviceId": 0
        },
        "options": {
          "host": "",
          "port": 3480
        }
      }
    }
  },
  "log": {
    "packet": {
      "enabled": false,
      "logToConsole": true,
      "logToFile": false,
      "filename": "packetLog",
      "invalid": true,
      "broadcast": {
        "enabled": true,
        "includeActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeActions": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "pump": {
        "enabled": true,
        "includeActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeActions": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "chlorinator": {
        "enabled": false,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "intellichem": {
        "enabled": false,
        "includeActions": [],
        "exclueActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "intellivalve": {
        "enabled": false,
        "includeActions": [],
        "exclueActions": [],
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "unidentified": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      },
      "unknown": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      }
    },
    "app": {
      "enabled": true,
      "level": "info",
      "captureForReplay": false
    }
  },
  "appVersion": "6.0.0"
```

5. Your pool equipment
  Intelliflo VS only

tagyoureit, 2020-07-18:
This tells me the "what" but not the "why".  Can you follow the directions for the [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach those log files here.

But regardless, VS Pumps don't have GPM at all.  You shouldn't expect to see GPM 55 or GPM 0.  I'm not sure why that is showing in addition to RPM

dkossman, 2020-07-18:
good point - i've updated the issue.  will work on a packet capture, thanks.

dkossman, 2020-07-19:
replay.zip is attached - the activity captured is just reloading the webclient on the browser.
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/4944702/replay.zip)

rstrouse, 2020-07-20:
Does this value ever change?  Your pump is actually reporting 55 in the flow byte.  You also appear to have 44 in the ppc byte and 255 in the driveState when the power is 0.  When the pump is running it would be interesting to see what the values are for these three bytes.  They could simply be undefined for your pump model.

tagyoureit, 2020-07-20:
I was just going to add a check for the pumps to not add speed to VF or flow to VS, but @rstrouse's points are valid.

I also did notice the first `mode` value is 9 means `Overcurrent condition 2`.  It went away with the 2nd status packet.

@dkossman Can you do a longer run with the pumps running and post back up?

dkossman, 2020-07-21:
OK - attached.  This run was: pump off, then 750RPM, then 1500RPM, then 750RPM, then off.
[replay2.zip](https://github.com/tagyoureit/nodejs-poolController/files/4951019/replay2.zip)

tagyoureit, 2020-07-21:
Hi, what model/brand is your pump?  It is giving different packets than we've seen before and reporting things that @rstrouse mentioned above.  I'll go out on a limb here and say that your pump is reporting some wacky data. 

We've seen enough VS pumps to have a really good handle on how they work and this one isn't fitting the mold.  Not sure there is really anything we can do to adjust the code unless we can find other pumps to corroborate this one's story.

dkossman, 2020-07-21:
it's a Pentair Intelliflo VS - model 011055.   I've never had any problems with it...

rstrouse, 2020-07-21:
This is actually not a VS3050.  This is an IntelliFlo 2 VST.  Interesting, I'll have a look at the log tomorrow.  It may just be that they are reporting something else int the flow byte for this pump.

rstrouse, 2020-07-21:
@tagyoureit your initial instincts are correct.  There appears to be no value in either byte 7 or 8.  I am guessing 55, 44 respectively are simply filler bytes for when the structure is first initialized.  My hope was that the ppc actually contained backpressure and 7 was calculated flow.  Alas, this is not to be.  I modified the pumpStateMessage to look at the pump type then apply speed, flow, or both depending on the capabilities.  If you grab an update to njspc it will no longer report flow for a VST pump.

dkossman, 2020-07-23:
Thanks - I no longer see the flow being reported in the web client, and it comes back as 0 in /state/all.
I'm going to close this issue.

closed 2020-07-23 by dkossman