tagyoureit/nodejs-poolController #197: circuit 192 doesn't affect light circuits

emes, 2020-07-23:
### Expected behavior and actual behavior

Toggling individual lighting circuits results in expected behavior...the lights turn on and off.

Toggling circuit 192 results in the lightGroup being marked isOn, but does not change the individual lighting circuits.

Initial status looks good.  The correct circuits are in the lightGroup.

```
    "lightGroups": [
        {
            "id": 192,
            "action": {
                "val": 0,
                "name": "ready",
                "desc": "Ready"
            },
            "isActive": true,
            "isOn": true,
            "type": {
                "val": 3,
                "name": "intellibrite",
                "desc": "IntelliBrite"
            },
            "name": "Intellibrite",
            "lightingTheme": {
                "val": 196,
                "name": "white",
                "desc": "White",
                "type": "intellibrite"
            },
            "equipmentType": "lightGroup",
            "circuits": [
                {
                    "id": 1,
                    "circuit": {
                        "id": 3,
                        "showInFeatures": true,
                        "isOn": false,
                        "name": "East Light",
                        "type": {
                            "val": 16,
                            "name": "intellibrite",
                            "desc": "Intellibrite",
                            "isLight": true
                        },
                        "nameId": 207,
                        "lightingTheme": {
                            "val": 196,
                            "name": "white",
                            "desc": "White",
                            "type": "intellibrite"
                        },
                        "equipmentType": "circuit"
                    },
                    "position": 1,
                    "color": 8,
                    "swimDelay": 0,
                    "isActive": true
                },
                {
                    "id": 2,
                    "circuit": {
                        "id": 4,
                        "showInFeatures": true,
                        "isOn": false,
                        "name": "West Light",
                        "type": {
                            "val": 16,
                            "name": "intellibrite",
                            "desc": "Intellibrite",
                            "isLight": true
                        },
                        "nameId": 208,
                        "lightingTheme": {
                            "val": 196,
                            "name": "white",
                            "desc": "White",
                            "type": "intellibrite"
                        },
                        "equipmentType": "circuit"
                    },
                    "position": 2,
                    "color": 8,
                    "swimDelay": 5,
                    "isActive": true
                }
            ]
        }
    ]
```

### What is actually happening (screenshots are helpful)

lights do not turn on or off via lightGroup

### Steps to reproduce the problem

PUT /state/circuit/toggleState {"id":192}

```
{
    "id": 192,
    "action": {
        "val": 0,
        "name": "ready",
        "desc": "Ready"
    },
    "isActive": true,
    "isOn": true,
    "type": {
        "val": 3,
        "name": "intellibrite",
        "desc": "IntelliBrite"
    },
    "name": "Intellibrite",
    "lightingTheme": {
        "val": 196,
        "name": "white",
        "desc": "White",
        "type": "intellibrite"
    },
    "equipmentType": "lightGroup"
}
```

but light circuits are not turned on.

```
        {
            "id": 2,
            "showInFeatures": true,
            "isOn": false,
            "name": "Levolor",
            "type": {
                "val": 0,
                "name": "generic",
                "desc": "Generic"
            },
            "nameId": 200,
            "equipmentType": "circuit"
        },
        {
            "id": 3,
            "showInFeatures": true,
            "isOn": false,
            "name": "East Light",
            "type": {
                "val": 16,
                "name": "intellibrite",
                "desc": "Intellibrite",
                "isLight": true
            },
            "nameId": 207,
            "lightingTheme": {
                "val": 196,
                "name": "white",
                "desc": "White",
                "type": "intellibrite"
            },
            "equipmentType": "circuit"
        },
        {
            "id": 4,
            "showInFeatures": true,
            "isOn": false,
            "name": "West Light",
            "type": {
                "val": 16,
                "name": "intellibrite",
                "desc": "Intellibrite",
                "isLight": true
            },
            "nameId": 208,
            "lightingTheme": {
                "val": 196,
                "name": "white",
                "desc": "White",
                "type": "intellibrite"
            },
            "equipmentType": "circuit"
        }
```

### Specifications

1. What version of the code are you using?
50a8311ffafb3121eadad984ac0b74dc0c5fcd64
2. All output from the file log 
[capture](https://www.dropbox.com/s/q7heyze2e0r15zm/2020-07-23_16-36-23.zip?dl=0
)

3. Any errors that are in the console
none
4. Your config.json or in-use configuration file

5. Your pool equipment

IntelliTouch i5+3S

tagyoureit, 2020-07-26:
I have this fixed but I'm in the middle of another change.  Hopefully will push to `next` branch by tonight.  I'll reply here when I do.

tagyoureit, 2020-07-27:
Pull the latest `next` branch and this should be fixed.  Ironically, we have just always assumed that setting a theme would turn on all of the associated light circuits.  IntelliCenter doesn't have an "Off" theme (but *Touch does).  I had to add in logic to turn off a light group from the toggleState.

emes, 2020-07-27:
Tested.  Works!  Thanks!

closed 2020-07-27 by emes