tagyoureit/nodejs-poolController #199: neither gas nor solar heater showed up

jbarneson, 2020-07-27:
loaded up version 6 went very well -- but neither of my heaters show up in the states array: 

[image]

I'd really like to monitor their run times - are they supposed to be returning state data? I do have the ability to set my desired heat temperature on the pool circuit... but i was hoping to see their running state as well.

tagyoureit, 2020-07-27:
Hi,
can you please give us the missing information from the list below?  The answer may depend on your setup and what type of equipment you have.
thx.

**Packet Capture**
Follow the instructions to complete a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) and attach the resulting zip/log files.

**Pool Equipment**
 - Controller: [e.g. IntelliCenter, EasyTouch, IntelliTouch]
 - Controller Model: [e.g. IntelliCenter i5PS, EasyTouch2 8P]
 - Controller Firmware Version: [e.g. 1.047]
 - Pump(s) manufacturer and model: [e.g. IntelliFlow 2 VST 011056]
 - Chlorinator: [e.g. iChlor, IntelliChlor-40]
 - Heater(s): [e.g. gas, solar, heatpump, ultratemp]
 - Chemical controller: [e.g. IntelliChem, homegrown]
 - Valves: [e.g. Intellivalve]
 - Any other relevant equipment:

**Additional context**
Add any other context about the problem here.

rstrouse, 2020-07-27:
That heaters array is for hybrid, ultratemp, and specialized heatpumps that will come later since these are the only types of heaters that report on the RS485 bus.  

However, there are two places where the state of the running heaters are emitted.  First is on the temps array.  There is a temps.bodies[n].heatStatus which will indicate the current status of the body heating.  These values will be off, heater, or solar.  If you have a heatpump it will be off, heater, heatpump in *Touch controllers.  There are a few other options in IntelliCenter if you have those types of heaters installed.  If you have a solar preference it will flip between solar and heater depending on whether the solar heater has sufficient temperature deltas to heat the body.

The second place where this information is located is in virtualCircuits.  There are virtual states for the heaters in these values for solar and gas heaters.  *Touch emits solar virtual circuits when a heatpump is engaged.

gadget-monk, 2020-07-27:
I added a comment on Gitter, but thought I would mention here as well. Heater status is inaccurate in “off” mode (I.e not auto). Even after the heater switches on in response to a Spa ON command, njsPC shows heater status as off. If the heating mode is Audo, the status is accurate.

rstrouse, 2020-07-28:
Are you referring to spa manual heat when the heater mode is set to off.  On IntelliCenter this changes the heat mode to heater and operates as if it is heating to a setpoint.  Is this different in *Touch?  If so what do you see in ScreenLogic or the control panels?

gadget-monk, 2020-07-28:
Yes, referring to spa manual heat when the heater mode is set to Off. The heat mode stays Off, but the heater kicks in and heats to a set point. Will take a capture and open a separate issue. Thanks!

closed 2020-07-28 by jbarneson