tagyoureit/nodejs-poolController #200: Heat state not accurate when using manual spa heat with EasyTouch

gadget-monk, 2020-07-31:
**Describe the bug**
poolController heater status is not accurate when heat mode is set to Off and Manual Heat is enabled on EasyTouch systems. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to web client, ensure Heat mode is set to off. In EasyTouch settings enable Manual Heat.
2. Click on 'Spa' button to enable Spa body. Pump and heater turns on. 
3. Scroll down to 'Heat' status in webclient
4. See error - heater status is off (no flame)

**Expected behavior**
Heater status is webclient is consistent with actual heater status. Heater status is accurate in web client when heat mode is auto and heater comes on. 

**Screenshots**
None

**Packet Capture**
Attached

**Pool Equipment**
 - Controller: EasyTouch
 - Controller Model:  EasyTouch2 8
 - Controller Firmware Version: ?
 - Pump(s) manufacturer and model: Intellipro VF
 - Chlorinator: None
 - Heater(s): gas
 - Chemical controller: None
 - Valves: None
 - Any other relevant equipment: None

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.
[replay-heater.zip](https://github.com/tagyoureit/nodejs-poolController/files/5004994/replay-heater.zip)

rstrouse, 2020-08-01:
@tagyoureit I sent you an e-mail regarding heatStatus and heatMode.  This is the replay that I was reviewing at around message id 36201 it starts broadcasting the heatStatus.  Btw, I fixed an issue in messageManager for the response objects this morning.

tagyoureit, 2020-08-02:
Pull the `next` branch and try now.  Replay looks accurate.

gadget-monk, 2020-08-02:
hi, I ran the update, haven't had a chance to test with heater on yet but noticed two issues: 
a. in web client, "either heater" virtual circuit is On even when heater is off (Spa heater and Pool heater virtual circuits are off
b. my homebridge client is broken because the body emit no longer includes HeatStatus

tagyoureit, 2020-08-03:
I just pushed up a quick fix.  I excluded the heatStatus when the tbody was off.  I'm seeing the heatStatus in the emit now.  I'm not seeing "either heater" on, however.  That should be complemented by "pool heater" or "spa heater".  Pull the latest and let me know if that fixes it.

gadget-monk, 2020-08-03:
everything works perfectly, heatStatus is working as it used to earlier (i.e. it is included even if body is off). thanks!

closed 2020-08-03 by gadget-monk