tagyoureit/nodejs-poolController #202: Intellichem data readings ver. 6

rchalloner, 2020-08-11:
[Capture1.zip](https://github.com/tagyoureit/nodejs-poolController/files/5054379/Capture1.zip)
[Capture2.zip](https://github.com/tagyoureit/nodejs-poolController/files/5054380/Capture2.zip)
[Capture3.zip](https://github.com/tagyoureit/nodejs-poolController/files/5054381/Capture3.zip)

**Describe the bug**
Intellichem readings need to be dialed in for new version.  Per tagyourit request, captures attached.

**Packet Capture**
Three captures from three different times have been attached in separate folders along with a text file that includes the correct readings from the Intellichem unit itself at that time of capture.

**Pool Equipment**
 - Controller: [ EasyTouch, ]
 - Controller Model: [EasyTouch2 8]
 - Controller Firmware Version: [2.140]
 - Pump(s) manufacturer and model: [e.g. IntelliFlow 2 VST 011056]
 - Chlorinator: [IntelliChlor-40]
 - Heater(s): [gas, solar, ]
 - Chemical controller: [IntelliChem,]

**Desktop (please complete the following information):**
 - OS: [Rasp]
 - Browser [Chromium]

tagyoureit, 2020-08-11:
per my Gitter message, pull the latest `next` release and try this it.  Should be good.

rchalloner, 2020-08-14:
Intellichem Calcium Hardness reading wrong--on my unit, supposed to be 250 and reads 2048 and the Water balance was 2.3--way high.  (The Calcium Hardness, Alkalinity, and Cyanic Acid are typed in manually by the user at the control panel.  Then the Intellichem calculates the water balance (SI) from ORP, PH, and temperature data using what I typed in.)   My Intellichem math shows a true Alkalinity of 90, Cyanic Acid of 18, ORP of 784, PH of 7.46 and temp of 78 which equals a water balance of -.16, --attached is log from today
[2020-08-14_10-03-28.zip](https://github.com/tagyoureit/nodejs-poolController/files/5076315/2020-08-14_10-03-28.zip)

tagyoureit, 2020-08-17:
These should both be fixed now.  Please re-open if you still have these issue.

closed 2020-08-17 by tagyoureit